alexa-pi/AlexaPi #105: Playback of some radio stations fail

dgonano, 2016-12-11:
**Your OS (including version) where you are running AlexaPi:**
```
Linux raspberrypi 4.4.26-v7+ #915 SMP Thu Oct 20 17:08:44 BST 2016 armv7l GNU/Linux
```

**Your hardware platform and model you are running on:**
```
RaspberryPi 2 (B+?)
```

**Python release (`python2 --version`):**
```
Python 2.7.9
```

**Description of problem:**
```
Playback fails on some radio stations and on some systems. Eg "play triple j radio" works fine but "play nova 1069 radio" gets stuck on my system. 
```

**Steps to reproduce:**
In python
1.  `import tunein`
2. `tunein_parser = tunein.TuneIn(5000)`
3. `tunein_parser.parse_stream_url("http://streaming.novaentertainment.com.au/nova1069")`
Function never returns (or takes exceptionally long)

^c produces same traceback each time:
```
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
  File "tunein.py", line 308, in parse_stream_url
    playlist, content_type = self._get_playlist(url)
  File "tunein.py", line 55, in _memoized
    value = func(*args)
  File "tunein.py", line 386, in _get_playlist
    data = resp.content.decode('utf-8', errors='ignore')
  File "/usr/lib/python2.7/dist-packages/requests/models.py", line 724, in content
    self._content = bytes().join(self.iter_content(CONTENT_CHUNK_SIZE)) or bytes()
  File "/usr/lib/python2.7/dist-packages/requests/models.py", line 653, in generate
    for chunk in self.raw.stream(chunk_size, decode_content=True):
  File "/usr/lib/python2.7/dist-packages/urllib3/response.py", line 256, in stream
    data = self.read(amt=amt, decode_content=decode_content)
  File "/usr/lib/python2.7/dist-packages/urllib3/response.py", line 186, in read
    data = self._fp.read(amt)
  File "/usr/lib/python2.7/httplib.py", line 602, in read
    s = self.fp.read(amt)
  File "/usr/lib/python2.7/socket.py", line 380, in read
    data = self._sock.recv(left)
```

lechat, 2016-12-11:
Looks like network timeout. How long is "exceptionally long"?

dgonano, 2016-12-11:
renekliment had 3 minutes, mine never returns. Not sure how long to leave it...

renekliment, 2016-12-11:
Yeah, this is something we need to solve. I haven't really looked into the `tunein.py` yet, so I don't have ideas (yet). One way or another, we have to implement a real timeout so it doesn't block our code no matter what.

lechat, 2016-12-11:
I've just tried link above, http://streaming.novaentertainment.com.au/nova1069, and it won't open in my browser. I can only see spinner in my Firefox tab...

renekliment, 2016-12-11:
It works for me in Chrome.

dgonano, 2016-12-11:
@lechat works for me in safari and chrome, I don't have firefox.

lechat, 2016-12-11:
Tried again, now it works ok. Hmm...