alexa-pi/AlexaPi #106: Write SoX playback handler

renekliment, 2016-12-11:
This should:
- solve audio problems:
    - fix #63 
    - sound problems on legacy kernel Orange Pi
    - CHIP with internal sound card
- enable us to support OpenWRT
- save space and maybe (just maybe) be a little faster

The current VLC handler isn't great and the whole thing needs a refactor regarding `.stop()` and resuming radio playback and stuff, but that doesn't block this. Also some methods could be reused, so it might not be a bad idea to move them to the base handler, but we'll see how that turns out.

@illperipherals showed interest in doing this, but it is currently free to take.

Random links: https://github.com/rabitt/pysox

enclarify, 2017-01-06:
I have this feature mostly working. I need to cleanup the code and add the sox packages to all platforms this weekend and then will submit a PR for review

renekliment, 2017-01-06:
@enclarify Dope!

ImEmJay, 2017-01-06:
+reneliment Sounds reasonable. It should work better than the vlc handler but including it fast as a optional feature can get a broader test-base faster.

enclarify, 2017-01-06:
@renekliment That makes sense given that I don't have access to all supported platforms for testing

renekliment, 2017-01-16:
Fixed by #140.

commit a97eaf63 (2017-01-16): Add SoX playback handler as alt to VLC - fix #106

The new playback handler is a wrapper for the SoX sound editing/playback
application. In testing it seems to work better than the VLC handler for
some platforms. However it does not currently handle m3u8 streams.
closed 2017-01-16 by renekliment