alexa-pi/AlexaPi #107: Flash Briefing Issues

mochman, 2016-12-12:
**Your OS (including version) where you are running AlexaPi:**
```
Raspbian GNU/Linux 8 (jessie)
```

**Your hardware platform and model you are running on:**
```
raspberry pi 3
```

**Python release (`python2 --version`):**
```
Python 2.7.9
```

**Description of problem:**
```
When running AlexaPi under daemon mode (tested with both systemctl & classic).  When trying to get the "Flash Briefing" by asking "What's New", Alexa replies with "Here's your flash briefing" then goes back to idle.  When running AlexaPi from the command line with "/opt/AlexaPi/src/main.py -d", Alexa works normally by saying "Here's your flash briefing." the continuing with the brief  "Today in the news....".

When Alexa fails, this appears in the logs:

python[6233]: [718094a0] filesystem access error: cannot open file /tmp/AlexaPi-runtime-RaKTUW/DailyBriefingPrompt.750758e8-2ac8-4568-bc4a-7add1a025a1a:ChannelIntroduction:0.
python[6233]: [718094a0] core access error: File reading failed
python[6233]: [718094a0] core access error: VLC could not open the file "/tmp/AlexaPi-runtime-RaKTUW/DailyBriefingPrompt.750758e8-2ac8-4568-bc4a-7add1a025a1a:ChannelIntroduct
python[6233]: [71610f30] core input error: open of `file:///tmp/AlexaPi-runtime-RaKTUW/DailyBriefingPrompt.750758e8-2ac8-4568-bc4a-7add1a025a1a:ChannelIntroduction:0.mp3' fai
python[6233]: [71610f30] core input error: Your input can't be opened
python[6233]: [71610f30] core input error: VLC is unable to open the MRL 'file:///tmp/AlexaPi-runtime-RaKTUW/DailyBriefingPrompt.750758e8-2ac8-4568-bc4a-7add1a025a1a:ChannelI

Looking at the /tmp folder, whenever AlexaPi runs, it creates a folder "AlexaPi-runtime-XXXXXX" chmodded to 700 owned by alexapi.  

Inside that folder, after I ask for the briefing, I see the files appear then dissapear after AlexaPi can't find them.

Running "sudo --user=alexapi /opt/AlexaPi/src/main.py -d" work as it should.   

Also, asking Alexa for the weather, my calendar, general questions works normally.  This problem only seems to occur when AlexaPi has to cycle through multiple audio streams.
```

**Expected:**
```
Alexa will continue with the briefing.
```

**Problem-relevant `config.yaml` entries:**
```

```

**Steps to reproduce:**

1.  Ask AlexaPi "What's New"

**Additional info:**
```
I've tried running everything as root to see if that helps.  It produces the same errors.
```

maso27, 2016-12-14:
I just tried this, and can't reproduce the problem.

Using an RPi 2, with a fresh install from main.

"What's New" yields the Flash Briefing, both when running from command line and after a reboot (systemctl style).

renekliment, 2016-12-14:
I haven't tried yet, but I think I know why this happens (a change in #103 that I didn't think through). I'll try and report / fix later.

dgonano, 2016-12-14:
I can't reproduce this either

mochman, 2016-12-14:
Hmm,  is there a command I can use to produce a detailed log file to see what's going on (like with the command line --debug argument) using systemctl?  The `sudo journalctl -u AlexaPi.service` command just shows me the errors only.

renekliment, 2016-12-16:
@mochman Funny thing is that this doesn't happen in debug mode :smile: No matter, the fix is there (open PR). Please test.

mochman, 2016-12-16:
Thanks!  I'll give it a try tonight.

mochman, 2016-12-17:
@renekliment Your fix works for me!  Thanks.

renekliment, 2016-12-17:
@mochman Great! Merging into `master`. Sorry for the bug.

commit a66c7dad (2016-12-17): revert 'do not block record upload by deleting files' from 14dbe51c5e82afb6c1859294761e369dccd403ce, fixes #107
closed 2016-12-17 by renekliment