alexa-pi/AlexaPi #108: Custom python code

abhigyan2311, 2016-12-13:
Can we have alexa to run custom python files on specific speech recognition ?

maso27, 2016-12-13:
AlexaPi currently never sees a text version of what has been said.  It goes straight to Amazon's servers to be processed.

We would need to add a preliminary step to process the audio ourselves before doing whatever Alexa wanted to do.

In [beagleman](https://github.com/fcooper/beagleman), the developer shunted commands off to https://wit.ai to handle "intents" so even if you didn't say the thing exactly it would determine what you meant.  It would then dump Alexa's response on the floor if the command was something that wit.ai understood.
I played with it for a while but wit.ai deprecated some things I was working on and I haven't gone back to it.

A more simple possibility would be to have a second thread running, listening for a different trigger word, that would process the audio and run scripts.

renekliment, 2016-12-17:
Yeah, this is not something we can do. AlexaPi is a simple interface to the AVS API. For this, you'd need to setup like a Home Assistant with your own skill and then you could do _Alexa, tell home assistant to ..._. Basically this could be done either via Alexa Skills, or your own code based on Pocketsphinx for example. 

Not relevant to this project. Closing.

closed 2016-12-17 by renekliment