alexa-pi/AlexaPi #112: Do I still need a push button called on the gpio pins for Alexa to work?

codyli2002hk, 2016-12-23:
Thank you for reporting an issue with **AlexaPi**.

Make sure you are running the latest version of **AlexaPi**, and look through [issues](https://github.com/alexa-pi/AlexaPi/issues?utf8=%E2%9C%93&q=is%3Aissue) before submitting.

Check out our [Wiki](https://github.com/alexa-pi/AlexaPi/wiki/), and make sure to look at [Debugging](https://github.com/alexa-pi/AlexaPi/wiki/Debugging/), and [Audio debugging](https://github.com/alexa-pi/AlexaPi/wiki/Audio-setup-&-debugging) in particular before filing an issue.

In order to help troubleshooting, be sure to include the following information:

**Output of the [audio debugging script](https://github.com/alexa-pi/AlexaPi/wiki/Audio-setup-&-debugging#audio-debugging-script).**

**Your OS (including version) where you are running AlexaPi:**
```
```

**Your hardware platform and model you are running on:**
```
Raspberry Pi
```

**Python release (`python2 --version`):**
```

```

**Description of problem:**
```

```

**Expected:**
```

```

**Problem-relevant `config.yaml` entries:**
```

```

**Steps to reproduce:**

1. 
2. 
3. 

**Traceback (if applicable):**
```

```

**Additional info:**
```

```

renekliment, 2016-12-23:
No, you don't. You can use voice activation only.

closed 2016-12-23 by renekliment

datafaust, 2017-01-12:
Quick question on this. I managed to get Alexapi running on my CHIP through voice activation, that is she boots straight up runs and I don't need to press a button, I can just call out and she will answer. Is this a heavenly error on my part or an update from the repo? As I see in the documentation 'devices' section there is still a reference to using gpio pins and button for recording and calling Alexa with CHIP. Thanks!

renekliment, 2017-01-12:
@datafaust You can use both, that's the point.

datafaust, 2017-01-13:
Understood thank you, following up on that , if i choose to go with voice activation, is it still possible to set lighting while recording and lighting while playing? My platform is set to "dummy", does it need to be changed to"CHIP" in order to activate lighting response from pin XIO-p5 and if so does this sacrifice lighting? Appreciate the help,