alexa-pi/AlexaPi #113: Adding Hyperion support for playback/record status

penfold42, 2016-12-26:
I want to use Hyperion (ambient lighting) to display the playback record status.

It's a really simple json request over tcp to set the led colours.

Should I do this as a separate platform or could we do this for all platforms ?

renekliment, 2016-12-26:
Hey. Currently, the best approach is indeed a new _device_platform_. You can inherit from whatever existing platform you like (if you want to keep its function) and just add your code.

renekliment, 2017-01-01:
@penfold42 Do you have any further questions / help requests / code to share?

ImEmJay, 2017-01-02:
@renekliment Because I also like to have Hyperion support and your 'device_platform' makes it really easy I already started implementing it.

@penfold42 I think I'm ready to push a first into a public branch in my repo within the next days. Would you like to do some testing then?

penfold42, 2017-01-03:
Yep - I have to get it working again tho which is why inhavemt done anything yet.

I was going to NOT use Hyperion-remote by rather use the json interface over tcp to set the colour.
We should expose the actual json string in the config file too

ImEmJay, 2017-01-05:
You can already set the colors independent for each state. Also you can choose between static color or 'knight rider' effect for all states. Additionaly you can override each state t use flashing notifications as you can see in the video when alexa is triggering the playback state.
I think not all effects would look good as notification and it's a bit tricky cause they all have different parameters. But im looking to integrate/customize more of the default effects. Still i would avoid to just pass the full args string as a config option.

renekliment, 2017-01-13:
This was fixed in `dev` by #127.

closed 2017-01-13 by renekliment