alexa-pi/AlexaPi #114: Install script fails at yaml

Kenno9000, 2016-12-27:
Gosh, I really want this to work, my idea needs it live in the pi and have other things going on.  I had hoped to get to the point where the sound card insanity hits again.  That part about making a dx script for the sound issues?  Please!  This seems so much cleaner than the Java implementation, where I fell down on the sound issues for the keyword detectors. 

pi 3 B fresh NOOBS heś got a head on, 3.5mm speaker, usb mic, cat 6 wire,  

Configuration file /etc/opt/AlexaPi/config.yaml exists already. What do you want to do?
[0] Keep and use current configuration file.
[1] Edit existing configuration file.
[2] Delete the configuration file and start with a fresh one.
Which option do you prefer? [hit Enter for 0]: 2
Creating configuration file /etc/opt/AlexaPi/config.yaml ...
Enter your Device Type ID []: RPIBot4
Enter your Security Profile Description []: RPIBot4_Stella-Bot
Enter your Security Profile ID []: amzn1.a-------I removed this stuff-----------------------0f
Enter your Client ID []: amzn1.a-------I removed this stuff-----------------------
Enter your Client Secret []: amzn1.a-------I removed this stuff-----------------------
___________________________________________________________________________________
Traceback (most recent call last):
  File "./auth_web.py", line 8, in <module>
    import yaml
ImportError: No module named yaml
_-------------------------------------------------------------------------------------------------------------------

renekliment, 2016-12-27:
It's not an issue with the config, but the python module missing.

> `ImportError: No module named yaml`

That should have been installed though. Can you provide the whole install log?

Kenno9000, 2016-12-27:
Thanks for that.  I have gotten past that point now.

closed 2016-12-27 by Kenno9000

Renard2323, 2017-01-23:
Thanks for sharing how you did it, so It can help new people? you know, people with trouble seeking help, like you were back then?

Exact same issue (very very beggining here)

I get this when checking status

pi@raspberrypi:~ $ sudo systemctl status AlexaPi.service
● AlexaPi.service - Alexa client for all your devices
   Loaded: loaded (/usr/lib/systemd/system/AlexaPi.service; enabled)
  Drop-In: /etc/systemd/system/AlexaPi.service.d
           └─restart.conf
   Active: failed (Result: start-limit) since Mon 2017-01-23 09:17:35 +05; 4min 59s ago
     Docs: https://github.com/alexa-pi/AlexaPi/wiki
  Process: 870 ExecStart=/usr/bin/python /opt/AlexaPi/src/main.py (code=exited, status=1/FAILURE)
 Main PID: 870 (code=exited, status=1/FAILURE)

Jan 23 09:17:34 raspberrypi systemd[1]: Unit AlexaPi.service entered failed state.
Jan 23 09:17:35 raspberrypi systemd[1]: AlexaPi.service holdoff time over, scheduling restart.
Jan 23 09:17:35 raspberrypi systemd[1]: Stopping Alexa client for all your devices...
Jan 23 09:17:35 raspberrypi systemd[1]: Starting Alexa client for all your devices...
Jan 23 09:17:35 raspberrypi systemd[1]: AlexaPi.service start request repeated too quickly, refusing to start.
Jan 23 09:17:35 raspberrypi systemd[1]: Failed to start Alexa client for all your devices.
Jan 23 09:17:35 raspberrypi systemd[1]: Unit AlexaPi.service entered failed state.
Jan 23 09:22:10 raspberrypi systemd[1]: Stopped Alexa client for all your devices.

Warning: Unit file changed on disk, 'systemctl daemon-reload' recommended.