alexa-pi/AlexaPi #115: After reboot, hearing "Hello" but AlexaPi is not working - No errors

supakdee, 2016-12-27:
Hi,

I'm setting up AlexaPi on Raspberry Pi 3 running Raspbian.

I could finish installation & getting token installed. Once reboot, I could hear "Hello" but it's not working when I say "Alexa"... No particular error as you see below.

When I use "run AlexaPi interactively in debug mode", it runs & stop after saying "Hello".

Can you share what could be the issue? or what i should do to fix this?

By the way, are there any videos that I can see what I should get by running those commands if it works fine?

**pi@raspberrypi:~ $ sudo systemctl status AlexaPi.service -l**
● AlexaPi.service - Alexa client for all your devices
   Loaded: loaded (/usr/lib/systemd/system/AlexaPi.service; enabled)
  Drop-In: /etc/systemd/system/AlexaPi.service.d
           └─restart.conf
   Active: active (running) since Tue 2016-12-27 14:42:14 ICT; 1min 19s ago
     Docs: https://github.com/alexa-pi/AlexaPi/wiki
 Main PID: 628 (python)
   CGroup: /system.slice/AlexaPi.service
           └─628 /usr/bin/python /opt/AlexaPi/src/main.py

Dec 27 14:42:14 raspberrypi systemd[1]: Started Alexa client for all your devices.

**pi@raspberrypi:~ $ sudo journalctl -u AlexaPi.service**
-- Logs begin at Tue 2016-12-27 14:42:10 ICT, end at Tue 2016-12-27 14:46:37 ICT. --
Dec 27 14:42:14 raspberrypi systemd[1]: Starting Alexa client for all your devices...
Dec 27 14:42:14 raspberrypi systemd[1]: Started Alexa client for all your devices.

**/opt/AlexaPi/src/main.py –d**
Checking Internet Connection...
Connection OK
Play_Audio Request for: /opt/AlexaPi/src/resources/hello.mp3
Player State: State.Opening
Player State: State.Playing
Player State: State.Ended

supakdee, 2016-12-27:
Oh... My set up is the audio through 3.5mm jack /// recorder through USB Mic.

Below is sound section of my config.yaml

**sound:**
  input_device: "plughw:1,0"
  playback_handler: "vlc"
  output: "alsa"
  output_device: "plughw:0,0"
  default_volume: 80
  media_output:
  media_output_device:
  media_default_volume:

renekliment, 2016-12-27:
What do you mean by 

>  it runs & stop after saying "Hello"

-- do you get an error or something?

It looks like everything is running fine according to the logs. Sometimes it's tricky to trigger it via voice. Have you tried a GPIO button? Also, you might need to adjust your audio input levels in `alsamixer`.

renekliment, 2016-12-27:
... and no videos. I might make one if there is demand.

Kenno9000, 2016-12-29:
Hi,
I'm making some progress (i.e. learning a lot), and I now appear to be at the same spot as supakdee, but I have an indication in the log.  It's long & repeatable, mic is set at 80 on a usb Soundblaster Play 2, speaker at about 46 on the 3.5 plug.  Noobs clean, update, installed.  AlPi got by git today.  Logs to Amazon and says Hello, running in a terminal, but same error when run from boot. Error report follows...

"I have the greatest enthusiasm for the mission, Dave."   I'm a Social Worker in New Zealand, wanting to make a bespoke companion for people with dementia, my blue-sky is to add Alexa Skills on top.  Any thoughts on whether this seems feasible?  

Keep up the good work, if we can get it going, there are a lot of old people (and some not so old) whose lives will really benefit. 

.root@rpi5:/home/pi# /opt/AlexaPi/src/main.py -d
INFO: pocketsphinx.c(152): Parsed model-specific feature parameters from /usr/local/lib/python2.7/dist-packages/pocketsphinx/model/en-us/feat.params
Current configuration:
[NAME]			[DEFLT]		[VALUE]
-agc			none		none
-agcthresh		2.0		2.000000e+00
-allphone				
-allphone_ci		no		no
-alpha			0.97		9.700000e-01
-ascale			20.0		2.000000e+01
-aw			1		1
-backtrace		no		no
-beam			1e-48		1.000000e-48
-bestpath		yes		yes
-bestpathlw		9.5		9.500000e+00
-ceplen			13		13
-cmn			live		batch
-cmninit		40,3,-1		41.00,-5.29,-0.12,5.09,2.48,-4.07,-1.37,-1.78,-5.08,-2.05,-6.45,-1.42,1.17
-compallsen		no		no
-debug					0
-dict					/usr/local/lib/python2.7/dist-packages/pocketsphinx/model/cmudict-en-us.dict
-dictcase		no		no
-dither			no		no
-doublebw		no		no
-ds			1		1
-fdict					
-feat			1s_c_d_dd	1s_c_d_dd
-featparams				
-fillprob		1e-8		1.000000e-08
-frate			100		100
-fsg					
-fsgusealtpron		yes		yes
-fsgusefiller		yes		yes
-fwdflat		yes		yes
-fwdflatbeam		1e-64		1.000000e-64
-fwdflatefwid		4		4
-fwdflatlw		8.5		8.500000e+00
-fwdflatsfwin		25		25
-fwdflatwbeam		7e-29		7.000000e-29
-fwdtree		yes		yes
-hmm					/usr/local/lib/python2.7/dist-packages/pocketsphinx/model/en-us
-input_endian		little		little
-jsgf					
-keyphrase				alexa
-kws					
-kws_delay		10		10
-kws_plp		1e-1		1.000000e-01
-kws_threshold		1		1.000000e-05
-latsize		5000		5000
-lda					
-ldadim			0		0
-lifter			0		22
-lm					
-lmctl					
-lmname					
-logbase		1.0001		1.000100e+00
-logfn					
-logspec		no		no
-lowerf			133.33334	1.300000e+02
-lpbeam			1e-40		1.000000e-40
-lponlybeam		7e-29		7.000000e-29
-lw			6.5		6.500000e+00
-maxhmmpf		30000		30000
[... 83 lines cut ...]
INFO: dict.c(213): Dictionary size 134723, allocated 1016 KiB for strings, 1679 KiB for phones
INFO: dict.c(336): 134723 words read
INFO: dict.c(358): Reading filler dictionary: /usr/local/lib/python2.7/dist-packages/pocketsphinx/model/en-us/noisedict
INFO: dict.c(213): Dictionary size 134728, allocated 0 KiB for strings, 0 KiB for phones
INFO: dict.c(361): 5 words read
INFO: dict2pid.c(396): Building PID tables for dictionary
INFO: dict2pid.c(406): Allocating 42^3 * 2 bytes (144 KiB) for word-initial triphones
INFO: dict2pid.c(132): Allocated 21336 bytes (20 KiB) for word-final triphones
INFO: dict2pid.c(196): Allocated 21336 bytes (20 KiB) for single-phone word triphones
INFO: kws_search.c(406): KWS(beam: -1080, plp: -23, default threshold -113, delay 10)
Checking Internet Connection...
Connection OK
Play_Audio Request for: /opt/AlexaPi/src/resources/hello.mp3
Player State: State.Opening
Player State: State.Playing
Player State: State.Ended
Traceback (most recent call last):
  File "/opt/AlexaPi/src/main.py", line 587, in <module>
    loop()
  File "/opt/AlexaPi/src/main.py", line 518, in loop
    _, buf = inp.read()
alsaaudio.ALSAAudioError: Capture data too large. Try decreasing period size
INFO: kws_search.c(448): TOTAL kws 0.00 CPU nan xRT
INFO: kws_search.c(451): TOTAL kws 0.00 wall nan xRT
root@rpi5:/home/pi#

Kenno9000, 2017-01-01:
HI again,

I am please to report that AlexaPi is working for me now.
The key to the solution seems to have been installing Jessie as a simple disk image and not, as I had previously used NOOBS.  A Newbi mistake, for sure, I get that now, but I bet I'm not alone.  It was the clear remark in another project's install guide that tipped me to this, so I was able to come back and get it right. 

For me this issue is resolved. Thanks for your time.

boblautenbach, 2017-01-03:
I am having the same hello issue....I loaded Jessie...but any other tips/config I should know?

maso27, 2017-01-03:
@boblautenbach 
When you say you loaded Jessie - did you install Jessie Lite, or the full desktop version?
If you loaded the full desktop version, you will encounter issues with Pulse Audio.

Are you seeing errors? (`sudo journalctl -u AlexaPi`)
Is it crashing or still running? (`ps ax | grep AlexaPi`, should show _main.py_ running)

You can debug it real-time with:
`sudo systemctl stop AlexaPi` (to stop the daemon)
`sudo /opt/AlexaPi/src/main.py -d` (to start it in the terminal, in _debug_ mode)

boblautenbach, 2017-01-03:
Its the full version.  I will try logging and get back to you

boblautenbach, 2017-01-03:
here is what I am seeing 
[image]

maso27, 2017-01-03:
@boblautenbach unless you need the desktop version of Raspbian, I'd highly recommend running Jessie Lite.

If you want to make the desktop version work, check the PulseAudio section [here](https://github.com/alexa-pi/AlexaPi/wiki/Audio-setup-&-debugging).

boblautenbach, 2017-01-03:
I think I will just go the Jessie Lite route..thanks for the help !

boblautenbach, 2017-01-03:
@maso27 won't I need at least lightdm for when I need to auth in the browswer?  Or are you doing that on another box and just putting that IP in the return URLS on the alexa site?

maso27, 2017-01-03:
@boblautenbach 
Authorization is done from the browser of a different machine on the same network.  

As part of the installation process, you register the IP address of you Pi at developer.amazon.com (as instructed [here](https://github.com/alexa-pi/AlexaPi/wiki/Installation)) and after the setup script has run, it instructs you to browse to that address.  

The auth key is automatically generated and saved.

boblautenbach, 2017-01-03:
@maso27 I realized how dumb that question was after I typed it : )

maso27, 2017-01-03:
@boblautenbach saw your edit, but you never need the IP of the second box. Just the address of the pi.  The auth script run by setup.sh takes care of it.

renekliment, 2017-01-06:
Seems resolved, closing. Feel free to reopen when needed.

closed 2017-01-06 by renekliment

pfiol, 2017-02-06:
Hi,

I'm affraid I have to reopen... I've exactly the same issue.
- I've installed Raspbian Jessy Lite from scratch.
- Followed all the installation instructions.
- If I reboot I can hear the word "Hello". 
- But if I say "alexa" nothing happens (I've tried a hundred times, just in case).

No errors on the logs, nothing that could give me a clue on where the problem is...

Here they are:
INFO: pocketsphinx.c(152): Parsed model-specific feature parameters from /usr/local/lib/python2.7/dist-packages/pocketsphinx/model/en-us/feat.params
Current configuration:
[NAME]			[DEFLT]		[VALUE]
-agc			none		none
-agcthresh		2.0		2.000000e+00
-allphone				
-allphone_ci		no		no
-alpha			0.97		9.700000e-01
-ascale			20.0		2.000000e+01
-aw			1		1
-backtrace		no		no
-beam			1e-48		1.000000e-48
-bestpath		yes		yes
-bestpathlw		9.5		9.500000e+00
-ceplen			13		13
-cmn			live		batch
-cmninit		40,3,-1		41.00,-5.29,-0.12,5.09,2.48,-4.07,-1.37,-1.78,-5.08,-2.05,-6.45,-1.42,1.17
-compallsen		no		no
-debug					0
-dict					/usr/local/lib/python2.7/dist-packages/pocketsphinx/model/cmudict-en-us.dict
-dictcase		no		no
-dither			no		no
-doublebw		no		no
-ds			1		1
-fdict					
-feat			1s_c_d_dd	1s_c_d_dd
-featparams				
-fillprob		1e-8		1.000000e-08
-frate			100		100
-fsg					
-fsgusealtpron		yes		yes
-fsgusefiller		yes		yes
-fwdflat		yes		yes
-fwdflatbeam		1e-64		1.000000e-64
-fwdflatefwid		4		4
-fwdflatlw		8.5		8.500000e+00
-fwdflatsfwin		25		25
-fwdflatwbeam		7e-29		7.000000e-29
-fwdtree		yes		yes
-hmm					/usr/local/lib/python2.7/dist-packages/pocketsphinx/model/en-us
-input_endian		little		little
-jsgf					
-keyphrase				alexa
-kws					
-kws_delay		10		10
-kws_plp		1e-1		1.000000e-01
-kws_threshold		1		1.000000e-05
-latsize		5000		5000
-lda					
-ldadim			0		0
-lifter			0		22
-lm					
-lmctl					
-lmname					
-logbase		1.0001		1.000100e+00
-logfn					
-logspec		no		no
-lowerf			133.33334	1.300000e+02
-lpbeam			1e-40		1.000000e-40
[... 81 lines cut ...]
INFO: ptm_mgau.c(595): Using memory-mapped I/O for senones
INFO: ptm_mgau.c(838): Maximum top-N: 4
INFO: phone_loop_search.c(114): State beam -225 Phone exit beam -225 Insertion penalty 0
INFO: dict.c(320): Allocating 138824 * 20 bytes (2711 KiB) for word entries
INFO: dict.c(333): Reading main dictionary: /usr/local/lib/python2.7/dist-packages/pocketsphinx/model/cmudict-en-us.dict
INFO: dict.c(213): Dictionary size 134723, allocated 1016 KiB for strings, 1679 KiB for phones
INFO: dict.c(336): 134723 words read
INFO: dict.c(358): Reading filler dictionary: /usr/local/lib/python2.7/dist-packages/pocketsphinx/model/en-us/noisedict
INFO: dict.c(213): Dictionary size 134728, allocated 0 KiB for strings, 0 KiB for phones
INFO: dict.c(361): 5 words read
INFO: dict2pid.c(396): Building PID tables for dictionary
INFO: dict2pid.c(406): Allocating 42^3 * 2 bytes (144 KiB) for word-initial triphones
INFO: dict2pid.c(132): Allocated 21336 bytes (20 KiB) for word-final triphones
INFO: dict2pid.c(196): Allocated 21336 bytes (20 KiB) for single-phone word triphones
INFO: kws_search.c(406): KWS(beam: -1080, plp: -23, default threshold -113, delay 10)
Checking Internet Connection...
Connection OK
Play_Audio Request for: /opt/AlexaPi/src/resources/hello.mp3
Player State: State.Opening
Player State: State.Playing
Player State: State.Ended

Any idea on where should I look?

Thanks!

pfiol, 2017-02-06:
I forgot to mention that it doesn't crash nor exit, it just don't work. Nothing happens.

renekliment, 2017-02-07:
@pfiol Looks like AlexaPi works, but is having problems with recognition.

Try the `dev` branch https://github.com/alexa-pi/AlexaPi/wiki/Q&A-(FAQ)#how-to-try-the-dev-branch and if the new threshold value doesn't work for you, try to tweak it.

pfiol, 2017-02-07:
@renekliment OK. Will try it later when I get home. Is there any other way to debug pocketsphinx?

I must say that I've tried other Alexa projects and using snowboy was the best option so far I hope you add it to this project as well.  Right now the main reason to use your code is launching alexa from command-line without any GUI interface. So, if you add Snowboy it will be just perfect :)

renekliment, 2017-02-07:
@pfiol Not really. You can try a standalone example from https://pypi.python.org/pypi/pocketsphinx/0.1.3

> ```
> from pocketsphinx import LiveSpeech
> 
> speech = LiveSpeech(lm=False, keyphrase='forward', kws_threshold=1e+20)
> for phrase in speech:
>     print(phrase.segments(detailed=True))
> ```

Yes, we will add snowboy (#53). We have refactored the whole triggering thing, so it can be a module now and you'll actually be able to choose between pocketsphinx and snowboy (not like other repos, where there is only one hardcoded). It is prepared, so it should be relatively easy to do, but no one got around implementing it. You can give it a try, if you want to :smile:

pfiol, 2017-02-15:
Yesterday I finally had time to test the dev branch. Exactly same problem :(

I'm starting to think that is not taking the correct input device for voice recognition although arecord -l only shows one result and "arecord -d 5 test.wav" works smoothly. 

I've also tried to modify the keyphrase to something else just in case there's a problem with my "alexa" pronunciation. I've tried several, nothing worked so far...

renekliment, 2017-02-15:
@pfiol And have you tried to tweak the threshold on the `dev` branch? More docs here: https://github.com/alexa-pi/AlexaPi/wiki/Triggers#pocketsphinx

pfiol, 2017-02-15:
I didn't had time enough. I will do some more testing this afternoon. I will let you know my findings.

Just for curiosity, is this setup supposed to "say" something after you call "alexa"? Something like "yes" a "beep" or similar? Maybe I'm just waiting to listen something and I'm not supposed to...

pfiol, 2017-02-15:
Ok, I've tried the magic number (threshold), and it finally worked. However, I had to increase it A LOT for it to be functional, and in triggers even when it shouldn't.

Previously I've tried the original Amazon GitHub repo, which uses the Kitt_ai/snowboy as voice trigger and it works MUCH better. I'm not a python developer, however if you could point me out where to look at maybe I can try to replace pocketsphinx for snowboy.

renekliment, 2017-02-15:
@pfiol See #53. It should be implemented as another trigger, so people can choose whichever engine they want to. A lot of work has been put into refactoring the triggering stuff (it's in `dev`), so we could also support snowboy, but no one had time to actually do the snowboy trigger.

pfiol, 2017-02-23:
Hi @renekliment 

I would say I'm  almost there. I've now integrated snowboy into the dev branch, it launches, and it triggers when I pronounce the word. But... I'm stuck there, I don't know how to give back the control to the app... I've noticed that in the other triggers you just put this:

`self._trigger_callback(self)`

And it does the magic... but If I use that same code on snowboy It just won't work. It's the first time I program in python actually. I'm sure it must be some stupid detail, but still I'm not capable of moving forward.

This is the code I'm currently using on my snowboytrigger.py:
```
                self._decoder.start(detected_callback=self._trigger_callback(self),
                        interrupt_check=interrupt_callback,
                        sleep_time=0.03)
```

I've essentially copied the file structure you created for the pocketsphinxtrigger, and as I said, the snowboy software is lunched and it detects when I say the magic word (I'm capable of printing some text when it does), but I don't know how to give control to the rest of the tool... Any idea?

renekliment, 2017-02-24:
@pfiol Great news!

I think the problem is in `detected_callback=self._trigger_callback(self)`, because you need to pass the function pointer (or whatever equivalent in python), but when you do `()`, that's evaluating the function. But you also need to pass the `self`, so try

```
                self._decoder.start(detected_callback=lambda: self._trigger_callback(self),
                        interrupt_check=interrupt_callback,
                        sleep_time=0.03)
```

I'm not sure how is that gonna work with the `self` in lambda ... Maybe not. See http://stackoverflow.com/a/13783227 for other options.

You can open a PR even if it is not finished yet so other people can have a look at the whole thing / test, etc.

lechat, 2017-02-24:
`lambda` shall work, but currying with `partial` looks a bit cleaner. I always stumble on lambdas and never had any problems with partials. :)

pfiol, 2017-02-24:
@renekliment  @lechat  Thank you both for the tips. I will try them out later today when I get home and will let you know how it goes :)

pfiol, 2017-02-24:
OK, I had time now to test your suggestion. There's some progress but still no joy... :(

I've traced the app and now it gets stuck at the "trigger_callback" function (on main.py), on the line where it says: `triggers.disable()`. It gets stuck there and doesn't move forward... Any idea?

renekliment, 2017-02-24:
@pfiol Open a PR with your code.

pfiol, 2017-02-24:
@renekliment Done: https://github.com/alexa-pi/AlexaPi/pull/174

Please don't laugh at me, it's the first time I do it...

ausaleh, 2018-07-01:
I am Having the same issue. But try checking this site. [https://github.com/alexa/alexa-avs-sample-app/issues/575](url)