alexa-pi/AlexaPi #117: Music is stopped if interrupted with a new query

ksanghai, 2016-12-28:
The music/media stops playing, if it is interrupted with a new query/request. Is this expected? I am guessing it is the limitation of the AVS v1 api and to be fixed with the new avs api?

renekliment, 2016-12-28:
Hey! It is expected, but probably not desired, meaning it's the behavior that was implemented in the code and no one has gotten around changing that yet. It requires a few things, so it's not a five-minute fix.

And no, it doesn't have anything to do with the API version.

ksanghai, 2017-01-03:
SoX playback handler might be a more elegant way to do this but I managed to make this work with software mixing using alsa. My USB sound card can now play multiple streams simultaneously. Here's my [https://github.com/ksanghai/AlexaPi/blob/duplex_audio/asoundrc](asoundrc file) that does the audio dmixing and here
s the original link from [https://www.alsa-project.org/main/index.php/Asoundrc#dmix](Alsa) on software mixing.

Also, I have both the media and the speech working pretty well with the [https://github.com/ksanghai/AlexaPi/tree/duplex_audio](older repo)... 
Its what I was working with originally and was familiar with. 

Ofcourse, this is one way to do it.. just wanted to share that.

ksanghai, 2017-01-03:
SoX playback handler might be a more elegant way to do this but I managed to make this work with software mixing using alsa. My USB sound card can now play multiple streams simultaneously. Here's my [asoundrc file](https://github.com/ksanghai/AlexaPi/blob/duplex_audio/asoundrc) does the audio dmixing and here's
the [original link from alsa](https://www.alsa-project.org/main/index.php/Asoundrc#dmix) on software mixing.

Also, I have both the media and the speech working pretty well with the [older repo](https://github.com/ksanghai/AlexaPi/tree/duplex_audio)...
Its what I was working with originally and was familiar with.

Ofcourse, this is one way to do it.. just wanted to share that.

renekliment, 2017-01-07:
As pointed out by @enclarify, the current progress reporting is broken. Acceptable values for playerActivity are IDLE, PAUSED and PLAYING and we feed it various crap. Also, we should send `offsetInMilliseconds`, so the playback can be later resumed. Basically go over the docs for all the events and make sure we honor the API.

Links:
- https://developer.amazon.com/public/solutions/alexa/alexa-voice-service/rest/audioplayer-events-requests

renekliment, 2017-01-07:
This should also fix audiobook pausing / resuming (#118).