alexa-pi/AlexaPi #118: Audiobook resuming

lsumner4, 2016-12-29:
I am using V1.3 of AlexaPi on my Raspberry Pi and I have found an issue with the playing of audiobooks through audible. Although Alexa says she is resuming the book, she instead restarts at the beginning of the book each time.

renekliment, 2016-12-29:
Could you write up steps to reproduce? (exactly what you say to Alexa) Ideally with some public / free book (I'm not sure how it works, I've tried it like once).

lsumner4, 2016-12-30:
I was thinking that the new API may be needed but I just wanted to present the issue so it was known.

renekliment, 2016-12-30:
Yep, good that you did. We will hopefully fix this. You can dig deeper if you want to, it's possible that I'm wrong about the API.

renekliment, 2017-01-07:
Correction: This is probably a bug on our side, because we use the current API incorrectly. Task #117.

renekliment, 2017-01-07:
Closing this as a duplicate of that other task.

closed 2017-01-07 by renekliment