alexa-pi/AlexaPi #121: Improve config file updating

dgonano, 2017-01-01:
During setup it asks the following:
```
echo "Configuration file $CONFIG_FILE exists already. What do you want to do?"
    echo "[0] Keep and use current configuration file."
    echo "[1] Edit existing configuration file."
    echo "[2] Delete the configuration file and start with a fresh one."
    read -p "Which option do you prefer? [hit Enter for 0]: " config_action
```

It would be good that when you selection option 2 the default security parameters are pulled from the current file like option 1. 
if it's a new file the defaults should be blank like normal.

renekliment, 2017-01-01:
Yes, that makes sense. For example when the user upgrades to a new version and doesn't want to manually update the config, at least the AVS settings are kept.

lechat, 2017-01-14:
I think that it is better to have this as another option: "Upgrade existing configuration file"

dgonano, 2017-01-14:
You will be give the option as normal. The current setting will be in the brackets (default) so you can hit enter or put your new value in.