alexa-pi/AlexaPi #122: Make all the sound paths configurable

renekliment, 2017-01-01:
We distribute some sound files in the `src/resources/` directory. We should make them all configurable. Particularly Alexa's "yes" and "hello" might get annoying over a time, so users might want to replace them with something else.

We should use something like `.replace('{resources_path}', resources_path)` and set the paths in the config relative to the `{resources_path}` directory.

lechat, 2017-01-09:
Why it needs to be relative to `{resources_path}` directory?

renekliment, 2017-01-09:
It doesn't have to be, but we need to link to our resources directory somehow. Also, it's not a good idea to hard-code paths in the python code - let's figure it our relatively to the running script. And, if the thing is empty, we wouldn't play anything.

lechat, 2017-01-09:
We can always convert relative path to absolute and we can keep our resource dir hardcoded. But asking users to move files into our resources dir is a bit too much. We can have "user_resources" config item and use it for file locations.

Agree on emptiness and same shall be for file not exist/not accessible (with warning in log about this).

renekliment, 2017-01-09:
Well, how would you mark / recognize this relative path?
The intention is definitely not to make users put files in our app folder.

sammachin, 2017-03-10:
As I mentioned in #194 I think making these config vars with the full file path seems the way to go then we can include our default sounds in resources and users can put custom ones anywhere they choose

renekliment, 2017-03-10:
I'm not sure, what you mean by "full file path", but it should work like this:

1. Empty = don't play anything
2. Something - replace `{resources_path}` (if it's there, else it does nothing) with the path to our `resources` directory, so we have our files linked. This should be default ... e.g. `{resources_path}/alexayes.mp3` and if people want their own stuff, they will do an absolute path like `/etc/opt/AlexaPi/sounds/my_custom_voice_confirm.mp3`

sammachin, 2017-03-10:
yeah sorry `full file path` = `absolute path`

Is an empty value going to be the most efficient, I guess that only really matters in terms of the wakeword prompt which already has a bool to switch it off.

renekliment, 2017-03-10:
> Is an empty value going to be the most efficient, I guess that only really matters in terms of the wakeword prompt which already has a bool to switch it off.

Well, yeah, there will be two approaches to actually turn off the "yes", though they will be semantically different. I wouldn't worry about that much.