alexa-pi/AlexaPi #124: amazon has an error and not accepting my authorization

baum1234, 2017-01-02:
I just installed everything on my Raspberry Pi.
After the setup I went to http://192.168.2.2:5050 and accepted the authorization to my account. After pressing "Ok", I get this message (in German):

> Tut uns leid!
Beim Bearbeiten Ihres Auftrags ist ein Fehler aufgetreten. Wir arbeiten bereits daran und haben das Problem in Kürze gelöst.

That just means that there is an error and that they are working on it.
When I visit https://www.amazon.de/ap/adam or https://www.amazon.com/ap/adam there is no entry.

Had anyone the same problem and found a solutoin

melvinhertz, 2017-01-03:
I had this problem. I had to create a new device and call it my_device and then it worked. I think I found that solution on the amazon alexa github page somewhere.

melvinhertz, 2017-01-03:
Also, make sure your return url code is exactly right in your security profile.

hackerseraph, 2017-01-04:
my_device resolved this for me as well. thanks team.

baum1234, 2017-01-04:
silly question, how do I start AlexPi after I ran the setup once?

dgonano, 2017-01-04:
`sudo service AlexaPi Start` or `src/main.py -d` for debugging

renekliment, 2017-01-06:
@baum1234 If this is resolved, please close the issue. Thank you.

closed 2017-01-07 by baum1234