alexa-pi/AlexaPi #126: Sinatra on Tunein Radio

jswalls110, 2017-01-04:
**Output of the [audio debugging script](https://github.com/alexa-pi/AlexaPi/wiki/Audio-setup-&-debugging#audio-debugging-script).**

**Your OS (including version) where you are running AlexaPi:**

Most current pull

**Your hardware platform and model you are running on:**
RPi 3

**Python release (`python2 --version`):**

2.7.9

**Description of problem:**
Alexa does basic things very well. It will play iHeart radio for my selections. I have tried a few Tunein stations.

When I ask for Frank Sinatra, or a specific station "Mostly Sinatra" Alexa will crash. Most time only a reboot will bring it back to working order

**Expected:**
Play the station

**Problem-relevant `config.yaml` entries:**
None

**Steps to reproduce:**

1. Alexa
2. Play Mostly Sinatra on Tunein Radio

1. Alexa Play Sinatra 100 Radio on Tunein Radio

**Additional info:**
Not devastating but this is mostly what I listen to so disturbing.

renekliment, 2017-01-04:
Confirming.
For 1. I get normal AVS response - Alexa telling me "I'm not sure what went wrong". There's nothing we can do AFAIK.
For 2. I get a TuneIn URL, where Alexa handles it badly. The URL doesn't work regardless, but it shouldn't knock out Alexa. I think this is similar to #105. We need to handle errors from the TuneIn lib better and stuff.

jswalls110, 2017-01-04:
Thanks for confirming Rene.

I wanted to make sure I wasn't doing something wrong.