alexa-pi/AlexaPi #137: Windows support

renekliment, 2017-01-07:
Should be doable after #45. Custom install might be needed.

EmerickH, 2017-06-01:
New repo created
https://github.com/alexa-pi/AlexaPi-Windows
Project in progress!

renekliment, 2017-06-01:
Great! You should base it on this repo though so we can track the changes and merge it when it's ready.

EmerickH, 2017-06-01:
It's a little bit complicated because I had to change completely certain files..
Is it possible to link just main.py for example?

renekliment, 2017-06-01:
Yes, I've expected that the install scripts would have to be completely different. You don't have to use the same names though. Install scripts for Linux and Windows can be separate, side by side in one repo. If there is something that can be shared, it can.

Not really. And the later you make this, the more difficult it would be to merge. It should have the same base from the start.

EmerickH, 2017-06-14:
@renekliment can you close this issue now?
Can I have he permission to add Labels to issues (and maybe also close them) because its very messy!

Thank you,
Emerick

renekliment, 2017-06-14:
I'd leave this open until we have proper docs about it in our wiki.

If you are sure that everything from https://github.com/alexa-pi/AlexaPi-Windows-DEPRECATED is merged, delete the whole commit history with `git rebase` and just leave a message to go to this repo so we keep things nice and clean.

Yep, will do.

renekliment, 2017-06-14:
@EmerickH Added. Be sure to comply with the project rules. Especially https://github.com/alexa-pi/AlexaPi/blob/master/CONTRIBUTING.md#for-core-developers