alexa-pi/AlexaPi #139: Issue with audio output front_center.wav

datafaust, 2017-01-07:
Ran into some audio issues after installing. Appreciate any help!

**Output of the [audio debugging script](https://github.com/alexa-pi/AlexaPi/wiki/Audio-setup-&-debugging#audio-debugging-script).**

http://www.alsa-project.org/db/?f=1dad84c07847476c65f18e561ba4bf9e8f331366

**Your OS (including version) where you are running AlexaPi:**
```I'm running CHIP o.s. debian kernal 4.3
``Kernel release:    4.3.0-ntc
Operating System:  GNU/Linux

**Your hardware platform and model you are running on:**
``` CHIP

```

**Python release (`python2 --version`):**
```I've been running under python 3 

```

**Description of problem:**
``` I followed installation directions without issue, under root and configured with the "root" issues as instructed. I wanted to test audio so I've been playing:

 aplay /usr/share/sounds/alsa/Front_Center.wav

to test audio ouput but I get a pause, then 

 : Signed 16 bit Little Endian, Rate 48000 Hz, Mono and then a response aplay:

 pcm_write:1939: write error: Input/output error

**Expected:**

``` I expected playback of the deafult .wav file to test audio ouput and then followup with mic test.

**Problem-relevant `config.yaml` entries:**

``` I didn't receive any errors from this file

`

renekliment, 2017-01-08:
Hm, your audio is messed up. Did you configure ALSA yourself?

Try setting the `input_device` to `usb` in `config.yaml`.

Also, if you don't have other stuff on the CHIP I suggest you do a clean install of the latest system. I believe they have an OS with kernel at least 4.4 for it.

enclarify, 2017-01-08:
Based on what you pasted I think your playback device should be `plughw:0` and your input device `plughw:1`. You can test your devices like this:

`aplay -f S16_LE -D plughw:0 /usr/share/sounds/alsa/Front_Center.wav`

and 

`arecord -f S16_LE -D plughw:1 /tmp/test.wav`

enclarify, 2017-01-08:
Additionally `src/scripts/inc/chip.sh` should likely be changed to:

`config_set 'output_device' 'plughw:0'`

datafaust, 2017-01-08:
@enclarify , it seems that the file, src/scripts/inc/chip.sh does not exist, which seems to point to messup @renekliment  is mentioning. Seems like I can't output any sound at all I continue to get the input writeout error. I'm going to try a fresh install, on another chip I have or a reboot of the software on this chip and see what happens.

datafaust, 2017-01-08:
So I tried all the suggestions and no luck, short of re flashing which I might try. Quick question before I do though. I've been using the GUI through VNC, so I decided to test the rca connection again (was working 2 days ago) on my tv under the suspicion that there might be an issue withoutput overall and I can't get an image out to my tv through the 3.5mm jack anymore. Do you guys know how this might be related to this issue? Is there a general messup with files that not allowing any output through the jack?

enclarify, 2017-01-08:
@datafaust Unfortunately, I don't know much about using the 3.5mm jack because I have an audio amplifier that's fed from the U14 header and operate it headless. One of my CHIPs is older and has not been flashed yet. I tested it prior to my first reply and it functions as expected. On all my CHIPs the buildin audio device is 0 and the USB mic that I am using is 1. The NTC forums are great for specific CHIP info. You might have more luck with your question there

renekliment, 2017-01-08:
@datafaust Damn, don't delete comments. 
Anyways, you should first get audio in your OS working (aplay, arecord). Closing for now as it doesn't concern AlexaPi. Feel free to come back and reopen / file a new issue when needed.

closed 2017-01-08 by renekliment