alexa-pi/AlexaPi #143: Not creating log file

jacobjiggler, 2017-01-11:
**Output of the [audio debugging script](https://github.com/alexa-pi/AlexaPi/wiki/Audio-setup-&-debugging#audio-debugging-script).**
http://www.alsa-project.org/db/?f=4cbd1bc88b77a0b58e7569a8f41b41d2dd87fff7

**Your OS (including version) where you are running AlexaPi:**
pi@raspberrypi:~ $ cat /etc/debian_version
8.0
 on raspberry pi model B 1

**Python release (`python2 --version`):**

python 2.7.9
```

**Description of problem:**
Hey guys,

I tried the normal install under the base user  pi but I'm still getting crashes. Alexa also hasn't generated a log file under /var/log/AlexaPi.log, or anything to go off of. The only error I can find is by running sudo systemctl status AlexaPi.service

Jan 10 22:16:34 raspberrypi pulseaudio[1326]: [pulseaudio] server-lookup.c: Unable to contact D-Bus: org.freedesktop.DBus.Error.NotSupported: Unable to autolaunch a dbus-daemon without a $DISPLAY for X11
Jan 10 22:16:34 raspberrypi pulseaudio[1326]: [pulseaudio] main.c: Unable to contact D-Bus: org.freedesktop.DBus.Error.NotSupported: Unable to autolaunch a dbus-daemon without a $DISPLAY for X11
Jan 10 22:16:34 raspberrypi pulseaudio[1326]: [pulseaudio] bluez4-util.c: org.bluez.Manager.GetProperties() failed: org.freedesktop.DBus.Error.UnknownMethod: Method "GetProperties" with signature "" on interface "org.bluez.Manager" doesn't exist

My first thought was that this was related to issue #38 , so I tried Disabling PA. Sure, enough the errors went away but at the same time my pi crashes and wont respond to alexa, yet I have no log file to check against. 

**Expected:**
A working alexa(amazon example runs just fine)

renekliment, 2017-01-11:
Hey.
You probably use systemd/journal, therefore you get the log via `journalctl` - see https://github.com/alexa-pi/AlexaPi/wiki/Debugging#logs
Provide proper logs, so your issue can be resolved.

jacobjiggler, 2017-01-11:
Woops sorry. 

Jan 10 22:53:21 raspberrypi pulseaudio[2232]: [pulseaudio] server-lookup.c: Unable to contact D-Bus: org.freedesktop.DBus.Error.NotSupported: Unable to autolaunch a dbus-daemon without a $DISPLAY for X11
Jan 10 22:53:21 raspberrypi pulseaudio[2232]: [pulseaudio] main.c: Unable to contact D-Bus: org.freedesktop.DBus.Error.NotSupported: Unable to autolaunch a dbus-daemon without a $DISPLAY for X11
Jan 10 22:53:21 raspberrypi pulseaudio[2232]: [pulseaudio] bluez4-util.c: org.bluez.Manager.GetProperties() failed: org.freedesktop.DBus.Error.UnknownMethod: Method "GetProperties" with signature "" on interface "org.bluez.Manager" doesn
Jan 10 22:53:25 raspberrypi python[2194]: [025f4b30] alsa audio output error: cannot estimate delay: Input/output error
Jan 10 22:53:27 raspberrypi python[2194]: [025f4b30] alsa audio output error: cannot estimate delay: Input/output error

renekliment, 2017-01-11:
So you have PulseAudio disabled? What exactly did you do?
You might want to try running PA in system-wide mode.

jacobjiggler, 2017-01-11:
I ran the exact commands you listed but have since reverted. I just switch to system-wide mode and rebooted. It fixed my issue with pulse audio, and responds to a "alexa" from very close, but does not responded to anything after. I do have desktop enabled though. Is it probable that this is causing issues with load on the pi? My logs and status are green across the board. I also had no issue with the java gui client except for the annoying web authentication. 
 sudo journalctl -u AlexaPi.service
-- Logs begin at Tue 2017-01-10 23:00:47 PST, end at Tue 2017-01-10 23:08:22 PST
Jan 10 23:01:01 raspberrypi systemd[1]: Starting Alexa client for all your devic
Jan 10 23:01:01 raspberrypi systemd[1]: Started Alexa client for all your device

renekliment, 2017-01-18:
Yes, there very probably is an issue with the load. The trigger word recognition can sometimes be tricky even on a more powerful machine, but on a RPi 1 it probably eats the whole CPU and if there is also a desktop environment, it can be an issue. You can always use a button on a Pi 1, that works fine.

What you can do:
- try a device that has a more powerful CPU ... anything else than RPi 1 really
- wait for _v1.4_ which will have snowboy support, which should eat less resources than pocketsphinx for trigger word recognition

Closing as the original _issue_ is resolved.

closed 2017-01-18 by renekliment