alexa-pi/AlexaPi #153: USB Microphone resource busy problem...

peshko, 2017-01-15:
I have seen multiple requests in that matter. I tried to update the config, etc. Still, I constantly get resource busy...

Traceback (most recent call last):
File "/opt/AlexaPi/src/main.py", line 587, in
loop()
File "/opt/AlexaPi/src/main.py", line 505, in loop
inp = alsaaudio.PCM(alsaaudio.PCM_CAPTURE, alsaaudio.PCM_NORMAL, config['sound']['input_device'])
alsaaudio.ALSAAudioError: Device or resource busy [plughw:CARD=Device,DEV=0]

It works for a little bit, but then I get this error.

The output of arecord -l

card 1: Device [USB PnP Sound Device], device 0: USB Audio [USB Audio]
Subdevices: 0/1
Subdevice #0: subdevice #0

Obviously, it tries to restart itself multiple times...Works for a bit and then kicks the error again...

If I stop the service I can record and hear my recordings.

Not sure why it happens.

Any help to solve this frustrating issue is appreciated

renekliment, 2017-01-18:
> It works for a little bit, ...

What exactly do you mean by that?

renekliment, 2017-02-06:
Do you have a desktop OS? https://github.com/alexa-pi/AlexaPi/wiki/Audio-setup-&-debugging#pulseaudio

renekliment, 2017-02-10:
Closing due to inactivity. Please re-open if needed.

closed 2017-02-10 by renekliment

peshko, 2017-02-10:
René, sorry I should have responded to this thread earlier. Yes, the solution to run PA in system-wide mode solved my problem. One difference to me was that I had to sudo adduser root. Pretty weird, but before I did that, I was getting some other errors...Now everything seems to be working great...