alexa-pi/AlexaPi #154: Question: Keyboard shortcut?

Silent-Hunter, 2017-01-19:
I am curious if this can be made to work with a keyboard shortcut instead of voice activation? I want to test it on my laptop.

Thanks!

renekliment, 2017-01-19:
Yes. Use the `desktop` device platform.

closed 2017-01-19 by renekliment

Silent-Hunter, 2017-02-10:
I would like to be able to use this on a debian based pocket computer, with a keyboard shortcut as the trigger. Is there any way to disable the hotword portion completely, and avoid needing pocketsphinx?

renekliment, 2017-02-11:
@Silent-Hunter Yes, in the upcoming version _v1.4_ which should be released soon the hotword pocketsphinx stuff is a module that can be easily disabled in the config. You can use the `dev` branch (https://github.com/alexa-pi/AlexaPi/wiki/Q&A-(FAQ)#how-to-try-the-dev-branch), if you don't wanna wait. Comment it out in `requirements.txt` so it doesn't get installed, if you care.

renekliment, 2017-02-11:
Running in the background + keyboard shortcut is currently not supported. You can however write a custom trigger in the new version.

Silent-Hunter, 2017-02-11:
Thank you! I'll wait, since the pocket computer doesn't come out for at least a few more months. But thank you!