alexa-pi/AlexaPi #157: Help! Blank Screen with MMM integration

lucallmon, 2017-01-22:
I don't know what I did wrong but I need some help. 

Firstly, I installed the Magic Mirror2, which worked fine.

Then, I installed AlexaPi from here: https://github.com/alexa-pi/AlexaPi/wiki/Installation

When I went through the install process and it asked which one to install, I chose magicmirror instead of the standard, default one. I added the module line from https://github.com/dgonano/MMM-AlexaPi to the config.json file. Now when I load MMM, it's just a blank screen. 

Also, the Alexa voice integration is not working properly. I got my token from the Amazon Dev site. 

Questions:
1. should I have not installed it that way?
2. if not, how do I uninstall the package and re-install the way from the MMM-AlexaPi?
3. Or should I just install the way of the MMM-AlexaPi? Will it overwrite the already installed version?

Thanks. 

Any thoughts will help.

lucallmon, 2017-01-22:
Ok. I fixed the blank screen by adding the MMM-AlexaPi module to the modules folder. Now I can't get alexa wake word to work.

dgonano, 2017-01-22:
Hmm try restarting alexapi if you haven't already. The wake word is independent of MM.

lucallmon, 2017-01-22:
How do you restart AlexaPi independent of MM?

dgonano, 2017-01-22:
Or restart the Pi. 
If you followed the install procedure for both you should be fine. If you still have an issue can you just list them and I might get you to post some log files

lucallmon, 2017-01-22:
It seems I can get AlexaPi to work on the initial boot of the mirror (set to auto load at startup) but it seems to freeze after a bit. When I CTRL-Q to go back to the main screen and "pm2 stop mm"j or just let it load back up on its own, it doesn't work after that. 

Also, the response is a bit slow and when she comes back with an answer, the audio seems to cut in and out? Power issue? I have a Pi3 model B with a Canakit case and supplied power cord. I'm not getting any power warnings in the OS. 

Could the power be cutting in and out because the microphone is too sensitive? I don't feel like that is the answer...
]Thoughts?

lucallmon, 2017-01-22:
Perhaps the playback_handler of vlc is what's causing it to be choppy since VLC is a very high demand app?

lucallmon, 2017-01-22:
@dgonano would downgrading my Raspbian to Jessie Lite improve this perhaps? Will the mirror with AlexaPi still work?

dgonano, 2017-01-24:
To restart alexapi run `sudo service AlexaPi restart`

Just note magic mirror and alexapi are seperate applications, pm2 won't affect alexapi. 

Is your internet connection slow? that currently causes issues with alexapi. 

lets try,
1. restarting alexapi
2. maybe something to do with the HDMI audio fix mentioned in the wiki

I'm not sure about downgrading, I think I'm running the latest but i'm away at the moment so can't check.

lucallmon, 2017-01-24:
Thanks for getting back to me. 

1. My internet connection is pretty fast actually. I'm getting 30-60mbps depending on the time of day.
2. restarting AlexaPi only had the same problems
   - when I restart it comes up with [Warning: Unit file of AlexaPi.service changed on disk, 'systemctl daemon-reload' recommended]
3. I'm not currently using HDMI audio. I'm using the 3.5mm audio jack. 

Also, I just overclocked my CPU to 1300MHz to see if that would work... and no luck. Still the same results. 

Perhaps I'll make a video showing what's going on. Do you think that will help?

lucallmon, 2017-01-24:
Video #1 - https://goo.gl/photos/KVb2RBvCK5YLk8GX7

Video #2 - https://goo.gl/photos/cgvE71U3QHAMEMyE8

I hope this helps.

dgonano, 2017-01-24:
I'm not really an audio expert because I never had any issues. 
@renekliment are you familiar with this issue or another person who may know?

maso27, 2017-01-24:
@lucallmon have you tried running AlexaPi in debug mode?

Kill MM and AlexaPi, then run `sudo /opt/AlexaPi/src/main.py -d` and events will be displayed in the terminal as they happen.  Perhaps you can learn something from what they say.

lucallmon, 2017-01-24:
@maso27 how do you kill AlexaPi? I can stop MM, but not sure how to actually stop AlexaPi. Is it `sudo service AlexaPi stop`?

maso27, 2017-01-24:
Yeah that should work.  I use `sudo systemctl stop AlexaPi` but I think it all comes out the same.

Or, if you'd rather use a big hammer, find the process with `ps ax grep AlexaPi`, then `sudo kill xxx` where _xxx_ is the process number.

lucallmon, 2017-01-24:
I keep getting this error:
`Sending MM Heatbeat
URLError:  [Errno 111] Connection refused
INFO: cmn_live.c(88): Update from < 32.83  6.36 -22.48 -7.00 -8.68 -4.72 -12.31  8.11  6.79 -3.15 -0.96  3.09 -0.61 >
INFO: cmn_live.c(105): Update to   < 32.63  6.95 -22.60 -6.89 -7.01 -4.07 -10.36  7.78  5.64 -2.70 -1.80  0.68 -1.31 >
`
thoughts?

maso27, 2017-01-24:
Sorry.  I don't know MM at all.

lucallmon, 2017-01-24:
Here's a different set:
`Tue Jan 24 11:45:40 2017 Magic Mirror HTTP Server - :5055
Checking Internet Connection...
Connection OK
Indicating Success
Calling URL: http://localhost:8080/alexapi?action=AVSSTATUS&status=success
URLError:  [Errno 111] Connection refused
Play_Audio Request for: /opt/AlexaPi/src/resources/hello.mp3
Player State: State.Opening
Player State: State.Playing
Indicate Start Playing
Calling URL: http://localhost:8080/alexapi?action=AVSSTATUS&status=playback
URLError:  [Errno 111] Connection refused
Player State: State.Ended
Indicate Stop Playing
Calling URL: http://localhost:8080/alexapi?action=AVSSTATUS&status=idle
Starting Magic Mirror platform HTTP Server
Starting Magic Mirror heartbeat with 5 second interval`

Do you think there's an error with the localhost server? Why would the connection be refused?  @maso27 regardless of MM, is there anything that looks strange or off? I can post the entirety of the debug log if it will help. Thoughts @dgonano ?

renekliment, 2017-01-24:
> ```
> Sending MM Heatbeat URLError: [Errno 111] Connection refused
> Calling URL: http://localhost:8080/alexapi?action=AVSSTATUS&status=success URLError: [Errno 111] Connection refused
> Calling URL: http://localhost:8080/alexapi?action=AVSSTATUS&status=playback URLError: [Errno 111] Connection refused 
> Calling URL: http://localhost:8080/alexapi?action=AVSSTATUS&status=idle
> ```

Obviously, MMM-AlexaPi isn't set-up properly or something. It looks like it's not running / running on another address or port / firewalled / ...

dgonano, 2017-01-24:
Yeah that's an issue between the two, I'll have a look when I get home later. 

That's seperate to the audio though, Alexa should run fine even with those issues

dgonano, 2017-01-24:
Check your magic mirror is running on the same device at port 8080

lucallmon, 2017-01-24:
Hhmmm. I'll go back and troubleshoot that.

lucallmon, 2017-01-24:
So, check out the video: https://goo.gl/photos/eLsDTau2UgmPcrkh7

In debug mode, Alexa works great when she's on port 5050. it looks like MMM is running on port 8080. I tried to switch her to 8080 on the dev site, but it just shut down and didn't do anything. I don't know how to change the port in AlexaPi to 8080 or change the port in  MMM to 5050. Thoughts @dgonano @maso27  @renekliment ?

maso27, 2017-01-24:
Check _/etc/opt/AlexaPi/config.yaml_, towards the bottom

lucallmon, 2017-01-24:
I have to edit that in the terminal right?

maso27, 2017-01-24:
You can use whatever works for you.  The only thing that could cause problems is the whole windows/linux/mac carriage_return/line_feed incompatibilities that crop up.

But if you're in the terminal, the easiest editor is probably _nano_.  So, type `sudo nano /etc/opt/AlexaPi/config.yaml` to get started.

lucallmon, 2017-01-24:
Changing the config files did not change anything. Actually it made AlexaPi crash after the first use in MMM.

lucallmon, 2017-01-24:
See video of everything working: https://goo.gl/photos/koXtZYN6jv4wLfcQA

So, Alexa Pi works great when incorporated in debug mode with an open terminal while MMM is running. Otherwise, if I just let it open at boot as originally configured, she continues to work the same... poorly. 

Any additional thoughts? Perhaps setting a script to open terminal to run Alexa in debug mode at the start/boot? I don't know how to do this, but would like to know if someone does.

maso27, 2017-01-24:
I'm expecting there's a significance in which user is starting AlexaPi.
The service is started by user _alexapi_, but when you start it from the terminal, with "sudo", you're starting it as _root_.

My bet is that there's some sort of incompatibility between users and groups for AlexaPi and Magic Mirror.

@dgonano, any thoughts?

@lucallmon this is a little heavy-handed, but you could try making the adjustments for "running as root" in the wiki and see if that helps.

lucallmon, 2017-01-24:
Hot damn! @maso27  you're a genius! It works brilliantly right from startup!

I followed the SystemD run as root wiki and rebooted. And BLAM! Perfect!

Thanks for all your help! From a total NOOB, I appreciate it.

dgonano, 2017-01-25:
Yeah the communication between the two is done over tcp so users won't be an issue for integrating the two. Only for AlexaPi as you found. 

Glad you got it working! 
Is it showing the correct stuff on the mirror now?

renekliment, 2017-02-10:
Okay, this is all weird. I see no reason why running it as root would help, but if it does ...

@lucallmon If your issues are resolved, please close this.

bamartos, 2017-02-26:
Hey guys, 

A bit late for the answer but concerning the error "URLError: [Errno 111] Connection refused" the solution for me was toto open my port 5055 in the router (portforwarding). 

Cheers

renekliment, 2017-10-14:
Closing due to inactivity.

closed 2017-10-14 by renekliment