alexa-pi/AlexaPi #158: """ImportError: No module named portend/yaml"""

Renard2323, 2017-01-23:
So im very very new and Im sorry to disturb you but I checked and could not find any solution to my trouble.

Raspberry PI 3/Raspbian/Python... 2.7? 

So I installed first the "official" Alexa from Amazon and it was running so I tried to get it headless without the script to runat every boot. Reddit brought me here. Im working from VNC/Putty

I got this message error using the script and following instruction

**Command /usr/bin/python -c "import setuptools, tokenize;__file__='/tmp/pip-build-VFSBGd/pyalsaaudio/setup.py';exec(compile(getattr(tokenize, 'open', open)(__file__).read().replace('\r\n', '\n'), __file__, 'exec'))" install --record /tmp/pip-UvcO7S-record/install-record.txt --single-version-externally-managed --compile failed with error code 1 in /tmp/pip-build-VFSBGd/pyalsaaudio
Storing debug log for failure in /root/.pip/pip.log**

Then I had a **ImportError: No module named 'yaml'**

I tried to get this yaml and typed 

 **sudo aptitude install python-yaml**

That sounded like a good idea... Then I re-ran the whole script using default choices.

Finally I get

**Traceback (most recent call last):
  File "./auth_web.py", line 9, in <module>
    import cherrypy
  File "/usr/local/lib/python2.7/dist-packages/cherrypy/__init__.py", line 83, in <module>
    from cherrypy import _cpserver
  File "/usr/local/lib/python2.7/dist-packages/cherrypy/_cpserver.py", line 8, in <module>
    from cherrypy.process.servers import ServerAdapter
  File "/usr/local/lib/python2.7/dist-packages/cherrypy/process/__init__.py", line 14, in <module>
    from cherrypy.process import plugins, servers  # noqa
  File "/usr/local/lib/python2.7/dist-packages/cherrypy/process/servers.py", line 126, in <module>
    import portend
ImportError: No module named portend**

`sudo systemctl status AlexaPi.service

Get me 

AlexaPi.service - Alexa client for all your devices
   Loaded: loaded (/usr/lib/systemd/system/AlexaPi.service; enabled)
  Drop-In: /etc/systemd/system/AlexaPi.service.d
           └─restart.conf
   Active: failed (Result: start-limit) since Mon 2017-01-23 09:47:59 +05; 39min ago
     Docs: https://github.com/alexa-pi/AlexaPi/wiki
 Main PID: 966 (code=exited, status=1/FAILURE)

Jan 23 09:47:58 raspberrypi systemd[1]: Unit AlexaPi.service entered failed ....
Jan 23 09:47:59 raspberrypi systemd[1]: AlexaPi.service holdoff time over, s....
Jan 23 09:47:59 raspberrypi systemd[1]: Stopping Alexa client for all your d....
Jan 23 09:47:59 raspberrypi systemd[1]: Starting Alexa client for all your d....
Jan 23 09:47:59 raspberrypi systemd[1]: AlexaPi.service start request repeat....
Jan 23 09:47:59 raspberrypi systemd[1]: Failed to start Alexa client for all....
Jan 23 09:47:59 raspberrypi systemd[1]: Unit AlexaPi.service entered failed ....
Hint: Some lines were ellipsized, use -l to show in full.

Help please?`

renekliment, 2017-01-23:
> Command /usr/bin/python -c "import setuptools, tokenize;file='/tmp/pip-build-VFSBGd/pyalsaaudio/setup.py';exec(compile(getattr(tokenize, 'open', open)(file).read().replace('\r\n', '\n'), file, [...]
Storing debug log for failure in /root/.pip/pip.log

For some reason, you installation failed. You can try to re-run it until it works. If you could share `/root/.pip/pip.log`, that might help.

Renard2323, 2017-01-24:
So I started a new installation on a vanilla brand new raspbian jesse. I guess something was interacting poorly between the official "alexa" and alexapi. maybe python version (seriously no idea)?

Installation seem to have been smoothly. 

**I have a brand new batch of issues now: No trigger word is working, sound is crappy (but I swear I could hear the beggining of a hello if I try to talk to the damn stuff for half an hour).**

 But when I do talk to my mic, it makes a weak hissing sound for 5-6 seconds...

renekliment, 2017-01-24:
Can you try `sudo apt-get install vlc`? Maybe the _lite_ vlc is causing problems.

maso27, 2017-01-24:
@Renard2323 how is your audio configured?

I know the pi has a mediocre sound card, and you can actually hear very light hissing on the 3.5mm jack as different processes run in the background.

So I'm wondering if your audio is going to a different device, but the slight noises you're hearing are the processor ramping up and dropping back.

Renard2323, 2017-01-24:
Thank you for helping me and sorry for the delay, I had a new batch of linissues, the fun never stop (see * for rant)

> sudo apt-get install vlc

seem to have been already here

Reading package lists... Done
Building dependency tree
Reading state information... Done
vlc is already the newest version.
0 upgraded, 0 newly installed, 0 to remove and 116 not upgraded.

> So I'm wondering if your audio is going to a different device, but the slight noises you're hearing are the processor ramping up and dropping back.

That was my understanding, I use a small bluetooth/jack speaker and it go once static cracking when powering up

**I included a sound sample, Alexa is gulping a couple syllabus, but it sound like the hour answer around 10'. Static are genuine**
[record20170124155426.zip](https://github.com/alexa-pi/AlexaPi/files/727949/record20170124155426.zip)

Uselles frustrated rant below
* I had a kernel panic (179,2) so I tried to correct it without starting from scratch. Only solution were Linux based and my two USB linux refused to boot (its snowing today, maybe they decided not to work). I installed a virtual machine and it wouldn't access sd card. Finally someone lended me a kali USB and the notorious rude-mannered thug OS (why would it get to change to keyboard layout in keyboard menu? Anarchy in the OS!) finally did his job (guess it liked the snow). Also: is there something linux actually reliable?

tonylandek, 2017-01-31:
Just an FYI, in case anyone gets to the end of the install, puts in all their credentials and gets: "ImportError: No module named portend" you need to install portend. sudo pip install portend. It was the last bug I had to fix before I got everything going!

Renard2323, 2017-01-31:
I had only this issue when installing on top of the "official" Alexa (the one without voice activation).

I have a bunch of new issues now

renekliment, 2017-01-31:
@Renard2323 Can you try the `dev` branch with the `sox` playback handler? Fresh config is needed. Thanks.

Renard2323, 2017-01-31:
Renekliment: With pleasure but you'll have to be more specific about the "how" part. i can start a new raspbian sd card but can't find the instructions for this branch.

renekliment, 2017-01-31:
@Renard2323 After the `git clone ...` do `git checkout dev` and then just normal install. After that is done, change `vlc` to `sox` in the config.

Renard2323, 2017-01-31:
> change vlc to sox in the config.

You mean `sudo apt-get install sox`?

renekliment, 2017-01-31:
No, I mean change the string `vlc` to `sox` in the configuration file - `/etc/opt/AlexaPi/config.yml`.

renekliment, 2017-02-10:
@Renard2323 Any progress on this? Have you tried the `sox` handler?

Renard2323, 2017-02-10:
I have tremendous issues with Raspbian itself these days (I think most of my cards are not suited).

I will tell you my progress when I will get a OS that survive 3 boots.

Renard2323, 2017-02-14:
So update.

I have a authentification error signing up on amazon.

We're sorry!
An error occurred when we tried to process your request. Rest assured, we're already working on the problem and expect to resolve it shortly.

renekliment, 2017-02-15:
@Renard2323 Are there any details? That looks like wrong URL on Amazon.

Renard2323, 2017-02-15:
Hello Renee Kliment

So I finally got my token from amazon. Problem (sound and so on) with main branch persist as expected

git checkout dev 
fatal: Not a git repository (or any of the parent directories): .git

I edited config.yaml
playback_handler: "sox"

Is that what you meant?

renekliment, 2017-02-15:
@Renard2323 No. See https://github.com/alexa-pi/AlexaPi/wiki/Q&A-(FAQ)#how-to-try-the-dev-branch ... do a fresh config and then set that to _sox_.

Renard2323, 2017-02-16:
Holy Molly!

It look like its working indeed!

renekliment, 2017-03-05:
Seems resolved. SoX is now in `master`, with the _v1.4_ release. Closing.

closed 2017-03-05 by renekliment