alexa-pi/AlexaPi #162: Auto start issue

Texas-Mark, 2017-01-29:
I am sort of new with the Raspberry Pi 3, so please bear with me.  Let me start at the end to alleviate some questions and possibilities. 

This was my third attempt. I started with a fresh image of the latest Raspian Pixel that I already had VLC installed on. I ran the setup.sh (install is in the recommended  /opt directory). I chose NOT to auto start. When I do the install this way everything works fine when I manually start main.py. So I know everything is working. That should eliminate 99% of the possible issues.

Now, this is what I originally did, which is exactly the same as above, only I chose the auto start option. I actually did this twice (starting with the exact same fresh image each time, and with the same results).  When I choose auto start, I have problems. It sort of works. By that I mean I get an occasional partial word response, but nothing else, often with a very long lag (like 15 seconds). I've never gotten a full word out of it. But I know it is answering my question. Like I ask what time it is and I can make out part of the word "time" or "pm". But 99% of the time it does nothing at all.  So it seems like it is at least partly working but is having some sort of issue due to auto start. If I stop the service and restart it manually, all works fine. 

So, what I would like to know is if there is any known issues with auto start on the latest Raspian.  And to help troubleshoot the issue without starting all over yet again, how can I enable auto start now that I have already chosen not to? I would like to do it in a way that I can easily undo it. 

I have read most of the threads here relating to auto start, but they all seem to deal with it not working at all.  Any help would be appreciated.

maso27, 2017-01-31:
There are two different things you may be reading about.  

First, there's the "start as a service/init.d" option, which to my knowledge has always worked quite well.  This is the actual auto-start that I think you're referring to.

Second, there's the "auto restart" that watches to make sure AlexaPi is running and if it isn't it force-starts it again.  This one does what it says on the tin, but is unnecessary in a functioning installation and only causes confusion in a broken one.  That's the one we're looking to remove from the base because it makes debug a nightmare.

As to your problem: you're installing a full desktop distro, which uses PulseAudio.  There've been a number of complications around PulseAudio causing unwanted behavior, and there are notes in the wiki about either disabling it or setting it to run in system-wide mode.

Simplest solution: start with Raspbian Lite instead of a full desktop if don't need the desktop environment.

Texas-Mark, 2017-01-31:
Yes, you are correct that I was talking about the "start as a service" option. 

Yes I am starting a  desktop and had read about issues with audio. However, it seems to go deeper than that.  As I noted, if I start it up manually after the desktop is running, it works perfectly. 

In any case, can you tell me how to manually set it to auto start so that I can see if it still does the same thing (rather than using the install script), and perhaps to make it easier to troubleshoot why it's not working when auto-started?

I also think I tried it at least one time with auto start enabled and with auto-restart disabled, and it did the same thing. So I don't think it is in some sort of loop. And as I noted if I manually stop the service (when it was set to auto start) and start it manually, it would work fine. I also used the interactive mode and there were no errors. 

FWIW, each time I did a reinstall to try again, I always started with a "working" image made prior to installing AlexaPi.

It's not the end of the world, as it only takes a second to start it up, just trying to figure out what is going wrong.

renekliment, 2017-01-31:
https://github.com/alexa-pi/AlexaPi/wiki/Audio-setup-&-debugging#pulseaudio

maso27, 2017-01-31:
@Texas-Mark 
When you install AlexaPi, select the "start as a service" option.
Then you can disable the auto-start with `sudo systemctl disable AlexaPi`.  
(If you ever want to re-enable the auto-start, use `sudo systemctl enable AlexaPi`)

Then it would be worth seeing if there's a difference between starting it as a service and starting it manually.

Service: `sudo systemctl start AlexaPi` (You can use `stop` and `restart` in there to control it further)
Manually: `sudo /opt/AlexaPi/src/main.py` (add `-d` if you want to watch the debug notes)

Correct me if I'm wrong @renekliment but I believe the service runs with user _alexapi_ but the manual method runs with user _root_.

There was someone else having issues with a Magic Mirror install the other day, which were fixed when he followed the "run as root" instructions from the wiki.

If the manual method works but the service method doesn't we may need to look closer into permissions issues.

Texas-Mark, 2017-01-31:
@maso27
I'm not sure how what you suggest would help. As I said, when auto start is enabled, if I stop the service and then start it back up manually, it works just fine.  How would what you suggest be any different than what I have already done?  I know auto start is not working for me, but a manual start always does.  

So what I was looking for was how to manually enter whatever is normally done to get it to auto start (now that I have it working by not enabling auto start with the setup script), and see if that works.  I guess I was just trying to see if the script is not playing nice with my setup. 

Maybe another bit of info. When auto start is enabled with the setup script, even if I stop the AlexaPi service, when I do a reboot or shutdown, it hangs for a LONG time before shutting down (like a minute or more).  If I do not enable auto start with the script, start AlexaPi and reboot or shutdown even when it is running, it shuts down normally (like a few seconds).

@renekliment 
I have already read that link, and I really don't think it's an audio issue. If it were, why does it work just fine when I manually start it?

maso27, 2017-01-31:
What I'm suggesting is a way for us to try to debug your problem.  
I'm pointing out that there is a difference between how the program is run when it's done as a service and when it's done as a user.  If you try that, we can rule some things out.

With "start as a service", there isn't a manual set of commands that we run at boot.  The program is registered as a service with _systemd_.  Which is kind of like putting the program in the "Start" menu of Windows vs. hunting for and double-clicking the .exe file.

That's where the `sudo systemctl disable AlexaPi` command comes in.  It takes it off of _systemd_'s list of things to run at boot, but keeps it registered properly.

Also, you don't have to go through the whole rigamarole of freshly imaging the SD card or anything.  You can get away with running _setup.sh_ again and choosing "start as a service" and it'll be put in there.  Since all the packages are already installed, It should run quite a bit faster too.

maso27, 2017-01-31:
As a follow-up to your "another bit of info" -- which is quite interesting, btw -- have you tried `sudo systemctl disable AlexaPi` before shutting down to see if it still takes a long time?

...and for something that will just bypass the _systemd_ altogether and avoid the issue, you could do this:

`sudo crontab -e`

enter this at the bottom:

`@reboot /opt/AlexaPi/src/main.py`

Texas-Mark, 2017-02-01:
Ok, give me a day or two to try what you suggest. I will make a clone of my working SD card and then rerun the script as you suggested on the clone, and try some of the other things mentioned. I just really don't want to mess up my working card since I added some other stuff after finally getting it to work. 

I don't mind having to manually start it, I was just wondering why it would not work for me. 

BTW, I did try to rerun the script to remove auto start after I had set it up for auto start (by selecting no at that prompt), but it did not change it. It still auto started.  That is one of the reasons  I had to start all over. So I am not sure if rerunning the script and saying no should have undone the auto start, but it didn't for me.

Texas-Mark, 2017-02-01:
Ok, here is what I did so far. I cloned my SD card and re-ran the script with auto start on and auto restart off.
Did not work. Note" when I say did not work, I mean "sort of works" like I described in my first post.

Ran the sudo systemctl disable AlexaPi command and rebooted.
Manually started with sudo /opt/AlexaPi/src/main.py
Works fine

Ran sudo systemctl enable AlexaPi to turn auto start back on and rebooted.
Did not work

Next I disabled auto start again, and added  @reboot /opt/AlexaPi/src/main.py to crontab
Works fine

Some more data. I have LEDs connected (green for Alexa talking and red for recording). This gives me a bit of a visual indication of what is going on without running in interactive mode (which always works anyway since it's a manual start). When auto start is on, sometimes the green LED comes on when she is supposed to say hello at boot and just stay on for like 30 seconds. Sometimes it never comes on at all. When it is "sort of working", the green LED only comes on about 1 out of 10 times that I make a request, And I have to be really close to the mic to get to it to respond at all. And I almost never get any audio. And any audio that I do get are just partial words. Sometimes the red record LED will light, but usually not. 

Now when I manually start, or use crontab, it responds 99% of the time, even if I am 10 feet away. 

So, I am fine with using crontab, but if there is anything else you want me to try, let me know.

maso27, 2017-02-01:
Hmmm.

First question: does it take a long time to shut down when you have AlexaPi as a service, but disabled?  Or is that just when it's enabled?

And if you don't mind, could you try this:

1. Comment out the _reboot..._ line in crontab by putting a # in front of it
2. `sudo systemctl enable AlexaPi`
3. Reboot
4. Exercise the faulty behavior, such that she responds with a partial word
5. Use `sudo journalctl -u AlexaPi` to display the log and scroll to the end  (Alternatively, `sudo journalctl -u AlexaPi | tail` will display the last screen's worth of log)  (you can exit out of it with "q")
6. Paste what happens after _Started Alexa client for all your devices._ here

I understand if you're happy enough using crontab, but I also thank you for pointing out this issue.  I'd like to determine if we can fix whatever quirk this is for everyone.

Texas-Mark, 2017-02-01:
Ok, here is what I got with the  sudo journalctl -u AlexaPi | tail  command.  I did not get any words, but I got the LEDs to light 

Jan 31 20:44:58 raspberrypi python[700]: Checking Internet Connection...
Jan 31 20:44:58 raspberrypi python[700]: Connection OK
Jan 31 20:44:58 raspberrypi python[700]: Traceback (most recent call last):
Jan 31 20:44:58 raspberrypi python[700]: File "/opt/AlexaPi/src/main.py", line 587, in <module>
Jan 31 20:44:58 raspberrypi python[700]: loop()
Jan 31 20:44:58 raspberrypi python[700]: File "/opt/AlexaPi/src/main.py", line 505, in loop
Jan 31 20:44:58 raspberrypi python[700]: inp = alsaaudio.PCM(alsaaudio.PCM_CAPTURE, alsaaudio.PCM_NORMAL, config['sound']['input_device'])
Jan 31 20:44:58 raspberrypi python[700]: alsaaudio.ALSAAudioError: Device or resource busy [plughw:1]
Jan 31 20:44:59 raspberrypi systemd[1]: AlexaPi.service: main process exited, code=exited, status=1/FAILURE
Jan 31 20:44:59 raspberrypi systemd[1]: Unit AlexaPi.service entered failed state.

BTW, I just read that other thread about the guy with Magic Mirror that was having a similar issue. I was going to try the "root" fix. But I will hold off on that for now. 

As for the hanging shutdown, oddly that does not seem to be happening anymore. However, I did not select the auto restart option.  I can try that later too and see if that issue comes back.

maso27, 2017-02-01:
Thank you for doing this.

I don't know what it means but it seems like this issue has come up more than once.  Looks like the microphone is "busy" from the perspective of user _alexapi_.

I'd say that the "run as root" solution and the crontab hack are equivalent in what they are doing.  It's not the best idea from a security standpoint to let a program have root access, so in either case be forewarned.

And I would not recommend auto restart.  It just opens the door for more headaches.

Texas-Mark, 2017-02-01:
Well, just for the heck of it, I tried running as root per the wiki instructions. I thought I had success because I got a nice "hello" at boot. However, now I can't get it to respond to anything. No LEDs, nothing.

The journalctl -u AlexaPi | tail command just gives me this (and I had to run it as root to get anything)

-- Logs begin at Wed 2017-02-01 14:05:40 CST, end at Wed 2017-02-01 14:08:45 CST. --
Feb 01 14:05:45 raspberrypi systemd[1]: Starting Alexa client for all your devices...
Feb 01 14:05:45 raspberrypi systemd[1]: Started Alexa client for all your devices.

Running in debug mode (as root) just seems to bomb out

Checking Internet Connection...
Connection OK
Play_Audio Request for: /opt/AlexaPi/src/resources/hello.mp3
Player State: State.Opening
Player State: State.Playing
Player State: State.Ended
Traceback (most recent call last):
  File "/opt/AlexaPi/src/main.py", line 587, in <module>
    loop()
  File "/opt/AlexaPi/src/main.py", line 505, in loop
    inp = alsaaudio.PCM(alsaaudio.PCM_CAPTURE, alsaaudio.PCM_NORMAL, config['sound']['input_device'])
alsaaudio.ALSAAudioError: Device or resource busy [plughw:1]
INFO: kws_search.c(448): TOTAL kws 0.00 CPU nan xRT
INFO: kws_search.c(451): TOTAL kws 0.00 wall nan xRT
root@raspberrypi:~# 

This is why I made a clone of my sd card  before doing any of this. :-)

So unless there is anything else anyone wants me to try, I am just going to live with manually starting or using chrontab.

maso27, 2017-02-01:
Yeah, I think that sounds like the best bet.  There are scattered reports of this issue but I think many folks gave up and moved on, rather than debugging it.

Thanks for your help.

hanssipi, 2017-02-09:
Hi, Just to let you know, i got the same problem here. Rpi B+ takes a long time 90 seconds to reboot.
Like Texas-Mark says it sort of works, i can hear broken words. After disabeling autostart and added the line to crontab it works fine, well... when i push the button. Not when i call ALEXA. It did before....

renekliment, 2017-02-09:
Can you guys try https://github.com/alexa-pi/AlexaPi/wiki/Running-as-root with the autostart instead of crontab? Does is work the same way? Thanks.

Texas-Mark, 2017-02-09:
I did try the running as root and had even worse luck. See my post about four replies up.

For my project, manually starting it has worked the best and is not a big deal as I have to run a few other things manually anyway.

hanssipi, 2017-02-10:
1. #@reboot /opt/AlexaPi/src/main.py
2. cp /opt/AlexaPi/src/scripts/unit-overrides/usergroup-root.conf /etc/systemd/system/AlexaPi.service.d/
3. sudo systemctl enable AlexaPi
4. reboot

@renekliment did what you asked and yes it works now! 
forgot to say, [https://www.raspberrypi.org/downloads/noobs/](url) is the software on the pi.

unfortunally the wake word does not work well. If i say alexa nothing happens, sometimes when it's noisy in the room, i suddenly hear "Yes". Enabled that in the config file. Is there a simple way to correct that?
Thanks!

renekliment, 2017-02-10:
@hanssipi Great! The systemd override is a better solution to a bad thing :smile: 

In the next version, the recognition threshold will be configurable, so that's one thing. We'll improve the recognition in the near future.

Texas-Mark, 2017-02-11:
Just as a FWIW, I was having sort of the same issue as hanssipi , where it worked running as root, but I could not get it to recognize anything I said unless I was right on top of the mic. When I ran it manually, it would work fine from 10 feet away. So it seems that somehow running with auto start or as root, somehow messes up the mic input. At least that is what appears to be happening.

mano1979, 2017-03-05:
I had this issue with the pre- 1.4 release and it whas fixed by setting your default audio devices in the alexapi config.yaml file.

Though after updating to the 1.4 release this does not work anymore. I tried every way to name my audio device but nothing works. I will try to remove the cronjob and start it as sudo in the rc.local file.

renekliment, 2017-03-05:
@mano1979 What exactly doesn't work? What do you get with the same device as in pre-1.4?

mano1979, 2017-03-05:
The automatic start of Alexapi. I too get the half word ..lo instead of hello. And also when i start alexapi manually everything works great.

cgioconda, 2017-03-05:
Same issue - when I run it manually, I have no issue with the wakeword getting a "Yes?" response, but autostart never works.

renekliment, 2017-03-05:
Those of you who haven't already, post the output of `journalctl -u AlexaPi.service`.

mano1979, 2017-03-05:
pi@raspberrypi:~ $ journalctl -u AlexaPi.service
No journal files were found.

renekliment, 2017-03-05:
@mano1979 Did you choose _systemd_ or the classic init scripts? What about `systemctl status AlexaPi.service`?

mano1979, 2017-03-05:
i chose systemd.

**This is the output:**

systemctl status AlexaPi.service
● AlexaPi.service - Alexa client for all your devices
   Loaded: loaded (/usr/lib/systemd/system/AlexaPi.service; enabled)
  Drop-In: /etc/systemd/system/AlexaPi.service.d
           └─restart.conf
   Active: inactive (dead) since Fri 2017-03-03 11:38:12 UTC; 2 days ago
     Docs: https://github.com/alexa-pi/AlexaPi/wiki
 Main PID: 535 (code=exited, status=0/SUCCESS)

renekliment, 2017-10-14:
Closing due to inactivity.

closed 2017-10-14 by renekliment