alexa-pi/AlexaPi #164: Everything Works...Except Can't Authorize on Amazon/5050

tonylandek, 2017-02-01:
Have held off for a few days now, but need some help.

For some reason, just about everything is working perfect. No audio issues. I get playback and recording. Logs say the service has started, internet is ok, have double and triple checked all my security/ids from AVS.

But when I run the setup, it will get to the end and run the authorization script, and I go http://my.ip.address:5050 and also try http://localhost:5050 and they don't connect to Amazon to authorize. In Chromium I get "This site can't be reached--my.ip.address:5050 refused to connect"

I decided to install firefox as well, because I recall with the official Alexa install that I did a few months ago, there was some issues with Chromium not allowing authorization, but same problem with firefox.

Kind of lost at this point, any help? 

Any chance my Google Wifi router could be causing any problems? Just throwing that out there...

Here is the log as well

Thanks guys

/opt/AlexaPi/src/main.py -d
INFO: pocketsphinx.c(152): Parsed model-specific feature parameters from /usr/local/lib/python2.7/dist-packages/pocketsphinx/model/en-us/feat.params
Current configuration:
[NAME]			[DEFLT]		[VALUE]
-agc			none		none
-agcthresh		2.0		2.000000e+00
-allphone				
-allphone_ci		no		no
-alpha			0.97		9.700000e-01
-ascale			20.0		2.000000e+01
-aw			1		1
-backtrace		no		no
-beam			1e-48		1.000000e-48
-bestpath		yes		yes
-bestpathlw		9.5		9.500000e+00
-ceplen			13		13
-cmn			live		batch
-cmninit		40,3,-1		41.00,-5.29,-0.12,5.09,2.48,-4.07,-1.37,-1.78,-5.08,-2.05,-6.45,-1.42,1.17
-compallsen		no		no
-debug					0
-dict					/usr/local/lib/python2.7/dist-packages/pocketsphinx/model/cmudict-en-us.dict
-dictcase		no		no
-dither			no		no
-doublebw		no		no
-ds			1		1
-fdict					
-feat			1s_c_d_dd	1s_c_d_dd
-featparams				
-fillprob		1e-8		1.000000e-08
-frate			100		100
-fsg					
-fsgusealtpron		yes		yes
-fsgusefiller		yes		yes
-fwdflat		yes		yes
-fwdflatbeam		1e-64		1.000000e-64
-fwdflatefwid		4		4
-fwdflatlw		8.5		8.500000e+00
-fwdflatsfwin		25		25
-fwdflatwbeam		7e-29		7.000000e-29
-fwdtree		yes		yes
-hmm					/usr/local/lib/python2.7/dist-packages/pocketsphinx/model/en-us
-input_endian		little		little
-jsgf					
-keyphrase				alexa
-kws					
-kws_delay		10		10
-kws_plp		1e-1		1.000000e-01
-kws_threshold		1		1.000000e-05
-latsize		5000		5000
-lda					
-ldadim			0		0
-lifter			0		22
-lm					
-lmctl					
-lmname					
-logbase		1.0001		1.000100e+00
-logfn					
-logspec		no		no
-lowerf			133.33334	1.300000e+02
-lpbeam			1e-40		1.000000e-40
-lponlybeam		7e-29		7.000000e-29
-lw			6.5		6.500000e+00
-maxhmmpf		30000		30000
-maxwpf			-1		-1
-mdef					
-mean					
-mfclogdir				
-min_endfr		0		0
-mixw					
-mixwfloor		0.0000001	1.000000e-07
-mllr					
-mmap			yes		yes
-ncep			13		13
-nfft			512		512
-nfilt			40		25
-nwpen			1.0		1.000000e+00
-pbeam			1e-48		1.000000e-48
-pip			1.0		1.000000e+00
-pl_beam		1e-10		1.000000e-10
-pl_pbeam		1e-10		1.000000e-10
-pl_pip			1.0		1.000000e+00
-pl_weight		3.0		3.000000e+00
-pl_window		5		5
-rawlogdir				
-remove_dc		no		no
-remove_noise		yes		yes
-remove_silence		yes		yes
-round_filters		yes		yes
-samprate		16000		1.600000e+04
-seed			-1		-1
-sendump				
-senlogdir				
-senmgau				
-silprob		0.005		5.000000e-03
-smoothspec		no		no
-svspec					0-12/13-25/26-38
-tmat					
-tmatfloor		0.0001		1.000000e-04
-topn			4		4
-topn_beam		0		0
-toprule				
-transform		legacy		dct
-unit_area		yes		yes
-upperf			6855.4976	6.800000e+03
-uw			1.0		1.000000e+00
-vad_postspeech		50		50
-vad_prespeech		20		20
-vad_startspeech	10		10
-vad_threshold		2.0		2.000000e+00
-var					
-varfloor		0.0001		1.000000e-04
-varnorm		no		no
-verbose		no		no
-warp_params				
-warp_type		inverse_linear	inverse_linear
-wbeam			7e-29		7.000000e-29
-wip			0.65		6.500000e-01
-wlen			0.025625	2.562500e-02

INFO: feat.c(715): Initializing feature stream to type: '1s_c_d_dd', ceplen=13, CMN='batch', VARNORM='no', AGC='none'
INFO: acmod.c(166): Using subvector specification 0-12/13-25/26-38
INFO: mdef.c(518): Reading model definition: /usr/local/lib/python2.7/dist-packages/pocketsphinx/model/en-us/mdef
INFO: mdef.c(531): Found byte-order mark BMDF, assuming this is a binary mdef file
INFO: bin_mdef.c(336): Reading binary model definition: /usr/local/lib/python2.7/dist-packages/pocketsphinx/model/en-us/mdef
INFO: bin_mdef.c(516): 42 CI-phone, 137053 CD-phone, 3 emitstate/phone, 126 CI-sen, 5126 Sen, 29324 Sen-Seq
INFO: tmat.c(149): Reading HMM transition probability matrices: /usr/local/lib/python2.7/dist-packages/pocketsphinx/model/en-us/transition_matrices
INFO: acmod.c(117): Attempting to use PTM computation module
INFO: ms_gauden.c(127): Reading mixture gaussian parameter: /usr/local/lib/python2.7/dist-packages/pocketsphinx/model/en-us/means
INFO: ms_gauden.c(242): 42 codebook, 3 feature, size: 
INFO: ms_gauden.c(244):  128x13
INFO: ms_gauden.c(244):  128x13
INFO: ms_gauden.c(244):  128x13
INFO: ms_gauden.c(127): Reading mixture gaussian parameter: /usr/local/lib/python2.7/dist-packages/pocketsphinx/model/en-us/variances
INFO: ms_gauden.c(242): 42 codebook, 3 feature, size: 
INFO: ms_gauden.c(244):  128x13
INFO: ms_gauden.c(244):  128x13
INFO: ms_gauden.c(244):  128x13
INFO: ms_gauden.c(304): 222 variance values floored
INFO: ptm_mgau.c(476): Loading senones from dump file /usr/local/lib/python2.7/dist-packages/pocketsphinx/model/en-us/sendump
INFO: ptm_mgau.c(500): BEGIN FILE FORMAT DESCRIPTION
INFO: ptm_mgau.c(563): Rows: 128, Columns: 5126
INFO: ptm_mgau.c(595): Using memory-mapped I/O for senones
INFO: ptm_mgau.c(838): Maximum top-N: 4
INFO: phone_loop_search.c(114): State beam -225 Phone exit beam -225 Insertion penalty 0
INFO: dict.c(320): Allocating 138824 * 20 bytes (2711 KiB) for word entries
INFO: dict.c(333): Reading main dictionary: /usr/local/lib/python2.7/dist-packages/pocketsphinx/model/cmudict-en-us.dict
INFO: dict.c(213): Dictionary size 134723, allocated 1016 KiB for strings, 1679 KiB for phones
INFO: dict.c(336): 134723 words read
INFO: dict.c(358): Reading filler dictionary: /usr/local/lib/python2.7/dist-packages/pocketsphinx/model/en-us/noisedict
INFO: dict.c(213): Dictionary size 134728, allocated 0 KiB for strings, 0 KiB for phones
INFO: dict.c(361): 5 words read
INFO: dict2pid.c(396): Building PID tables for dictionary
INFO: dict2pid.c(406): Allocating 42^3 * 2 bytes (144 KiB) for word-initial triphones
INFO: dict2pid.c(132): Allocated 21336 bytes (20 KiB) for word-final triphones
INFO: dict2pid.c(196): Allocated 21336 bytes (20 KiB) for single-phone word triphones
INFO: kws_search.c(406): KWS(beam: -1080, plp: -23, default threshold -113, delay 10)
Checking Internet Connection...
Connection OK
INFO: kws_search.c(448): TOTAL kws 0.00 CPU nan xRT
INFO: kws_search.c(451): TOTAL kws 0.00 wall nan xRT

renekliment, 2017-02-01:
Do you actually use the IP of the machine with AlexaPi or your own?

tonylandek, 2017-02-01:
The actual IP of my machine. http://192.168.XX.X:5050

lion6230i, 2017-02-02:
Do you add your ip to "Allowed Origins" ?
It can set in your amazon develop account.

tonylandek, 2017-02-02:
Yeah did that as well.

lion6230i, 2017-02-03:
Do you run "python auth_web.py" before you use web browser to connect "http://192.168.XX.X:5050" ?

tonylandek, 2017-02-03:
Yeah and it runs fine from what I can tell. Gives me the correct go to
http://912.168.XX.X:5050 and hit crtl+c to end script response.

lion6230i, 2017-02-03:
@tonylandek 
What is your python version? This project is base on python2.7 different version may
have some compatible problem.

bamartos, 2017-02-06:
I have the same issue, I use python 2.7.6.

tonylandek, 2017-02-06:
@lion6230i 
Same as @bamartos 2.7.6

lion6230i, 2017-02-06:
@tonylandek @bamartos 
Version is fine. 
Second you can check your CherryPy module.
In below link , you can find a simple example to check CherryPy is work fine or not.
[CherryPy-8.8.0](https://pypi.python.org/pypi/CherryPy/8.8.0)

bamartos, 2017-02-06:
I confirm that Cherrpy is working for me. I still have the issue to connect..

bamartos, 2017-02-06:
Well, I tried several tricks and it seems that: 
1) adding my real ip address (get it from www.whatismyip.com) as Allowed Origins and Allowed Return URLs,
2) portforwarding the router to point the 5050 port to raspberrypi
3) open web browser from another-pc and connect to http://interne.ip:5050 

seemed that worked! (well still is not working the AlexaPi 100% but this will be another issue.)

Cheers

renekliment, 2017-02-06:
That seems like and it doesn't work w/o public IP. Glad you got it working though :)

Basically, you need to use the same local IP/hostname for setup and Amazon's page.

tonylandek, 2017-02-11:
Yeah I still have not had any luck. Tried the port forwarding deal, but same issue. Feel like there should be an easier way.

tonylandek, 2017-02-11:
Just got it going. It was an issue with Cherrypy. Didn't have cheroot installed. Thanks for the help guys.

renekliment, 2017-02-11:
Glad you got it working! The upcoming version should stop the install in case some dependency fails to install, so it is clear what needs to be fixed. Closing.

closed 2017-02-11 by renekliment