alexa-pi/AlexaPi #165: AlexaPi + Pocketsphinx Error (ImportError: cannot import name get_model_path)

likeadoc, 2017-02-02:
Latest Raspbian Jessie on RPi3 + PulseAudio
Python 2.7.9

**Description of problem:**
```
PocketSphinx on its own works without problems - but AlexaPi produces an Import Error
```

**Traceback (if applicable):**
```
Traceback (most recent call last):
Feb 02 05:41:48 magicmirrorpie python[610]: File "/opt/AlexaPi/src/main.py", line 23, in <module>
Feb 02 05:41:48 magicmirrorpie python[610]: from pocketsphinx import get_model_path
Feb 02 05:41:48 magicmirrorpie python[610]: ImportError: cannot import name get_model_path

```

lion6230i, 2017-02-02:
You can use python command line try to manual import pocketsphinx first.

likeadoc, 2017-02-02:
I tried to manually (via pdb) execute  " import pocketsphinx"  and still get the same message ( ImportError: cannot import name get_model_path)

renekliment, 2017-02-02:
@likeadoc I think you might have an old version. Be sure to have https://pypi.python.org/pypi/pocketsphinx/0.1.3 installed.

likeadoc, 2017-02-02:
pocketsphinx version is 0.1.3 ( installed  with  AlexaPi setup.sh) - If i change get_model_name to a variable containing the path to the models i dont get an error anymore.

likeadoc, 2017-02-02:
But you are right. it seems to be a problem with pocketsphinx because the test snippets return the same error

likeadoc, 2017-02-02:
solved ! was a faulty pocketsphinx 0.1.3 install! after uninstall and install via pip it works like a charm!

closed 2017-02-02 by likeadoc