alexa-pi/AlexaPi #166: main.py is hang

lion6230i, 2017-02-02:
Hi All,
I run the main.py . But always stop before play hello.mp3 .

> INFO: dict.c(213): Dictionary size 134728, allocated 0 KiB for strings, 0 KiB for phones
INFO: dict.c(361): 5 words read
INFO: dict2pid.c(396): Building PID tables for dictionary
INFO: dict2pid.c(406): Allocating 42^3 * 2 bytes (144 KiB) for word-initial triphones
INFO: dict2pid.c(132): Allocated 21336 bytes (20 KiB) for word-final triphones
INFO: dict2pid.c(196): Allocated 21336 bytes (20 KiB) for single-phone word triphones
INFO: kws_search.c(406): KWS(beam: -1080, plp: -23, default threshold -113, delay 10)
Checking Internet Connection...
Connection OK
MemCached: MemCache: inet:127.0.0.1:11211: connect: [Errno 111] Connection refused.  Marking dead.
Play_Audio Request for: /opt/AlexaPi/src/resources/hello.mp3

I can't hear hello.mp3 . The program just stop in this line.

I have try to use python command line import vlc to play hello.mp3 by manual.
It's work! I can hear hello from my speaker. I think the problem is not from python-vlc .

> import vlc
i = vlc.Instance()
p = vlc.MediaPlayer()
p.set_media(i.media_new("/opt/AlexaPi/src/resources/hello.mp3"))
p.play()

Could anyone guide me how to debug? or know what's wrong in my platform.

lion6230i, 2017-02-02:
## I add debug message in vlchandler.py .

>                 player.play() 
                print("debug 1")                                    
                if item['offset']:                                                                                       
                        player.set_time(item['offset'])                                                      
                                                                   
                print("debug 2")                                  
                self.current_item_lock.wait()
                print("debug 3")                                         
                self.current_item_lock.clear()                                                               
                print("debug 4")                                                                               
                                                          
                self.event_manager.event_detach(vlc.EventType.MediaStateChanged)                  
                self.player.stop()                        
                self.media_player.stop()                                                                     
                                                                                                                             
                self.is_playing = False                                                                      
                print("debug 5")                                                
                self.play_lock.set()                                                  
                print("debug 6")     

## The main.py will not hang, but have another error message ..

> MemCached:
 MemCache: inet:127.0.0.1:11211: connect: [Errno 111] Connection refused.  Marking dead.
setup_complete
Play_Audio Request for: /opt/AlexaPi/src/resources/hello.mp3
debug 1
debug 2
Traceback (most recent call last):
  File "main.py", line 586, in <module>
    setup()
Exception AttributeError:  in  ignored
Error in sys.exitfunc:
Traceback (most recent call last):
  File "/usr/lib/python2.7/atexit.py", line 13, in _run_exitfuncs
    def _run_exitfuncs():
Segmentation fault

renekliment, 2017-02-02:
Weird, that memcached is not running on your system ... maybe the install failed for some reason?

Can you try the `dev` branch? https://github.com/alexa-pi/AlexaPi/wiki/Q&A-(FAQ)#how-to-try-the-dev-branch

lion6230i, 2017-02-03:
@renekliment 
I have tried `dev` branch.
Error message in below:

> INFO: kws_search.c(406): KWS(beam: -1080, plp: -23, default threshold -225, delay 10)
2017-02-03 03:30:56,378 DEBUG: Setting up playback handler: VlcHandler
Type "a" to trigger Alexa: 2017-02-03 03:31:02,629 DEBUG: Stopping audio play
[ 1810.817621] xhci-hcd xhci-hcd.0.auto: xHCI xhci_drop_endpoint called with disabled ep cea55240
[ 1810.825259] xhci-hcd xhci-hcd.0.auto: Trying to add endpoint 0x82 without dropping it.
[ 1810.833144] usb 1-1: Not enough bandwidth for altsetting 1
[ 1810.838574] 2:2:1: usb_set_interface failed (-22)
2017-02-03 03:31:02,664 DEBUG: Playing audio: /opt/AlexaPi/src/resources/hello.mp3
Exception in thread Thread-1:
Traceback (most recent call last):
  File "/usr/lib/python2.7/threading.py", line 810, in __bootstrap_inner
    self.run()
  File "/usr/lib/python2.7/threading.py", line 763, in run
    self.__target(*self.__args, **self.__kwargs)
  File "/opt/AlexaPi/src/alexapi/triggers/pocketsphinxtrigger.py", line 55, in thread
    inp = alsaaudio.PCM(alsaaudio.PCM_CAPTURE, alsaaudio.PCM_NORMAL, self._config['sound']['input_device'])
ALSAAudioError: Input/output error [plughw:0]
2017-02-03 03:31:02,676 DEBUG: Cleaning up playback handler

I have input and output device in card0.

> root@OpenWrt:/opt/AlexaPi/src# aplay -l
**** List of PLAYBACK Hardware Devices ****
card 0: Device [USB Audio Device], device 0: USB Audio [USB Audio]
  Subdevices: 1/1
  Subdevice #0: subdevice #0
root@OpenWrt:/opt/AlexaPi/src# arecord -l
**** List of CAPTURE Hardware Devices ****
card 0: Device [USB Audio Device], device 0: USB Audio [USB Audio]
  Subdevices: 1/1
  Subdevice #0: subdevice #0

It seems alsaaudio can't accept config format. 
I have also tried use "plughw:0,0" but it didn't work too.

maso27, 2017-02-03:
I don't have anything valuable to contribute, but I think it's worth pointing out that it appears you're running this in OpenWRT.

Maybe these problems are happening within the OpenWRT-specific code.

renekliment, 2017-02-06:
Ah, shoot, I completely missed that! There might indeed be a problem with OpenWRT, but I'm not sure.

The command to list the devices, not the sound card itself is `arecord -L` ... please post the output of that.

Also, please try on the latest `dev` branch with a new install / fresh config. We've got some enhancements there. At the very least, it will print valid devices, if you choose an invalid one.

lion6230i, 2017-02-07:
@renekliment 
The log that I provide before already use `dev` branch. Did you commit any code in those two days?
Here is `arecord -L` log.

> root@OpenWrt:/opt/AlexaPi/src# arecord -L
null
    Discard all samples (playback) or generate zero samples (capture)
default:CARD=Device
    USB Audio Device, USB Audio
    Default Audio Device
sysdefault:CARD=Device
    USB Audio Device, USB Audio
    Default Audio Device
front:CARD=Device,DEV=0
    USB Audio Device, USB Audio
    Front speakers
surround21:CARD=Device,DEV=0
    USB Audio Device, USB Audio
    2.1 Surround output to Front and Subwoofer speakers
surround40:CARD=Device,DEV=0
    USB Audio Device, USB Audio
    4.0 Surround output to Front and Rear speakers
surround41:CARD=Device,DEV=0
    USB Audio Device, USB Audio
    4.1 Surround output to Front, Rear and Subwoofer speakers
surround50:CARD=Device,DEV=0
    USB Audio Device, USB Audio
    5.0 Surround output to Front, Center and Rear speakers
surround51:CARD=Device,DEV=0
    USB Audio Device, USB Audio
    5.1 Surround output to Front, Center, Rear and Subwoofer speakers
surround71:CARD=Device,DEV=0
    USB Audio Device, USB Audio
    7.1 Surround output to Front, Center, Side, Rear and Woofer speakers
iec958:CARD=Device,DEV=0
    USB Audio Device, USB Audio
    IEC958 (S/PDIF) Digital Audio Output

renekliment, 2017-02-07:
Yes, there is some new code.

lion6230i, 2017-02-07:
Hi @renekliment 

After test last `dev` branch , I set input and output device correctly.
But problem still there, main.py just hang after print 

> Play_Audio Request for: /opt/AlexaPi/src/resources/hello.mp3

I modify vlchandler.py to trace which step stop 

>              player.play()                                                     
                print("debug1")                                                                         
                if item.offset:                                                                             
                       player.set_time(item.offset)                                                                                                                                                                
                print("debug2")                                                                         
                self.current_item_lock.wait()                                                                
                print("debug3")                                                                         
                self.current_item_lock.clear()                                                               
                print("debug4")                      
                                                                                                             
                self.event_manager.event_detach(vlc.EventType.MediaStateChanged)
                self.player.stop()                                                                           
                self.media_player.stop() 
                                           

It's stop in debug1 

> INFO: kws_search.c(406): KWS(beam: -1080, plp: -23, default threshold -225, delay 10)
2017-02-07 11:09:44,703 DEBUG: Setting up playback handler: VlcHandler
Type "a" to trigger Alexa: 2017-02-07 11:09:50,582 DEBUG: Stopping audio play
2017-02-07 11:09:50,630 DEBUG: Playing audio: /opt/AlexaPi/src/resources/hello.mp3
debug1
2017-02-07 11:09:50,639 DEBUG: Cleaning up playback handler
INFO: cmn_live.c(120): Update from < 41.00 -5.29 -0.12  5.09  2.48 -4.07 -1.37 -1.78 -5.08 -2.05 -6.45 -1.42  1.17 >
INFO: cmn_live.c(138): Update to   < 43.84  4.59 -3.66  5.62 -3.40 -5.28 -7.87  0.52  3.94 -4.82 -7.65 -8.99 -5.36 >
INFO: kws_search.c(656): kws 28.84 CPU 11.582 xRT
INFO: kws_search.c(658): kws 16.42 wall 6.594 xRT
2017-02-07 11:10:07,052 DEBUG: Stopping audio play

I can't hear any voice from speaker.
Is it cause by python threading module?

May be like maso27 said it's only happening with the OpenWRT-specific code.
I use PC platform everything is work fine.

renekliment, 2017-02-07:
> OpenWRT-specific code

The is no OpenWRT-specific code.

I wonder, how do you have VLC installed on OpenWRT? Use the `sox` playback handler on OpenWRT.

Not related, just curious:
1. How did you actually install it on OpenWRT? 
2. What device are you using?

lion6230i, 2017-02-07:
I write a vlc Makefile for openwrt format after compile. It will auto
installs to my platform(build a image).
Yeah ,I will try sox playback handler tomorrow.

1. I use openwrt SDK to build a new firmware write it to flash with tftp.
2. IPQ4019 Qualcomm develop board.

maso27, 2017-02-07:
> OpenWRT-specific code
Yeah, I didn't say that well.  I remembered people were talking about OpenWRT earlier, and that we've got separate build instructions for different systems (debian/arch), so I assumed there was an OpenWRT one that had been used.
Sorry for the confusion!

lion6230i, 2017-02-08:
Hi @renekliment 
I change to use sox playback handler is work fine.
My vlc version is 2.2.4 , Which version is your recommendation?

renekliment, 2017-02-10:
@lion6230i Awesome! I don't think we will (and can) support VLC on OpenWRT, so I can't really tell about the version. Please just use the SoX handler.

Also note, that TuneIn support for the SoX handler is now experimental (some things work, some don't), but that's what you get for using experimental stuff :-) We will try to improve OpenWRT support, but that's a little bit in the future. This is the best we can offer now.

If the issue is resolved then, please close this.

lion6230i, 2017-02-11:
OK, Thanks for your help.

closed 2017-02-11 by lion6230i