alexa-pi/AlexaPi #168: Can't configure new skill? Device offline?

chimpy, 2017-02-04:
**Your OS (including version) where you are running AlexaPi:**
Raspbian 4.4

**Your hardware platform and model you are running on:**
 RPi 2

**Python release (`python2 --version`):**
2.7.9

**Description of problem:**
alexa.amazon.co.uk shows the device as offline.
Installed Plex skill, and then linked it to Plex account on alexa.amazon.co.uk. Alexa still says I need to link/configure the skill... but I have.

**Expected:**
Plex skill should work?

Is this normal or should this work? Thanks a lot for any help and for this great project!

renekliment, 2017-02-06:
Reported as working at gitter -> closing.

> In the end it was actually because I was using a UK account. And the Alexa was set up on a US account or something
> Anyway created a new account, works now 

Device shows up as _offline_ because https://github.com/alexa-pi/AlexaPi/wiki/Q&A-(FAQ)#why-do-i-see-the-alexapi-device-offline-in-the-alexa-app--why-do-i-get-timers-and-alarms-are-not-currently-supported-on-this-device

closed 2017-02-06 by renekliment

ArcUk, 2017-06-24:
Sorry to bring up a closed question, but the above issue was reported as fixed due to some issue with UK/US accounts? I am having the same issue (skills not being available on Pi after setup alex.amazon.com), could you clarify what the issue actually was here and how you addressed it. I can't feasibly create a new Amazon account if that is the suggestion as my account is tied to Prime etc.

Thanks.

B-Hard, 2017-10-23:
I have that issue too... any hints would be great!

ttruong15, 2017-10-26:
Same here, i setup my skills and can see in the apps but it seem the AlexaPI on my raspberry pi can't seem to find my skills.  I have test the skills in the website and it works fine.  Is there any plan to fix this issue?

dvbit, 2018-04-24:
Same for me