alexa-pi/AlexaPi #170: Alexa lifecycle proposal

ynk, 2017-02-07:
Hi,

First of all, thanks for the amazing work. I kickstarted my project in seconds.
As I work with Alexa for quite some time now, I found myself struggling while dealing with the `indicate` callbacks provided in devices ; sure enough, I made my own implementation to fit the need of our project.

I'd like to propose to make a distinction in playback between before and after processing, not only by giving booleans as arguments, but with a real steps reflecting the lifecycle of a conversation with Alexa. The terminology I chose in my proposal is toward something more human-like in the lifecycle, which make sens to me as we deal with a VUi.

For now, and as far as i traced it, here's the current lifecycle :

```
~ setup ~

1. indicate_success
2. indicate_playback True
3. indicate_playback False
4. after_setup

~ user says Alexa ~

5. indicate_playback True
6. indicate_playback False

7. indicate_recording True

~ user queries ~

8. indicate_recording False

9. indicate_processing True
10. indicate_processing False

11a. indicate_playback True
12a. indicate_playback False

11b. ø (if null response)
11c. indidate_failure (if 204)
```

**I would propose the following :**

```
~ setup ~

1. indicate_setup_success
2. indicate_greeting_start
3. indicate_greeting_end

~ user says Alexa ~

5. indicate_awaking_start
6. indicate_awaking_end

7. indicate_listening_start

~ user queries ~

8. indicate_listening_end

9. indicate_processing_start
10. indicate_processing_end

11a. indicate_answering_start
12a. indicate_answering_end

11b. indicate_muting_start (if null response)
12b. indicate_muting_end

11c. indicate_crashing_start (if 204)
12c. indicate_crashing_end
```

This is totally open to discussion, and in case of approval I would do my best to fill a proper PR with documentation.

Thanks,
J.

renekliment, 2017-02-07:
Hey!
Thanks for the idea. From a brief look it makes sense. Have you thought about how this would work with #52?

Couple of observations:
- `indicate_crashing_*` - it doesn't really mean something is _crashing_
- `indicate_muting_*` - _muting_? this brings association with mic muting
- with `indicate_greeting_*` and `indicate_awaking_*` you lose the indication of playback for _hello_ and _yes_ if I'm correct ... I don't have an opinion on if this is okay or not

ynk, 2017-02-07:
I do agree _crashing_ might be the best verb to define "a dead end because of server failure".

You are right about _muting_, maybe then _idling_ would be better ?

I do understand your concern about the other terms ; maybe it's all about finding the best verbs ; as for "awaking", my team and I described this phase that way because the real amazon echo does not speak while turning awake, rather emit a beep sound.

I'd be glad to implement it the way of #52 if more people like it this way ; I honestly didn't look at it before, and proposed simple callback fashion by personal preference.

renekliment, 2017-02-08:
@ynk Any work on AlexaPi is appreciated, so if you like to, knock yourself out :smile:

ynk, 2017-02-09:
@renekliment thank you. I've done a lot of work during the 2 last days. I'll submit my PR tonight hopefully.

closed 2017-02-09 by ynk