alexa-pi/AlexaPi #172: 400 Bad Request on http://localhost:5050 and Local IP?

AR019, 2017-02-18:
Thank you for reporting an issue with **AlexaPi**.

Make sure you are running the latest version of **AlexaPi**, and look through [issues](https://github.com/alexa-pi/AlexaPi/issues?utf8=%E2%9C%93&q=is%3Aissue) before submitting.

Check out our [Wiki](https://github.com/alexa-pi/AlexaPi/wiki/), and make sure to look at [Debugging](https://github.com/alexa-pi/AlexaPi/wiki/Debugging/), and [Audio debugging](https://github.com/alexa-pi/AlexaPi/wiki/Audio-setup-&-debugging) in particular before filing an issue.

In order to help troubleshooting, be sure to include the following information:

**Output of the [audio debugging script](https://github.com/alexa-pi/AlexaPi/wiki/Audio-setup-&-debugging#audio-debugging-script).**

**Your OS (including version) where you are running AlexaPi:**
```
Ubuntu (MATE)
```

**Your hardware platform and model you are running on:**
```
Pine64+ 2GB
```

**Python release (`python2 --version`):**
```
Python 2.7.12
```

**Description of problem:**
```
Getting 400 Bad Request when attempting to connect to http://localhost:5050 or http://The_Local_IP:5050 from amazon.com. Page also states that URLs are not whitelisted for application even though they are.
```

**Expected:**
```
Amazon Authorization Page
```

**Problem-relevant `config.yaml` entries:**
```
N/A or (Believed to be) Not applicable for this issue.
```

**Steps to reproduce:**

1. cd /opt/AlexaPi/src/scripts && sudo ./setup.sh
2. Go through setup, then "Ready goto" is presented along with the URLs 
3. Try to connect to http://localhost:5050 or the Local IP with :5050

**Traceback (if applicable):**
```

```

**Additional info:**
```
URLs are whitelisted, ran setup with following setup options:
For OS: debian (Ubuntu MATE)
For system/platform option: other
```

frisco82, 2017-02-18:
Are you sure you added that url to your app in amazon developer panel?
https://github.com/alexa-pi/AlexaPi/wiki/Installation

You need to add localhost:5050 and also the local ip you use to connect from your pc.

renekliment, 2017-02-18:
Make sure you add the url exactly as you enter it in your browser. Note also you need both URLs -`Allowed Origins` and `Allowed Return URLs` with the `/code`.

People often think they do it right, but they don't.

Also, there could be a problem with failed install - similar issue lately. That will be resolved in 2 weeks by releasing a new version. If you can't make it work, just wait for the new version.

EmerickH, 2017-06-14:
@AR019 you don't replied so I suppose you solved your issue...
Please re-open the issue if you still have the problem.

Emerick

closed 2017-06-14 by EmerickH