alexa-pi/AlexaPi #178: cant i run Alexa on my NOOBS operativ system? [ No start up no nothing]

NoBreakz, 2017-02-25:
Ive followed the installation guide to the point, only i have NOOBS insted of rasbian. 
but in the instalation setup it says that i have my OS is debian...

apreciate it alot for all help, ive started with Raspberry pi 2 weks ago so im a total noob to this. altså i use  MagicMirror 2 by michmich.

renekliment, 2017-02-25:
NOOBS is not an OS. It's an operating system installer which by default installs Raspbian (Debian-based OS), so choose Debian.

I'm not sure what your question is exactly.

NoBreakz, 2017-02-25:
i cant seem to get it to start, ive done what the instalation guide told me
to do, but nothing happends after i reboot. I thought it might be the OS
that was wrong..

Kenno9000, 2017-02-25:
I recall having a similar difficulty when I tried to install APi after NOOBS to install the OS.  

Try with the Raspbian downloaded from the RPi site onto a 16GB Class 10 and then 

 

*     sudo apt-get update

*     sudo apt-get dist-upgrade

Worst case is you end up with the cleanest freshest Pi.  

After  that, if there are probs, I’m around all day.  Not the greatest geek, but I prevailed in getting this to werk.

NoBreakz, 2017-02-25:
Thanks!

Funny thing happend, when i disconected the webcamere ( i use a logitech
c270 as a mic.... i know.. ill get a propper mic on monday)  Alexa keeps
saying hello at random times, but i dont get a interface or anything like
that, arent i supposed to?

But anyway im reading up how to set up the webcam properly on the
raspberry. Ill let you know how it goes
But as far as i know yoy can only use a usb mic, right?
#noob

Kenno9000, 2017-02-25:
Well, actually, there is no display interface, it’s a voice interface only.  You say her name, ask her a question and she answers you, that’s the whole package.  The Amazon echo is referred to as speaker (as in speaker interface).  You can hook up a couple of LEDs to see what she is up to (Name recognition, searching) or just clip on a Sense Hat, if you  have one, it lights up rainbow.  

 

What happens when you say Alexa (note this does not work for everyone, a friend’s accent never made it trigger), but the TV did, a lot.  

If you can trigger “hello” you have 8 secs to ask a question, give a command, and then a few seconds of latency before the reply.  IF this is happening, it’s working. What’s missing for you?

 

Yes, USB mic is the only way to do a mic on a RPi.

NoBreakz, 2017-03-03:
UPDATE

so I got myself a propper mic, but the random "HELLO" are still there.
Ive tested the mic and it works in raspberry.

she even say "HELLO" in a diffrent tone, but i dunno, she doesnt respond to any interaction from me

maso27, 2017-03-03:
Alexa only says "Hello" when the program is first run.  She says "yes" when the trigger word has been detected.
The fact that you're getting random "Hellos" indicates that the program is getting re-started after a crash somehow.

The first time you installed this, was there an option for an automatic restart that you selected?  This feature has recently been removed due to creating confusing debug scenarios.

If you've chosen this option, I strongly suggest that you disable it.  There's an entry about it in the wiki, [here](https://github.com/alexa-pi/AlexaPi/wiki/Restart-on-crashes).

As for debug, there are a couple things you can look at:
1.) List activity of the program with `sudo journalctl -u AlexaPi`.  _Q_ will quit it.
2.) Start in debug mode:
        `sudo systemctl stop AlexaPi` to stop the current instance.
        `sudo /opt/AlexaPi/src/main.py -d` to start with debug messages.  _Ctrl-C_ will stop the program.

NoBreakz, 2017-03-03:
yes I put YES on the option to restart Alexa if crash ( something like that in the install) 

thnx ill look into it, I'll let you know what happens :)

NoBreakz, 2017-03-03:
ok, ive put `sudo journalctl -u AlexaPi`

`mars 03 19:24:42 raspberrypi systemd[1]: Starting Alexa client for all your devi
mars 03 19:24:42 raspberrypi systemd[1]: Started Alexa client for all your devic
mars 03 19:25:30 raspberrypi pulseaudio[1085]: [pulseaudio] udev-util.c: Failed 
mars 03 19:25:30 raspberrypi pulseaudio[1085]: [pulseaudio] server-lookup.c: Una
mars 03 19:25:30 raspberrypi pulseaudio[1085]: [pulseaudio] main.c: Unable to co
mars 03 19:25:30 raspberrypi pulseaudio[1085]: [pulseaudio] bluez4-util.c: org.b
mars 03 19:25:36 raspberrypi pulseaudio[1085]: [pulseaudio] udev-util.c: Failed 
mars 03 19:26:09 raspberrypi pulseaudio[1085]: [pulseaudio] udev-util.c: Failed 
mars 03 19:26:09 raspberrypi python[592]: ALSA lib pcm_hw.c:1667:(_snd_pcm_hw_op
mars 03 19:26:09 raspberrypi python[592]: Checking Internet Connection...
mars 03 19:26:09 raspberrypi python[592]: Connection OK
mars 03 19:26:09 raspberrypi python[592]: Traceback (most recent call last):
mars 03 19:26:09 raspberrypi python[592]: File "/opt/AlexaPi/src/main.py", line 
mars 03 19:26:09 raspberrypi python[592]: loop()
mars 03 19:26:09 raspberrypi python[592]: File "/opt/AlexaPi/src/main.py", line 
mars 03 19:26:09 raspberrypi python[592]: inp = alsaaudio.PCM(alsaaudio.PCM_CAPT
mars 03 19:26:09 raspberrypi python[592]: alsaaudio.ALSAAudioError: No such file
mars 03 19:26:09 raspberrypi systemd[1]: AlexaPi.service: main process exited, c
mars 03 19:26:10 raspberrypi systemd[1]: Unit AlexaPi.service entered failed sta
mars 03 19:26:11 raspberrypi systemd[1]: AlexaPi.service holdoff time over, sche
mars 03 19:26:11 raspberrypi systemd[1]: Stopping Alexa client for all your devi
mars 03 19:26:11 raspberrypi systemd[1]: Starting Alexa client for all your devi
lines 1-23...skipping...
-- Logs begin at fr. 2017-03-03 19:24:36 CET, end at fr. 2017-03-03 20:15:06 CET. --
mars 03 19:24:42 raspberrypi systemd[1]: Starting Alexa client for all your devices...
mars 03 19:24:42 raspberrypi systemd[1]: Started Alexa client for all your devices.
mars 03 19:25:30 raspberrypi pulseaudio[1085]: [pulseaudio] udev-util.c: Failed to get card object.
mars 03 19:25:30 raspberrypi pulseaudio[1085]: [pulseaudio] server-lookup.c: Unable to contact D-Bus: org.freedesktop.DBus.Er
mars 03 19:25:30 raspberrypi pulseaudio[1085]: [pulseaudio] main.c: Unable to contact D-Bus: org.freedesktop.DBus.Error.NotSu
mars 03 19:25:30 raspberrypi pulseaudio[1085]: [pulseaudio] bluez4-util.c: org.bluez.Manager.GetProperties() failed: org.free
mars 03 19:25:36 raspberrypi pulseaudio[1085]: [pulseaudio] udev-util.c: Failed to get card object.
mars 03 19:26:09 raspberrypi pulseaudio[1085]: [pulseaudio] udev-util.c: Failed to get card object.
mars 03 19:26:09 raspberrypi python[592]: ALSA lib pcm_hw.c:1667:(_snd_pcm_hw_open) Invalid value for card
mars 03 19:26:09 raspberrypi python[592]: Checking Internet Connection...
mars 03 19:26:09 raspberrypi python[592]: Connection OK
mars 03 19:26:09 raspberrypi python[592]: Traceback (most recent call last):
mars 03 19:26:09 raspberrypi python[592]: File "/opt/AlexaPi/src/main.py", line 587, in <module>
mars 03 19:26:09 raspberrypi python[592]: loop()
mars 03 19:26:09 raspberrypi python[592]: File "/opt/AlexaPi/src/main.py", line 505, in loop
mars 03 19:26:09 raspberrypi python[592]: inp = alsaaudio.PCM(alsaaudio.PCM_CAPTURE, alsaaudio.PCM_NORMAL, config['sound']['i
mars 03 19:26:09 raspberrypi python[592]: alsaaudio.ALSAAudioError: No such file or directory [plughw:1]
mars 03 19:26:09 raspberrypi systemd[1]: AlexaPi.service: main process exited, code=exited, status=1/FAILURE
mars 03 19:26:10 raspberrypi systemd[1]: Unit AlexaPi.service entered failed state.
mars 03 19:26:11 raspberrypi systemd[1]: AlexaPi.service holdoff time over, scheduling restart.
mars 03 19:26:11 raspberrypi systemd[1]: Stopping Alexa client for all your devices...
mars 03 19:26:11 raspberrypi systemd[1]: Starting Alexa client for all your devices...
mars 03 19:26:11 raspberrypi systemd[1]: Started Alexa client for all your devices.
mars 03 19:26:18 raspberrypi pulseaudio[1187]: [pulseaudio] udev-util.c: Failed to get card object.
mars 03 19:26:19 raspberrypi pulseaudio[1187]: [pulseaudio] server-lookup.c: Unable to contact D-Bus: org.freedesktop.DBus.Er
mars 03 19:26:19 raspberrypi pulseaudio[1187]: [pulseaudio] main.c: Unable to contact D-Bus: org.freedesktop.DBus.Error.NotSu
mars 03 19:26:19 raspberrypi pulseaudio[1187]: [pulseaudio] bluez4-util.c: org.bluez.Manager.GetProperties() failed: org.free
mars 03 19:26:23 raspberrypi python[1176]: ALSA lib pcm_hw.c:1667:(_snd_pcm_hw_open) Invalid value for card
mars 03 19:26:23 raspberrypi python[1176]: Checking Internet Connection...
mars 03 19:26:23 raspberrypi python[1176]: Connection OK
mars 03 19:26:23 raspberrypi python[1176]: Traceback (most recent call last):
mars 03 19:26:23 raspberrypi python[1176]: File "/opt/AlexaPi/src/main.py", line 587, in <module>
mars 03 19:26:23 raspberrypi python[1176]: loop()
mars 03 19:26:23 raspberrypi python[1176]: File "/opt/AlexaPi/src/main.py", line 505, in loop
mars 03 19:26:23 raspberrypi python[1176]: inp = alsaaudio.PCM(alsaaudio.PCM_CAPTURE, alsaaudio.PCM_NORMAL, config['sound']['
mars 03 19:26:23 raspberrypi python[1176]: alsaaudio.ALSAAudioError: No such file or directory [plughw:1]
mars 03 19:26:24 raspberrypi systemd[1]: AlexaPi.service: main process exited, code=exited, status=1/FAILURE
[... cut ...]
mars 03 19:29:02 raspberrypi python[1631]: inp = alsaaudio.PCM(alsaaudio.PCM_CAPTURE, alsaaudio.PCM_NORMAL, config['sound']['
mars 03 19:29:02 raspberrypi python[1631]: alsaaudio.ALSAAudioError: No such file or directory [plughw:1]
mars 03 19:29:02 raspberrypi systemd[1]: AlexaPi.service: main process exited, code=exited, status=1/FAILURE
mars 03 19:29:02 raspberrypi systemd[1]: Unit AlexaPi.service entered failed state.
mars 03 19:29:03 raspberrypi systemd[1]: AlexaPi.service holdoff time over, scheduling restart.
lines 1-116
`

maso27, 2017-03-03:
It looks like you don't have the proper device specified for your microphone in _/etc/opt/AlexaPi/config.yaml_.

Specifically, this error shows up in there:
`alsaaudio.ALSAAudioError: No such file or directory [plughw:1]`

Check [this page](https://github.com/alexa-pi/AlexaPi/wiki/Audio-setup-&-debugging) in the wiki for a next step.

NoBreakz, 2017-03-04:
thnx! but do i make changes in the config.yaml?
and if so, where/what?

NoBreakz, 2017-03-04:
`arecord -D yourdevice test.wav
ALSA lib pcm.c:2239:(snd_pcm_open_noupdate) Unknown PCM yourdevice
arecord: main:722: audio open error: Ingen slik fil eller filkatalog
`

i get this code when i try to test the recording, the speakers work fine

NoBreakz, 2017-03-04:
ps! the last thing is in Norwegian, translated its " No such file or file category"

NoBreakz, 2017-03-04:
*catalog

renekliment, 2017-03-04:
@NoBreakz Update to latest version and then we can debug more.

https://github.com/alexa-pi/AlexaPi/wiki/Q&A-(FAQ)#how-do-i-upgrade-alexapi

NoBreakz, 2017-03-04:
Ive updated the Pi, both as shown in te link you posted and in the pi
config ( apt-get update )

But it seems like the pi has trouble finding the mic.

Ive tried the aplay test.mav bit i only get errors..

Any idé what can be the problem?

renekliment, 2017-03-04:
@NoBreakz Now that you've updated, post the log.

NoBreakz, 2017-03-05:
here is the debugging log 

`sudo journalctl -u AlexaPi.service
`

`-- Logs begin at lø. 2017-03-04 21:53:49 CET, end at sø. 2017-03-05 12:21:11 CET
mars 04 21:53:55 raspberrypi systemd[1]: Starting Alexa client for all your devi
mars 04 21:53:55 raspberrypi systemd[1]: Started Alexa client for all your devic
mars 04 21:54:01 raspberrypi python[575]: Traceback (most recent call last):
mars 04 21:54:01 raspberrypi python[575]: File "/opt/AlexaPi/src/main.py", line 
mars 04 21:54:01 raspberrypi python[575]: import coloredlogs
mars 04 21:54:01 raspberrypi python[575]: ImportError: No module named coloredlo
mars 04 21:54:01 raspberrypi systemd[1]: AlexaPi.service: main process exited, c
mars 04 21:54:01 raspberrypi systemd[1]: Unit AlexaPi.service entered failed sta
lines 1-9/9 (END)...skipping...
-- Logs begin at lø. 2017-03-04 21:53:49 CET, end at sø. 2017-03-05 12:21:11 CET. --
mars 04 21:53:55 raspberrypi systemd[1]: Starting Alexa client for all your devices...
mars 04 21:53:55 raspberrypi systemd[1]: Started Alexa client for all your devices.
mars 04 21:54:01 raspberrypi python[575]: Traceback (most recent call last):
mars 04 21:54:01 raspberrypi python[575]: File "/opt/AlexaPi/src/main.py", line 19, in <module>
mars 04 21:54:01 raspberrypi python[575]: import coloredlogs
mars 04 21:54:01 raspberrypi python[575]: ImportError: No module named coloredlogs
mars 04 21:54:01 raspberrypi systemd[1]: AlexaPi.service: main process exited, code=exited, status=1/FAILURE
mars 04 21:54:01 raspberrypi systemd[1]: Unit AlexaPi.service entered failed state.
~
`

NoBreakz, 2017-03-05:
`mars 04 21:54:01 raspberrypi systemd[1]: AlexaPi.service: main process exited, code=exited, status=1/FAILURE
mars 04 21:54:01 raspberrypi systemd[1]: Unit AlexaPi.service entered failed state.
`

NoBreakz, 2017-03-05:
ok so ive read the wiki, on the debuggin its stands 

> 
If you hear the word hello multiple times and your AlexaPi doesn't work, it's because the service keeps crashing (for example due to a faulty config) and gets restarted because you chose this option in setup. Follow the instructions above to find and fix your problem.

but how to i configure my config.yaml file, ive tried, i even manage to change the file permission so that i can edit it, but im not sure wich part im gonan change, im i going to add the source for my mic? 

and thanl you so much everybody for the help so far!

renekliment, 2017-03-05:
@NoBreakz You have to run the setup script again after the upgrade.

NoBreakz, 2017-03-06:
what script :)?

NoBreakz, 2017-03-06:
ok, I deleted AlexaPi and reinstalled it, then i tried this

pi@raspberrypi:~ $ sudo systemctl stop AlexaPi
pi@raspberrypi:~ $ sudo /opt/AlexaPi/src/main.py -d
Traceback (most recent call last):
  File "/opt/AlexaPi/src/main.py", line 536, in <module>
    capture = alexapi.capture.Capture(config, tmp_path)
  File "/opt/AlexaPi/src/alexapi/capture.py", line 31, in __init__
    self.validate_config()
  File "/opt/AlexaPi/src/alexapi/capture.py", line 35, in validate_config
    input_devices = alsaaudio.pcms(alsaaudio.PCM_CAPTURE)
AttributeError: 'module' object has no attribute 'pcms'

NoBreakz, 2017-03-06:
what is the attributeError?

renekliment, 2017-03-06:
Alright, I don't know. I don't see any circular dependencies or anything.

@alexa-pi/team Anybody?

NoBreakz, 2017-03-06:
can the config.yaml file be the problem?

renekliment, 2017-03-06:
No.

NoBreakz, 2017-03-06:
but shouldnt I spesify the sound input (mic), because the in the audio device settings on my PI the default is set to "bcm2835" and not the mic

maso27, 2017-03-06:
@NoBreakz yes, you use _/etc/opt/AlexaPi/config.yaml_ to set it as needed for your specific hardware setup.  If you're using a Raspberry Pi with a USB Microphone/camera everything should just work after a fresh install.
__

When you did the reinstall, did you choose to create a new config.yaml or did you keep the old one?  I think your installation was sufficiently dated to where the old one might not apply.

As far as the reinstall is concerned, here's how I think we're expecting it was performed:

-`sudo rm -r /opt/AlexaPi` or something to delete the old directory
-`cd /opt`
-`sudo git clone https://github.com/alexa-pi/AlexaPi`
-`sudo AlexaPi/src/scripts/setup.sh`
-Select all defaults provided for the install.
-Create new config.yaml.  Enter all the details requested, as listed on your amazon developer account.
-Navigate to <ip_address:5050> in a browser and log in.

Does that sound about right for what you did?

NoBreakz, 2017-03-06:
hi maso, yup, thats what i did, i even did the full uninstall and did the steps all over again, except i kept the same alexa information tho, but the refresh token worked fine so i rebooted, but nothing happends, not even the "hello" from before, actually i didnt get an option this time to add it this time :P

renekliment, 2017-03-06:
This has nothing to do with the configuration. It looks like a bug in the python code (or pyalsaaudio). It #worksForMe though ... I'm not sure what could go wrong.

NoBreakz, 2017-03-06:
but i used the same config.template.yaml, change the name like it say `sudo mv config.template.yaml config.yaml`

then I put in the amazon info, and put input and output on "default"

NoBreakz, 2017-03-06:
ok, im trying thi sagain, only now im makeing a new device aswell,

when i try to get the refresh token i can only get it if i follow the localhost:5050 not the ipaddress

when i try to follow the ip address this is the message

We're sorry!
An error occurred when we tried to process your request. Rest assured, we're already working on the problem and expect to resolve it shortly.

Hide Details
Error Summary
400 Bad Request
The redirect URI you provided has not been whitelisted for your application. Please add your redirect URI in the 'Allowed Return URLs' section under 'Web Settings' for your Security Profile on Amazon Developer Portal.

im not sure if this got anything to do with the errors but its there anyway.
but i still got the refresh token

NoBreakz, 2017-03-08:
i think the problem can be in the .asound.gif file ( wich i currently dont have) so i need to create it to set the usb device as default. my best guess at the problem

NoBreakz, 2017-03-10:
ok, i earesed the whole SSD card and reinstalled it, now this comes up

`i@raspberrypi:~ $ sudo journalctl -u AlexaPi.service
-- Logs begin at Fri 2017-03-10 19:12:32 CET, end at Fri 2017-03-10 19:14:50 CET
Mar 10 19:12:38 raspberrypi systemd[1]: Starting Alexa client for all your devic
Mar 10 19:12:38 raspberrypi systemd[1]: Started Alexa client for all your device
Mar 10 19:12:49 raspberrypi pulseaudio[885]: [pulseaudio] server-lookup.c: Unabl
Mar 10 19:12:49 raspberrypi pulseaudio[885]: [pulseaudio] main.c: Unable to cont
Mar 10 19:12:49 raspberrypi pulseaudio[885]: [pulseaudio] bluez4-util.c: org.blu
Mar 10 19:13:10 raspberrypi python[580]: INFO: Checking Internet Connection ...
Mar 10 19:13:10 raspberrypi python[580]: INFO: Starting new HTTPS connection (1)
Mar 10 19:13:10 raspberrypi python[580]: INFO: Connection OK
Mar 10 19:13:10 raspberrypi python[580]: INFO: AVS token: Requesting a new one
Mar 10 19:13:10 raspberrypi python[580]: INFO: Starting new HTTPS connection (1)
Mar 10 19:13:11 raspberrypi python[580]: INFO: AVS token: Obtained successfully
Mar 10 19:13:41 raspberrypi pulseaudio[885]: [alsa-source-USB Audio] alsa-source
Mar 10 19:13:41 raspberrypi pulseaudio[885]: [alsa-source-USB Audio] alsa-source
Mar 10 19:13:41 raspberrypi pulseaudio[885]: [alsa-source-USB Audio] alsa-source
lines 1-15/15 (END)...skipping...
-- Logs begin at Fri 2017-03-10 19:12:32 CET, end at Fri 2017-03-10 19:14:50 CET. --
Mar 10 19:12:38 raspberrypi systemd[1]: Starting Alexa client for all your devices...
Mar 10 19:12:38 raspberrypi systemd[1]: Started Alexa client for all your devices.
Mar 10 19:12:49 raspberrypi pulseaudio[885]: [pulseaudio] server-lookup.c: Unable to contact D-Bus: org.freedesktop.DBus.Error.NotSupported: Unable to autolaunch a dbus-daemon without a $DISPLAY for X11
Mar 10 19:12:49 raspberrypi pulseaudio[885]: [pulseaudio] main.c: Unable to contact D-Bus: org.freedesktop.DBus.Error.NotSupported: Unable to autolaunch a dbus-daemon without a $DISPLAY for X11
Mar 10 19:12:49 raspberrypi pulseaudio[885]: [pulseaudio] bluez4-util.c: org.bluez.Manager.GetProperties() failed: org.freedesktop.DBus.Error.UnknownMethod: Method "GetProperties" with signature "" on interface "org.bluez.Mana
Mar 10 19:13:10 raspberrypi python[580]: INFO: Checking Internet Connection ...
Mar 10 19:13:10 raspberrypi python[580]: INFO: Starting new HTTPS connection (1): api.amazon.com
Mar 10 19:13:10 raspberrypi python[580]: INFO: Connection OK
Mar 10 19:13:10 raspberrypi python[580]: INFO: AVS token: Requesting a new one
Mar 10 19:13:10 raspberrypi python[580]: INFO: Starting new HTTPS connection (1): api.amazon.com
Mar 10 19:13:11 raspberrypi python[580]: INFO: AVS token: Obtained successfully
Mar 10 19:13:41 raspberrypi pulseaudio[885]: [alsa-source-USB Audio] alsa-source.c: ALSA woke us up to read new data from the device, but there was actually nothing to read!
Mar 10 19:13:41 raspberrypi pulseaudio[885]: [alsa-source-USB Audio] alsa-source.c: Most likely this is a bug in the ALSA driver 'snd_usb_audio'. Please report this issue to the ALSA developers.
Mar 10 19:13:41 raspberrypi pulseaudio[885]: [alsa-source-USB Audio] alsa-source.c: We were woken up with POLLIN set -- however a subsequent snd_pcm_avail() returned 0 or another value < min_avail.
`

NoBreakz, 2017-03-10:
raspberrypi pulseaudio[885]: [alsa-source-USB Audio] alsa-source.c: ALSA woke us up to read new data from the device, but there was actually nothing to read!
 
raspberrypi pulseaudio[885]: [alsa-source-USB Audio] alsa-source.c: Most likely this is a bug in the ALSA driver 'snd_usb_audio'. Please report this issue to the ALSA developers.

 raspberrypi pulseaudio[885]: [alsa-source-USB Audio] alsa-source.c: We were woken up with POLLIN set -- however a subsequent snd_pcm_avail() returned 0 or another value < min_avail.

NoBreakz, 2017-03-14:
@renekliment  @maso27  doesn any of this make sense? sorry for beeing so noob, but nothing i try seems to work

renekliment, 2017-10-14:
A lot has changed in AlexaPi regarding audio. Try the latest version.

Closing due to inactivity.

closed 2017-10-14 by renekliment