alexa-pi/AlexaPi #179: Voice Activity Detector Doesn't trigger, MAX RECORDING LENGTH always used

sammachin, 2017-02-27:
Putting this here as a note for others, you may find that recording doesn't end just after you finish speaking, instead AlexaPi always records 8 seconds of audio which is the`MAX_RECORDING_LENGTH` as set in main.py line 187.

The issue is related to the choice of input_device in config.py, certain devices for example `default:CARD=Device` whilst works for capturing audio and sending to alexa does not work with the VAD. This is because when the alsa device is setup it refuses to accept the `setperiodsize` setting instead using a value derrived from the bit rate and format but slightly larger than would be expected, eg for 20ms frames at 16Khz 16bit the expected size of a frame is 320 but this device insists on returning 341. 

In turn this means that the comparison of frame length with expected length at line 473 fails (341 vs 320) and the frame is not passed through the VAD, The code then falls back to the max length.

I've tried  a couple of fixes including forcing the frame into VAD by disabling the comparison but the VAD then fails to parse the frame, also stripping the first 21 bytes from the frame before passing it to VAD which whilst that works felt very hacky and I wouldn't be confident in that working well on all platforms.

I finally found the solution is to use a different input device from `arecord -L` in my case the last one in the list `plughw:CARD=Device,DEV=0` worked well.

I have only been able to test this on the RasPi both Pi2 and Pi3, I have tried several USB devices, 2 microphones,a USB Sound Card and a USB Plantronics headset all with the same issues. I'm not sure if this is some quirk of alsa on the Pi or if other platforms exhibit the same behaviour.

I don't think this is a bug in the code that we can fix as such, rather we need to improve the selection and setup process for input_devices.

I am working on a script that will help guide the user through finding the right value in their mic, initially I propose we include this as a stand alone script that can be used for fault finding and then target adding it to the setup script for a future release once we have more feedback on how well it finds the right device.

renekliment, 2017-02-27:
@sammachin Dope. Please also note, that we'll be dropping pyalsaaudio in favor of pyaudio or sounddevice (#45), still not entirely clear which one, so it might be a good idea not to waste time with pyalsaaudio specifically.

sammachin, 2017-02-27:
Ok I'll keep working on it as its nearly there anyway, right now gives you a suggestion of 4-5 devices to use which at least pass the test above, I'm just adding a short record/playback test to the code so you can actually validate each mic properly.
I think it will be a useful line of work to have even once we switch audio framework as we can then port this code over and aim to still have a mic test/setup section to the setup guide. Seems to be a source of a lot of trouble for people and I don't think all the issues are alsa related, likely to be other things even as simple as not connecting the microphone properly!

renekliment, 2017-02-27:
Yes, you are right - a lot of people are struggling with the audio setup and some script would be very useful. I've filed #31, but no one had time to look at it.

Looking forward to it!

mano1979, 2017-03-01:
So i changed my config as you said and now it can start automatic without intervention (before i had to manually stop the service after boot and restart it).

Since this afternoon (after a reboot) everytime i ask alexa something, she goes into a startup loop and keeps saying "hello" in 10 sec. interval's. she only stops when i interrupt her with another question.
Then on the next question i ask she slips into the boot loop again.

I don't know yet is this is also the case when i manually start alexapi.

mano1979, 2017-03-01:
UPDATE : If i start it manually with the -d extension, i don't have this problem

renekliment, 2017-03-07:
@sammachin I've tried a shitty USB soundcard today. It:
1. Doesn't allow any other frame rate other than 48k (higher or lower) ... and it doesn't even say it via pyalsaaudio. Only when doing 

	```
	arecord -f S16_LE -r 16000 -D devicename -d 5 test16.wav
	arecord -f S16_LE -r 60000 -D devicename -d 5 test60.wav
	```
	I got 
	> Warning: rate is not accurate (requested = 16000Hz, got = 48000Hz)
	>         please, try the plug plugin 

    Therefore the voice trigger didn't work. Surprisingly, I think AVS coped with that (48kHz instead of 16kHz).
2. Got the same thing regarding length (341 instead of 480). When trying out with PyAudio, I got `IOError: [Errno -9981] Input overflowed`

The reason for devices `plughw:` is that obviously some (most?) audio cards don't support all the frame rates and the plug module in ALSA does the conversion. It's unfortunate that we don't list the `plughw:` stuff now and people have to know what to do and explicitly enable it.

renekliment, 2017-03-07:
So we have:
- frame rate issues (that can be solved by plughw in every case hopefully)
- invalid length / overflow issues - fun stuff - we could probably say "you got crap hardware, throw it away"
- rubbish data like you had when trying out the streaming before - I have the same when using pyaudio with this crap usb soundcard - on one specific device, not my laptop ... seriously, what the hell, I have absolutely no idea what went wrong here ... 

This whole audio business makes me wanna cry and say "use this setup, because it's hell to support anything else" :sob:

mkhansen-intel, 2017-07-24:
@sammachin - We're running into the same issue as described above where length = 341. We tried different input_device and output_device options but none of those work at all. You said you were working on a script? Is that the solution? We're using a Plantronics USB Headset as our input device.

EmerickH, 2017-07-25:
@mkhansen-intel in the dev branch of this repo, we have changed the Python audio lib, maybe it solves this issue?

Emerick

mkhansen-intel, 2017-07-27:
@EmerickH - we tried this and yes it seems to have fixed this issue. Now the question is will this be merged to the master branch any time soon?