alexa-pi/AlexaPi #182: Tracebacks from when I run the setup script.

samiultahir, 2017-03-01:
This is the error i receive when i run the setup script as told to in the installation guide.

I am in the right directory too.

pi@raspberrypi:/opt $ sudo ./AlexaPi/src/scripts/setup.sh
TIP: When there is a value in brackets like [default_value], hit Enter to use it.

Which operating system are you using?
         debian - Debian, Raspbian, Armbian, Ubuntu or other Debian-based
      archlinux - Arch Linux or Arch Linux-based
Your OS [debian]:
Which device are you using?
           chip - C.H.I.P.
       hyperion - Integrate with Hyperion Ambient Lightning Software
    magicmirror - Integrating with Magic Mirror project (MMM-AlexaPi)
       orangepi - Orange Pi or another H3 based board
          other - other SBCs, desktops, or anything else
    raspberrypi - all Raspberry Pi variants
         serial - an Arduino controled device for example; used for Teddy Ruxpin
Your device [raspberrypi]:
Do you want AlexaPi to run on boot?
You have these options:
0 - NO
1 - yes, use systemd (default, RECOMMENDED and awesome)
2 - yes, use a classic init script (for a very old PC or an embedded system)
Which option do you prefer? [1]: 1
Creating a user to run AlexaPi under ... done.
gpio:x:997:pi
audio:x:29:pi,pulse
Synchronizing state for AlexaPi.service with sysvinit using update-rc.d...
Executing /usr/sbin/update-rc.d AlexaPi defaults
Executing /usr/sbin/update-rc.d AlexaPi enable
Created symlink from /etc/systemd/system/default.target.wants/AlexaPi.service to /usr/lib/systemd/system/AlexaPi.service.
Would you like to also install Airplay support (Y/n)? n
Hit http://mirrordirector.raspbian.org jessie InRelease
Hit http://archive.raspberrypi.org jessie InRelease
Hit http://mirrordirector.raspbian.org jessie/main armhf Packages
Ign http://linux.teamviewer.com preview InRelease
Ign http://linux.teamviewer.com stable InRelease
Hit http://mirrordirector.raspbian.org jessie/contrib armhf Packages
Hit http://mirrordirector.raspbian.org jessie/non-free armhf Packages
Hit http://mirrordirector.raspbian.org jessie/rpi armhf Packages
Hit http://archive.raspberrypi.org jessie/main armhf Packages
Hit http://archive.raspberrypi.org jessie/ui armhf Packages
Hit http://linux.teamviewer.com preview Release.gpg
Hit http://linux.teamviewer.com stable Release.gpg
Hit http://linux.teamviewer.com stable Release
Hit http://linux.teamviewer.com preview Release
Hit http://linux.teamviewer.com stable/main armhf Packages
Hit http://linux.teamviewer.com preview/main armhf Packages
Ign http://mirrordirector.raspbian.org jessie/contrib Translation-en_GB
Ign http://mirrordirector.raspbian.org jessie/contrib Translation-en
Ign http://mirrordirector.raspbian.org jessie/main Translation-en_GB
Ign http://mirrordirector.raspbian.org jessie/main Translation-en
Ign http://linux.teamviewer.com stable/main Translation-en_GB
Ign http://mirrordirector.raspbian.org jessie/non-free Translation-en_GB
Ign http://mirrordirector.raspbian.org jessie/non-free Translation-en
Ign http://mirrordirector.raspbian.org jessie/rpi Translation-en_GB
Ign http://archive.raspberrypi.org jessie/main Translation-en_GB
Ign http://linux.teamviewer.com stable/main Translation-en
Ign http://mirrordirector.raspbian.org jessie/rpi Translation-en
Ign http://linux.teamviewer.com preview/main Translation-en
Ign http://archive.raspberrypi.org jessie/main Translation-en
Ign http://linux.teamviewer.com preview/main Translation-en_GB
Ign http://archive.raspberrypi.org jessie/ui Translation-en_GB
Ign http://archive.raspberrypi.org jessie/ui Translation-en
Reading package lists... Done
Reading package lists... Done
Building dependency tree
Reading state information... Done
curl is already the newest version.
git is already the newest version.
libpulse-dev is already the newest version.
libsox-fmt-mp3 is already the newest version.
python-dev is already the newest version.
python-pip is already the newest version.
python-setuptools is already the newest version.
sox is already the newest version.
swig is already the newest version.
vlc-nox is already the newest version.
libasound2-dev is already the newest version.
0 upgraded, 0 newly installed, 0 to remove and 148 not upgraded.
pip 1.5.6 from /usr/lib/python2.7/dist-packages (python 2.7)
Requirement already satisfied (use --upgrade to upgrade): requests>=2.4.3 in /usr/lib/python2.7/dist-packages (from -r ./requirements.txt (line 1))
Requirement already satisfied (use --upgrade to upgrade): CherryPy>=4.0.0 in /usr/local/lib/python2.7/dist-packages (from -r ./requirements.txt (line 2))
Downloading/unpacking python-vlc from git+https://github.com/oaubert/python-vlc.git (from -r ./requirements.txt (line 3))
  Cloning https://github.com/oaubert/python-vlc.git to /tmp/pip-build-ofBWBE/python-vlc
  Running setup.py (path:/tmp/pip-build-ofBWBE/python-vlc/setup.py) egg_info for package python-vlc
    WARNING:root:This script should be run from a VLC tree. Falling back to pre-generated file.

Downloading/unpacking pyalsaaudio>=0.8.2 (from -r ./requirements.txt (line 4))
  Downloading pyalsaaudio-0.8.4.tar.gz (315kB): 315kB downloaded
  Running setup.py (path:/tmp/pip-build-ofBWBE/pyalsaaudio/setup.py) egg_info for package pyalsaaudio

    warning: no files found matching '*.json' under directory 'doc'
Downloading/unpacking webrtcvad>=2.0.7 (from -r ./requirements.txt (line 5))
  Downloading webrtcvad-2.0.10.tar.gz (66kB): 66kB downloaded
  Running setup.py (path:/tmp/pip-build-ofBWBE/webrtcvad/setup.py) egg_info for package webrtcvad

Requirement already satisfied (use --upgrade to upgrade): pyyaml in /usr/local/lib/python2.7/dist-packages (from -r ./requirements.txt (line 6))
Downloading/unpacking pocketsphinx>=0.1.3 (from -r ./requirements.txt (line 7))
  Downloading pocketsphinx-0.1.3.zip (29.3MB): 5.0MB downloaded
Cleaning up...
Exception:
Traceback (most recent call last):
  File "/usr/lib/python2.7/dist-packages/pip/basecommand.py", line 122, in main
    status = self.run(options, args)
  File "/usr/lib/python2.7/dist-packages/pip/commands/install.py", line 290, in run
    requirement_set.prepare_files(finder, force_root_egg_info=self.bundle, bundle=self.bundle)
  File "/usr/lib/python2.7/dist-packages/pip/req.py", line 1198, in prepare_files
    do_download,
  File "/usr/lib/python2.7/dist-packages/pip/req.py", line 1376, in unpack_url
    self.session,
  File "/usr/lib/python2.7/dist-packages/pip/download.py", line 572, in unpack_http_url
    download_hash = _download_url(resp, link, temp_location)
  File "/usr/lib/python2.7/dist-packages/pip/download.py", line 433, in _download_url
    for chunk in resp_read(4096):
  File "/usr/lib/python2.7/dist-packages/pip/download.py", line 421, in resp_read
    chunk_size, decode_content=False):
  File "/usr/lib/python2.7/dist-packages/urllib3/response.py", line 256, in stream
    data = self.read(amt=amt, decode_content=decode_content)
  File "/usr/lib/python2.7/dist-packages/urllib3/response.py", line 201, in read
    raise ReadTimeoutError(self._pool, None, 'Read timed out.')
ReadTimeoutError: HTTPSConnectionPool(host='pypi.python.org', port=443): Read timed out.

Storing debug log for failure in /root/.pip/pip.log

samiultahir, 2017-03-01:
This is my pip.log file

------------------------------------------------------------
/usr/bin/pip2 run on Wed Mar  1 18:52:26 2017
Requirement already satisfied (use --upgrade to upgrade): requests>=2.4.3 in /u$
  skipping extra security
  skipping extra security
  skipping extra security
Requirement already satisfied (use --upgrade to upgrade): CherryPy>=4.0.0 in /u$
Downloading/unpacking python-vlc from git+https://github.com/oaubert/python-vlc$
  Cloning https://github.com/oaubert/python-vlc.git to /tmp/pip-build-ofBWBE/py$
  Found command 'git' at '/usr/bin/git'
  Running command /usr/bin/git clone -q https://github.com/oaubert/python-vlc.g$
  Running setup.py (path:/tmp/pip-build-ofBWBE/python-vlc/setup.py) egg_info fo$
    WARNING:root:This script should be run from a VLC tree. Falling back to pre$
    running egg_info
    creating pip-egg-info/python_vlc.egg-info
    writing pip-egg-info/python_vlc.egg-info/PKG-INFO
    writing top-level names to pip-egg-info/python_vlc.egg-info/top_level.txt
    writing dependency_links to pip-egg-info/python_vlc.egg-info/dependency_lin$
    writing manifest file 'pip-egg-info/python_vlc.egg-info/SOURCES.txt'

renekliment, 2017-03-01:
Hey! It looks like there was just some network error. Try again.

renekliment, 2017-03-04:
@samiultahir Have you tried again? Does it work?

samiultahir, 2017-03-04:
sorry not yet i've been busy lately trying to set up a nodemcu

samiultahir, 2017-03-04:
so far so good! fingers crossed!

samiultahir, 2017-03-04:
YAY! it worked but how do I actually run the script?
(The instructions are very brief!)

renekliment, 2017-03-04:
@samiultahir So now you have it installed? 
It gets installed as a system service - see https://github.com/alexa-pi/AlexaPi/wiki/Debugging

samiultahir, 2017-03-04:
ok thanks!

Yes it is installed.

samiultahir, 2017-03-04:
I started the service however Alexa won't respond... :(

I hear the 'Hello' at the start but alexa won't respond to my commands.
I have a button attached with pin 18 and GND.

Any help?

renekliment, 2017-03-04:
> Hint: Some lines were ellipsized, use -l to show in full.

@samiultahir Post the whole output.

samiultahir, 2017-03-05:
Here you go.

pi@raspberrypi:~ $ sudo systemctl status AlexaPi.service  -l
● AlexaPi.service - Alexa client for all your devices
   Loaded: loaded (/usr/lib/systemd/system/AlexaPi.service; enabled)
   Active: active (running) since Sat 2017-03-04 21:59:31 UTC; 13h ago
     Docs: https://github.com/alexa-pi/AlexaPi/wiki
 Main PID: 547 (python)
   CGroup: /system.slice/AlexaPi.service
           └─547 /usr/bin/python /opt/AlexaPi/src/main.py --daemon

Mar 04 21:59:31 raspberrypi systemd[1]: Started Alexa client for all your devices.
Mar 04 21:59:38 raspberrypi python[547]: Failed to create secure directory (/var/lib/AlexaPi/.config/pulse): Operation not permitted
Mar 04 21:59:38 raspberrypi python[547]: Failed to create secure directory (/var/lib/AlexaPi/.config/pulse): Operation not permitted
Mar 04 21:59:45 raspberrypi python[547]: Failed to create secure directory (/var/lib/AlexaPi/.config/pulse): Operation not permitted
Mar 04 21:59:45 raspberrypi python[547]: Failed to create secure directory (/var/lib/AlexaPi/.config/pulse): Operation not permitted
Mar 04 21:59:45 raspberrypi python[547]: INFO: Checking Internet Connection ...
Mar 04 21:59:45 raspberrypi python[547]: INFO: Starting new HTTPS connection (1): api.amazon.com
Mar 04 21:59:46 raspberrypi python[547]: INFO: Connection OK
Mar 04 21:59:46 raspberrypi python[547]: INFO: AVS token: Requesting a new one
Mar 04 21:59:46 raspberrypi python[547]: INFO: Starting new HTTPS connection (1): api.amazon.com
Mar 04 21:59:48 raspberrypi python[547]: INFO: AVS token: Obtained successfully
Mar 04 21:59:48 raspberrypi python[547]: Failed to create secure directory (/var/lib/AlexaPi/.config/pulse): Operation not permitted
Mar 04 21:59:48 raspberrypi python[547]: Failed to create secure directory (/var/lib/AlexaPi/.config/pulse): Operation not permitted
Mar 04 21:59:48 raspberrypi python[547]: Failed to create secure directory (/var/lib/AlexaPi/.config/pulse): Operation not permitted

samiultahir, 2017-03-07:
I had to reformat my SD card but now I get the tracebacks at the start again! 

PLZ Help!

renekliment, 2017-03-07:
The very same or something different?

samiultahir, 2017-03-08:
Yes i think so

samiultahir, 2017-03-08:
pi@raspberrypi:/opt/AlexaPi/src/scripts $ sudo ./setup.sh
TIP: When there is a value in brackets like [default_value], hit Enter to use it.

Which operating system are you using?
         debian - Debian, Raspbian, Armbian, Ubuntu or other Debian-based
      archlinux - Arch Linux or Arch Linux-based
Your OS [debian]: 
Which device are you using?
           chip - C.H.I.P.
       hyperion - Integrate with Hyperion Ambient Lightning Software
    magicmirror - Integrating with Magic Mirror project (MMM-AlexaPi)
       orangepi - Orange Pi or another H3 based board
          other - other SBCs, desktops, or anything else
    raspberrypi - all Raspberry Pi variants
         serial - an Arduino controled device for example; used for Teddy Ruxpin
Your device [raspberrypi]: 
Do you want AlexaPi to run on boot?
You have these options: 
0 - NO
1 - yes, use systemd (default, RECOMMENDED and awesome)
2 - yes, use a classic init script (for a very old PC or an embedded system)
Which option do you prefer? [1]: 
Creating a user to run AlexaPi under ... user already exists. That's cool - using that.
gpio:x:997:pi,alexapi
audio:x:29:pi,pulse,alexapi
Would you like to also install Airplay support (Y/n)? y
Get:1 http://ppa.launchpad.net trusty InRelease [15.5 kB]
Hit http://archive.raspberrypi.org jessie InRelease                        
Hit http://mirrordirector.raspbian.org jessie InRelease                    
Hit http://archive.raspberrypi.org jessie/main armhf Packages
Ign http://ppa.launchpad.net trusty InRelease        
Hit http://mirrordirector.raspbian.org jessie/main armhf Packages
Hit http://ppa.launchpad.net trusty/main Sources                               
Hit http://mirrordirector.raspbian.org jessie/contrib armhf Packages           
Hit http://ppa.launchpad.net trusty/main armhf Packages                        
Hit http://archive.raspberrypi.org jessie/ui armhf Packages                    
Hit http://mirrordirector.raspbian.org jessie/non-free armhf Packages          
Hit http://mirrordirector.raspbian.org jessie/rpi armhf Packages               
Hit http://ppa.launchpad.net trusty/main Translation-en                        
Ign http://ppa.launchpad.net trusty/main Translation-en_GB                     
Ign http://archive.raspberrypi.org jessie/main Translation-en_GB               
Ign http://archive.raspberrypi.org jessie/main Translation-en      
Ign http://mirrordirector.raspbian.org jessie/contrib Translation-en_GB        
Ign http://archive.raspberrypi.org jessie/ui Translation-en_GB                 
Ign http://mirrordirector.raspbian.org jessie/contrib Translation-en           
Ign http://mirrordirector.raspbian.org jessie/main Translation-en_GB           
Ign http://archive.raspberrypi.org jessie/ui Translation-en        
Ign http://mirrordirector.raspbian.org jessie/main Translation-en              
Ign http://mirrordirector.raspbian.org jessie/non-free Translation-en_GB       
Ign http://mirrordirector.raspbian.org jessie/non-free Translation-en          
Ign http://mirrordirector.raspbian.org jessie/rpi Translation-en_GB            
Ign http://mirrordirector.raspbian.org jessie/rpi Translation-en               
Fetched 15.5 kB in 20s (739 B/s)                                               
Reading package lists... Done
W: GPG error: http://ppa.launchpad.net trusty InRelease: The following signatures couldn't be verified because the public key is not available: NO_PUBKEY C2518248EEA14886
Reading package lists... Done
Building dependency tree       
Reading state information... Done
curl is already the newest version.
git is already the newest version.
libpulse-dev is already the newest version.
libsox-fmt-mp3 is already the newest version.
python-dev is already the newest version.
python-pip is already the newest version.
python-setuptools is already the newest version.
sox is already the newest version.
swig is already the newest version.
vlc-nox is already the newest version.
libasound2-dev is already the newest version.
The following packages were automatically installed and are no longer required:
[... 29 lines cut ...]
Cleaning up...
Exception:
Traceback (most recent call last):
  File "/usr/lib/python2.7/dist-packages/pip/basecommand.py", line 122, in main
    status = self.run(options, args)
  File "/usr/lib/python2.7/dist-packages/pip/commands/install.py", line 290, in run
    requirement_set.prepare_files(finder, force_root_egg_info=self.bundle, bundle=self.bundle)
  File "/usr/lib/python2.7/dist-packages/pip/req.py", line 1198, in prepare_files
    do_download,
  File "/usr/lib/python2.7/dist-packages/pip/req.py", line 1376, in unpack_url
    self.session,
  File "/usr/lib/python2.7/dist-packages/pip/download.py", line 572, in unpack_http_url
    download_hash = _download_url(resp, link, temp_location)
  File "/usr/lib/python2.7/dist-packages/pip/download.py", line 433, in _download_url
    for chunk in resp_read(4096):
  File "/usr/lib/python2.7/dist-packages/pip/download.py", line 421, in resp_read
    chunk_size, decode_content=False):
  File "/usr/lib/python2.7/dist-packages/urllib3/response.py", line 256, in stream
    data = self.read(amt=amt, decode_content=decode_content)
  File "/usr/lib/python2.7/dist-packages/urllib3/response.py", line 201, in read
    raise ReadTimeoutError(self._pool, None, 'Read timed out.')
ReadTimeoutError: HTTPSConnectionPool(host='pypi.python.org', port=443): Read timed out.

Storing debug log for failure in /root/.pip/pip.log
pi@raspberrypi:/opt/AlexaPi/src/scripts $

renekliment, 2017-03-19:
> `ReadTimeoutError: HTTPSConnectionPool(host='pypi.python.org', port=443): Read timed out.`

This still looks like network issue to me.

samiultahir, 2017-03-19:
oh

shivasiddharth, 2017-04-06:
Before trying to install Alexa, install python3-pip using 
sudo apt-get install python3-pip 
I ran into the same error as yours. installing python3-pip worked.

samiultahir, 2017-04-06:
I installed it using:
sudo apt-get install python3-pip
it said it was already installed but i'm running the setup script now, let's see if it makes a difference...

shivasiddharth, 2017-04-06:
Are you SSHing, through wifi?

samiultahir, 2017-04-06:
no. I'm on the pi itself

shivasiddharth, 2017-04-06:
Have the pi connected to internet through ethernet. Wifi has often caused such troubles.

samiultahir, 2017-04-06:
ok I have installed but managed to get some errors which say alexapi isn't running is this because i haven't set up pulse audio configuration?

shivasiddharth, 2017-04-06:
See the audio and debugging guide. For desktop OSes you need to install pulse audio. I would suggest running jessie lite if you can.

samiultahir, 2017-04-06:
It is running!
Now do i just add a button and to which GPIO pins?
OR
Can i just say 'Alexa'

shivasiddharth, 2017-04-06:
You can just say alexa

samiultahir, 2017-04-06:
i can't seem to hear alexa. Any reason why? my audio is set to the highest.

samiultahir, 2017-04-06:
Now there's no gui when i reboot!
it's just a black screen  with a cursor!

shivasiddharth, 2017-04-06:
You need to configure the audio. What kind of mic input are you using?

samiultahir, 2017-04-06:
i'm using a plantronics usb headset with mic

shivasiddharth, 2017-04-06:
in the terminal, run the following
`aplay -l`
get the device number of your headset. then run the following 
`sudo nano /etc/asound.conf`
paste the following code in that
```
pcm.!default  {
type hw card 1
}
ctl.!default {
type hw card 1
}
```
change "1"to match your device number. Reboot now.

I dont know whether i am allowed to share youtube urls here. Here is a guide that i posted in youtube. https://www.youtube.com/watch?v=WU2MLzqLKhk based on Jessie Lite. it is much easier and less buggy to run AlexaPi in Jessie Lite.

samiultahir, 2017-04-06:
I get this, which do i replace for '1' is it '0' or shall i just keep it as it is?

 aplay -l

**** List of PLAYBACK Hardware Devices ****
card 0: ALSA [bcm2835 ALSA], device 0: bcm2835 ALSA [bcm2835 ALSA]
  Subdevices: 8/8
  Subdevice #0: subdevice #0
  Subdevice #1: subdevice #1
  Subdevice #2: subdevice #2
  Subdevice #3: subdevice #3
  Subdevice #4: subdevice #4
  Subdevice #5: subdevice #5
  Subdevice #6: subdevice #6
  Subdevice #7: subdevice #7
card 0: ALSA [bcm2835 ALSA], device 1: bcm2835 ALSA [bcm2835 IEC958/HDMI]
  Subdevices: 1/1
  Subdevice #0: subdevice #0
card 1: DSP [Plantronics .Audio 400 DSP], device 0: USB Audio [USB Audio]
  Subdevices: 1/1
  Subdevice #0: subdevice #0

shivasiddharth, 2017-04-06:
leave it like that. Just paste the code
```
pcm.!default  {
type hw card 1
}
ctl.!default {
type hw card 1
}
```

samiultahir, 2017-04-06:
Now i get this...

pi@raspberrypi:~ $ sudo systemctl status AlexaPi.service
● AlexaPi.service - Alexa client for all your devices
   Loaded: loaded (/usr/lib/systemd/system/AlexaPi.service; enabled)
   Active: failed (Result: exit-code) since Thu 2017-04-06 13:11:21 UTC; 2s ago
     Docs: https://github.com/alexa-pi/AlexaPi/wiki
  Process: 1096 ExecStart=/usr/bin/python /opt/AlexaPi/src/main.py --daemon (code=exited, status=1/FAILURE)
 Main PID: 1096 (code=exited, status=1/FAILURE)

Apr 06 13:11:21 raspberrypi python[1096]: surround50:CARD=DSP,DEV=0
Apr 06 13:11:21 raspberrypi python[1096]: surround51:CARD=DSP,DEV=0
Apr 06 13:11:21 raspberrypi python[1096]: surround71:CARD=DSP,DEV=0
Apr 06 13:11:21 raspberrypi python[1096]: iec958:CARD=DSP,DEV=0
Apr 06 13:11:21 raspberrypi python[1096]: dmix:CARD=DSP,DEV=0
Apr 06 13:11:21 raspberrypi python[1096]: dsnoop:CARD=DSP,DEV=0
Apr 06 13:11:21 raspberrypi python[1096]: hw:CARD=DSP,DEV=0
Apr 06 13:11:21 raspberrypi python[1096]: plughw:CARD=DSP,DEV=0
Apr 06 13:11:21 raspberrypi systemd[1]: AlexaPi.service: main process exited...E
Apr 06 13:11:21 raspberrypi systemd[1]: Unit AlexaPi.service entered failed ....
Hint: Some lines were ellipsized, use -l to show in full.

samiultahir, 2017-04-06:
pi@raspberrypi:~ $ sudo systemctl status AlexaPi.service -l
● AlexaPi.service - Alexa client for all your devices
   Loaded: loaded (/usr/lib/systemd/system/AlexaPi.service; enabled)
   Active: failed (Result: exit-code) since Thu 2017-04-06 13:11:21 UTC; 55s ago
     Docs: https://github.com/alexa-pi/AlexaPi/wiki
  Process: 1096 ExecStart=/usr/bin/python /opt/AlexaPi/src/main.py --daemon (code=exited, status=1/FAILURE)
 Main PID: 1096 (code=exited, status=1/FAILURE)

Apr 06 13:11:21 raspberrypi python[1096]: surround50:CARD=DSP,DEV=0
Apr 06 13:11:21 raspberrypi python[1096]: surround51:CARD=DSP,DEV=0
Apr 06 13:11:21 raspberrypi python[1096]: surround71:CARD=DSP,DEV=0
Apr 06 13:11:21 raspberrypi python[1096]: iec958:CARD=DSP,DEV=0
Apr 06 13:11:21 raspberrypi python[1096]: dmix:CARD=DSP,DEV=0
Apr 06 13:11:21 raspberrypi python[1096]: dsnoop:CARD=DSP,DEV=0
Apr 06 13:11:21 raspberrypi python[1096]: hw:CARD=DSP,DEV=0
Apr 06 13:11:21 raspberrypi python[1096]: plughw:CARD=DSP,DEV=0
Apr 06 13:11:21 raspberrypi systemd[1]: AlexaPi.service: main process exited, code=exited, status=1/FAILURE
Apr 06 13:11:21 raspberrypi systemd[1]: Unit AlexaPi.service entered failed state.

shivasiddharth, 2017-04-06:
Did you edit your config file as mentioned in [https://github.com/alexa-pi/AlexaPi/wiki/Audio-setup-&-debugging](url)

samiultahir, 2017-04-06:
yes

shivasiddharth, 2017-04-06:
see as to what you get when you run 
`sudo nano /etc/modprobe.d/alsa-base.conf`

samiultahir, 2017-04-06:
I am getting responses!
Although they are quiet :(
How can i make them louder?

shivasiddharth, 2017-04-06:
open alsamixer increase the volume
`alsamixer`

samiultahir, 2017-04-06:
what do I press once in the terminal layout of alsamixer?

samiultahir, 2017-04-06:
i found out the problem!
The audio is on the loudest setting but it's coming out of the headphones!!!!!!
How can i make it come out of the speakers?

shivasiddharth, 2017-04-06:
where is the speaker connected and where is the headphone connected?

samiultahir, 2017-04-06:
the speakers are connected via the 3.5mm audio jack and the headset is connected to usb

shivasiddharth, 2017-04-06:
change the 1  to 0 in this code that i had earlier mentioned
```
pcm.!default  {
type hw card 1
}
ctl.!default {
type hw card 1
}
```

samiultahir, 2017-04-06:
Both 1s?

shivasiddharth, 2017-04-06:
please close this issue as i guess you have the alexa working now.

samiultahir, 2017-04-06:
You guessed wrong...
because alexa isn't fully working - I have audio issues!

shivasiddharth, 2017-04-06:
If that did not work, open raspi config,
`sudo raspi-config`
move to advanced options, audio and choose force audio through 3.5mm

samiultahir, 2017-04-06:
there was an error running option A4 Audio

This is what happens when i try to force 3.5mm

shivasiddharth, 2017-04-06:
open the asound.conf using
`sudo nano /etc/asound.conf`
delete the contents, save and reboot. Run the following
sudo su
modprobe snd_bcm2835
cp /etc/modules /etc/modules.orig
echo snd_bcm2835 >> /etc/modules
now choose force audio

samiultahir, 2017-04-06:
No Luck

shivasiddharth, 2017-04-06:
run the following speaker test codes and let me know as to what happens
```
speaker-test -c2 --test=wav -w /usr/share/sounds/alsa/Front_Center.wav
sudo apt-get install mpg123
mpg123 http://ice1.somafm.com/u80s-128-mp3
```
also run following and let me know the devices order
`aplay -l`

samiultahir, 2017-04-06:
the commands make the audio come out from the headset not the speakers.

aplay -l gives...

**** List of PLAYBACK Hardware Devices ****
card 0: ALSA [bcm2835 ALSA], device 0: bcm2835 ALSA [bcm2835 ALSA]
  Subdevices: 7/8
  Subdevice #0: subdevice #0
  Subdevice #1: subdevice #1
  Subdevice #2: subdevice #2
  Subdevice #3: subdevice #3
  Subdevice #4: subdevice #4
  Subdevice #5: subdevice #5
  Subdevice #6: subdevice #6
  Subdevice #7: subdevice #7
card 0: ALSA [bcm2835 ALSA], device 1: bcm2835 ALSA [bcm2835 IEC958/HDMI]
  Subdevices: 1/1
  Subdevice #0: subdevice #0
card 1: DSP [Plantronics .Audio 400 DSP], device 0: USB Audio [USB Audio]
  Subdevices: 1/1
  Subdevice #0: subdevice #0

shivasiddharth, 2017-04-06:
open alsamixer, press F6, change audio card to bcm2835

samiultahir, 2017-04-06:
When i unplug the headset and type:

speaker-test -c2 --test=wav -w /usr/share/sounds/alsa/Front_Center.wav

the audio came out from the speakers

samiultahir, 2017-04-06:
i tried that but still no luck

shivasiddharth, 2017-04-06:
I would suggest you to get a seperate mini USB Mic or normal USB Mic or a cheap CM108 USB Dac if you are planning to use the 3.5mm jack.

samiultahir, 2017-04-06:
i have got a usb sound card device which allows you to plug headphones and microphones in. Is that what you mean?

shivasiddharth, 2017-04-06:
Yes..

[2 more replies]

closed 2017-10-14 by renekliment