alexa-pi/AlexaPi #183: Calendar not working -> Error, no calendar associated to account

gemx, 2017-03-02:
Thank you for reporting an issue with **AlexaPi**.

Make sure you are running the latest version of **AlexaPi**, and look through [issues](https://github.com/alexa-pi/AlexaPi/issues?utf8=%E2%9C%93&q=is%3Aissue) before submitting.

Check out our [Wiki](https://github.com/alexa-pi/AlexaPi/wiki/), and make sure to look at [Debugging](https://github.com/alexa-pi/AlexaPi/wiki/Debugging/), and [Audio debugging](https://github.com/alexa-pi/AlexaPi/wiki/Audio-setup-&-debugging) in particular before filing an issue.

In order to help troubleshooting, be sure to include the following information:

**Output of the [audio debugging script](https://github.com/alexa-pi/AlexaPi/wiki/Audio-setup-&-debugging#audio-debugging-script).**

**Your OS (including version) where you are running AlexaPi:**
```
Linux raspberrypi 4.4.38-v7+ #938 SMP Thu Dec 15 15:22:21 GMT 2016 armv7l GNU/Linux
```

**Your hardware platform and model you are running on:**
```
raspberry pi 3
```

**Python release (`python2 --version`):**
```
Python 2.7.9
```

**Description of problem:**
```
When asking Alexa "what are my appointments for today" she answers "You have no calendar associated to your account...".
I connected a my outloook.com successfully in the alexa app and this works with an android app "reverb" and also with the alexa-avs-sample app, so everything should be configured ok
```

**Expected:**
```
Alexa tells me my appointments
```

**Problem-relevant `config.yaml` entries:**
```

```

**Steps to reproduce:**

1.  Ask Alexa "What are my appointments for today"
2. 
3. 

**Traceback (if applicable):**
```

```

**Additional info:**
```

```

maso27, 2017-03-02:
This is likely more an issue with Amazon than with AlexaPi.

But have you checked which account you're logged into?  You can ask "who am I" and she'll tell you which account you're in.  Then "switch accounts" will log her into a different account if you are linked to one.

This happens with my wife's account and mine, since hers is linked to Amazon Prime Music and mine has all my stuff.

gemx, 2017-03-02:
When asking "who am i" it says that this is the only device in my account. The name for the account is also correct

gemx, 2017-03-02:
Sorry, I meant Alexa answers: there is only 1 account for this device ;-)

maso27, 2017-03-03:
I'm wondering if this is an API 1 vs. API 2 issue.

AlexaPi is currently using API 1, and as a result some features are different.

(Registers as "offline" in alexa app, timers don't work, stuff like that.)

gemx, 2017-03-03:
Already thought about that. The API docs don't state anything about the calendar availability so it might be. On the other hand, Alarms will need a means for notifications will only the new api provides but the calendar just reads my appointments!? Very strange

gemx, 2017-03-07:
Since it seems API related we can close it

closed 2017-03-07 by gemx

B-Hard, 2017-10-23:
Are there any news? I would like to use the calendar function too, cheers!