alexa-pi/AlexaPi #184: Setup without Pulse audio using usb DAC

shivasiddharth, 2017-03-03:
basically this is not an issue. i was able to setup alexa using usb dac and without pulse audio easily, so posting it here hoping that someone might find it useful.

OS used: jessie-lite

After sudo apt-get upgrade
chaqnge the default audio device using 
sudo nano /etc/asound.conf

in that paste the following and save, make sure to change "1" acording to your usb dac number
        pcm.!default  {
 	type hw card 1
	}
	ctl.!default {
 	type hw card 1
	}
restart the system and complete the alexa installation as mentioned in the guide. Then, open the config file using
sudo nano /etc/opt/AlexaPi/config.yaml
change the following
     input_device: sysdefault:CARD=Device
     output_device: ""
restart and you should have crystal clear audio

renekliment, 2017-03-04:
@shivasiddharth Could you write this up in a pretty form and put in https://github.com/alexa-pi/AlexaPi/wiki/Audio-setup-&-debugging so we can close this issue? Thank you.

shivasiddharth, 2017-03-04:
I tried to edit that, but did not know as to how it must be done.

closed 2017-03-05 by shivasiddharth