alexa-pi/AlexaPi #185: No module named cheroot.server

frisco82, 2017-03-04:
Thank you for reporting an issue with **AlexaPi**.

Make sure you are running the latest version of **AlexaPi**, and look through [issues](https://github.com/alexa-pi/AlexaPi/issues?utf8=%E2%9C%93&q=is%3Aissue) before submitting.

Check out our [Wiki](https://github.com/alexa-pi/AlexaPi/wiki/), and make sure to look at [Debugging](https://github.com/alexa-pi/AlexaPi/wiki/Debugging/), and [Audio debugging](https://github.com/alexa-pi/AlexaPi/wiki/Audio-setup-&-debugging) in particular before filing an issue.

In order to help troubleshooting, be sure to include the following information:

**Output of the [audio debugging script](https://github.com/alexa-pi/AlexaPi/wiki/Audio-setup-&-debugging#audio-debugging-script).**

**Your OS (including version) where you are running AlexaPi:**
```
OSMC (Debian 8.7)
```

**Your hardware platform and model you are running on:**
```
Raspberry PI 2
```

**Python release (`python2 --version`):**
```
Python 2.7.9
```

**Description of problem:**
```
Setup did not finish, it halts after setting client info
```

**Expected:**
```
Setup completes correctly
```

**Problem-relevant `config.yaml` entries:**
```

```

**Steps to reproduce:**

1. 
2. 
3. 

**Traceback (if applicable):**
```
Traceback (most recent call last):
  File "./auth_web.py", line 8, in <module>
    import cherrypy
  File "/usr/local/lib/python2.7/dist-packages/cherrypy/__init__.py", line 76, in <module>
    from cherrypy import _cprequest
  File "/usr/local/lib/python2.7/dist-packages/cherrypy/_cprequest.py", line 10, in <module>
    from cherrypy import _cpreqbody, _cpconfig
  File "/usr/local/lib/python2.7/dist-packages/cherrypy/_cpreqbody.py", line 135, in <module>
    import cheroot.server
ImportError: No module named cheroot.server
```

**Additional info:**
```

```

renekliment, 2017-03-04:
Thank you for the report. It seems that CherryPy==4.0.0 doesn't have `cheroot` in requirements. I'll bump the requirements (probably in general) and I'll let you know to test. Thanks!

renekliment, 2017-03-04:
@frisco82 See #187. If you could test the PR, that would be great. Thank you!

renekliment, 2017-03-04:
Fixed in `dev`. I'll close this when it gets merged to `master`.

cgioconda, 2017-03-08:
Having the same issue as well - when do you anticipate dev will be merged to master?

renekliment, 2017-03-08:
@cgioconda Some time this week. Until Friday hopefully.

renekliment, 2017-03-10:
Merged into `master` in version 1.5. Closing.

closed 2017-03-10 by renekliment