alexa-pi/AlexaPi #189: SoX handler volume

renekliment, 2017-03-04:
Couple of things:

1. When I tell Alexa to _volume up_ I get

	```
	Traceback (most recent call last):
	File "/usr/lib/python3.6/threading.py", line 916, in _bootstrap_inner
		self.run()
	File "/usr/lib/python3.6/threading.py", line 864, in run
		self._target(*self._args, **self._kwargs)
	File "src/main.py", line 508, in trigger_process
		alexa_speech_recognizer()
	File "src/main.py", line 318, in alexa_speech_recognizer
		process_response(resp)
	File "src/main.py", line 445, in process_response
		volume = player.get_volume() + int(vol_token)
	File "src/main.py", line 158, in get_volume
		return self.pHandler.volume
	AttributeError: 'SoxHandler' object has no attribute 'volume'
	```

    because the attribute name is different.

2. Then

	```
	@staticmethod
	def __calculate_volume_gain(volume):
		'''
		The volume is a percentage in AlexaPi configs. With SoX, volume is specified as a gain value.
		To change the volume the gain is specified as a value greater than or less than 0.
		'''
		return (MAX_VOLUME_GAIN * volume / 100) - MAX_VOLUME_GAIN
	```

I'm not sure I understand it correctly ... if you give it max = 100 (%), it returns 0. So you go from `- MAX_VOLUME_GAIN` to `0` on a linear scale?

renekliment, 2017-03-04:
@enclarify Could you please have a look at this? You're the SoX guru and also the general volume design might need a change and I believe you are better at it in this case. Thank you.

enclarify, 2017-03-05:
@renekliment I see the access of the volume attribute that's causing the issue. The simple fix is to just maintain a volume attribute with the with the originally requested % volume that can be queried in the current way. I will also do some more testing of raising and lowering the volume with the alexa commands to ensure the volume management is intuitive. You are correct that the volume is scaled between -30dB and +30dB where a value of 100% equals 0dB

chriswnl, 2017-11-11:
Here's how I fixed this in my sox setup with everything running on Ubuntu 16.04 LTS. In `soxhandler.py`  in `__init__`:
`self.volume = None`

in `on_setup(self)`, change 
```python
if self.__config['sound']['default_volume']:
    self.on_set_volume(self.__config['sound']['default_volume'])
```
to
```python
 if self.__config['sound']['default_volume']:
    self.volume = self.__config['sound']['default_volume']
    self.on_set_volume(self.__config['sound']['default_volume'])
```

and then change `on_set_volume(self,voume)` to

```python
def on_set_volume(self, volume):
    self.volume_gain = self.__calculate_volume_gain(volume)
    self.volume = volume
```

I was also having issues playing the only radio streams I listen too - BBC. There's a warning about sox and streams in the wiki but as they're all available (in my case) as mp3 streams, I can change the `on_play` method has the sox_cmd line of `sox_cmd = ['sox', '-t', 'mp3', '-q', audio_file]`

Hope this helps someone.