alexa-pi/AlexaPi #191: Amazon Music no longer working?

cgioconda, 2017-03-05:
Is Amazon Music no longer approved for use with AlexaPi? I set up a new install on my RPi2, created a new device profile in the Amazon Dev console, and seemingly appear to be approved to use Amazon Music in my new security profile, yet AlexaPi still tells me "Amazon Music is not supported for this device".

On another website (Martian Watches) in their faq, they state: "– I have setup my music service, but got a response on my watch saying “Amazon music is not supported for this device”. Yes, at the time Amazon Alexa is only authorized to play music through Amazon Echo and Dot devices that are tethered by Bluetooth to a compatible music sound system."

cgioconda, 2017-03-05:
It appears I am still pending - closing this case.

closed 2017-03-05 by cgioconda

renekliment, 2017-03-05:
See #90 for details.