alexa-pi/AlexaPi #194: Make audio prompts cusomisable

sammachin, 2017-03-10:
Putting this down here while I think of it.

Within alexapi we have a couple of audio files that are used as prompts/indicators;

There is the 'hello' on startup
An (optional) 'Yes' when the wakeword is detected
And a 'shuttng down' that can be used when the app is shutdown.

We should really make these files configurable especially when we move to better support internationalisation (British and German) it would also be nice to be able to replace the voices with tones for some people.

I suggest we keep it simple by having 3 config vars for the above prompts and then just set the filenames in config.yaml probabbly keeping the current ones as defaults maybe rename the files slightly eg alexa_hello_us.mp3, alexa_yes_us.mp3 or something?

renekliment, 2017-03-10:
Duplicate of #122.

sammachin, 2017-03-10:
ha ha my bad! I did try searching for an open issue but I looked for `audio` `file` and `prompts` oops :)

closed 2017-03-10 by renekliment