alexa-pi/AlexaPi #197: The authorization doesn't work

infograafikko, 2017-03-10:
**Your OS (including version) where you are running AlexaPi:**
`Jessie Lite 4.4
`
**Your hardware platform and model you are running on:**
`Raspberry Zero W`

**Python release (`python2 --version`):**
`2.7.9
`

**Description of problem:**
`Problem with authorization. 
`
```
"Ready goto http://192.168.1.175:5050 or http://localhost:5050  to begin the auth process
(Press Ctrl-C to exit this script once authorization is complete)"
```

The authorization is not getting completed. I have triple checked that the information matches ones in Amazon Developer page. 

**Expected:**
`Authorization will be completed
`

**Steps to reproduce:**
```
1.  sudo ./AlexaPi/src/scripts/setup.sh
2. fill in the right information
```

renekliment, 2017-03-10:
So what do you see / happens when you enter the URL in your browser? We need more info there.

infograafikko, 2017-03-10:
The problem was solved. I didn't fully understand the steps. 

`sudo python auth_web.py` helped to get it to work.

closed 2017-03-10 by infograafikko