alexa-pi/AlexaPi #198: How to set sample rate?

derkalaender, 2017-03-11:
My RaspberryPi 2 Model B doesn't do anything when I say "Alexa". I figured out why: the recording I get when running `arecord -D hwplug:Device,0 test.wav` sounds horrible. I assume it's because of the low sample rate (when I'm doing `arecord -r 96000 -D hwplug:Device,0 test.wav` it sounds normal).

So how do you change the sample rate AlexaPi records with?

Btw: I'am using [this](https://www.amazon.de/gp/product/B01KLRBHGM) mic and when using it on my windows pc it works perfectly finde.

renekliment, 2017-03-11:
Hey!
1. AlexaPi uses the sample rate of 16 kHz and you can't currently change that. It uses the `plughw:` devices, where ALSA does sample rate conversion automatically, there should be no issue.
2. > My RaspberryPi 2 Model B doesn't do anything when I say "Alexa". 

    This doesn't have to be the case - voice triggering can be tricky in a lot of ways.
3. Try running in debug mode, triggering it with a button, keypress or something else and record something. Then, upload your `recording.wav`, which will be in `/tmp/AlexaPi-runtime-something`.

derkalaender, 2017-03-12:
OK. I seem to be able to trigger the hotword when I'm shouting Alexa, but the quality it records my voice with is still terrible. Here is the [recording.wav](https://drive.google.com/file/d/0ByK8wkmfMgK1dUtpRkYzMXM1NU0/view?usp=sharing)

delphiki, 2017-03-29:
I have the exact same problem (RPi 2, same mic). I managed to get it working properly once, but no success since.