alexa-pi/AlexaPi #199: AlexaPi server status shows Unable to contact D-Bus[pulseaudio error]

NoBreakz, 2017-03-15:
im running the AlexaPi on Raspberry pi3 version B and im using Noobs [debian]

ok, so ive had alot of problems, i found out that it was my Pi that was the problem.
im currently borrowing my friends Pi, who has the same as mine. anyhow,

after using the comand `sudo systemctl status AlexaPi.service -l`

this mess shows up

`● AlexaPi.service - Alexa client for all your devices
   Loaded: loaded (/usr/lib/systemd/system/AlexaPi.service; enabled)
   Active: active (running) since on. 2017-03-15 11:44:28 CET; 2min 37s ago
     Docs: https://github.com/alexa-pi/AlexaPi/wiki
 Main PID: 568 (python)
   CGroup: /system.slice/AlexaPi.service
           ├─568 /usr/bin/python /opt/AlexaPi/src/main.py --daemon
           └─857 /usr/bin/pulseaudio --start --log-target=syslog

mars 15 11:44:28 raspberrypi systemd[1]: Started Alexa client for all your devices.
mars 15 11:44:38 raspberrypi pulseaudio[857]: [pulseaudio] server-lookup.c: Unable to contact D-Bus: org.freedesktop.DBus.Error.NotSupported: Unable to autolaunch a dbus-daemon without a $DISPLAY for X11
mars 15 11:44:38 raspberrypi pulseaudio[857]: [pulseaudio] main.c: Unable to contact D-Bus: org.freedesktop.DBus.Error.NotSupported: Unable to autolaunch a dbus-daemon without a $DISPLAY for X11
mars 15 11:44:49 raspberrypi python[568]: INFO: Checking Internet Connection ...
mars 15 11:44:50 raspberrypi python[568]: INFO: Connection OK
mars 15 11:44:50 raspberrypi python[568]: INFO: AVS token: Requesting a new one
mars 15 11:44:51 raspberrypi python[568]: INFO: AVS token: Obtained successfully
`

and its expesially the ` [pulseaudio] server-lookup.c: Unable to contact D-Bus: org.freedesktop.DBus.Error.NotSupported: Unable to autolaunch a dbus-daemon without a $DISPLAY for X11
mars 15 11:44:38 raspberrypi pulseaudio[857]: [pulseaudio] main.c: Unable to contact D-Bus: org.freedesktop.DBus.Error.NotSupported: Unable to autolaunch a dbus-daemon without a $DISPLAY for X11` that worries me.

ive tried so much but i gotten rid of so many errors that to this point i feel so close.
ive reinstalled pulseaudio, that worked to a point and i got rid of errors.

but im kind of stuck here. what should i try next?
clear the ssd-card and try all over again ?

thanks in advance!

-Adrian

illperipherals, 2017-03-15:
Adrian,
Can you jump into gitter?

NoBreakz, 2017-03-15:
@illperipherals hi, how do i do that, ? im very new to this

illperipherals, 2017-03-15:
https://gitter.im/alexa-pi/Lobby

NoBreakz, 2017-03-15:
@illperipherals  ah nice, thnx :)

renekliment, 2017-03-15:
Closing as resolved on gitter.
The cause of this is _not_ running PulseAudio in system-wide mode.

closed 2017-03-15 by renekliment

jeubzzz, 2017-03-20:
I am so new to all this. Gitter doesn't work. But I'm having the same issue and it's been keeping me up at night. The alexapi can't access the tokens from the amazon developers website

renekliment, 2017-03-20:
@jeubzzz

> The cause of this is not running PulseAudio in system-wide mode.

See the docs - https://github.com/alexa-pi/AlexaPi/wiki/Audio-setup-&-debugging#pulseaudio