alexa-pi/AlexaPi #200: Choppy Audio and after an attempt to fix now won't start

znelson710, 2017-03-18:
**Your OS (including version) where you are running AlexaPi:**
Running on rpi 3 raspian

**Python release (`python2 --version`):**
not sure?

**Description of problem:**
Alexa audio was choppy and now won't start after messing around with audio troubleshooting

**Problem-relevant `config.yaml` entries:**
not sure

**Steps to reproduce:**

1. 
2. 
3. 

**Traceback (if applicable):**
```

```

**Additional info:**

First time really messing with raspberry pi. Install went easy enough with just installing and editing files as everything was well explained. I had Amazon's release working but the trigger kept crashing and it did not autostart. Found this one with a nifty button and LEDs and Autostart. I can post whatever is needed but I don't know what that is. Also when I try to re run set up it just says "command not found" This is eventually to be put into a smart calendar running Alexa in the background and chrome in kiosk mode.

renekliment, 2017-03-19:
Hey. Please check the newly added [Post-installation steps](https://github.com/alexa-pi/AlexaPi/wiki/Installation#3-post-installation-steps) section of the Installation guide.

EmerickH, 2017-06-14:
@znelson710 you don't replied so I suppose you solved your issue...
Please re-open the issue and provide more details if you still have the problem.

Emerick

closed 2017-06-14 by EmerickH