alexa-pi/AlexaPi #201: after "Hello" i am pressing button but alexa is not replay?

disheet, 2017-03-18:
Thank you for reporting an issue with **AlexaPi**.

Make sure you are running the latest version of **AlexaPi**, and look through [issues](https://github.com/alexa-pi/AlexaPi/issues?utf8=%E2%9C%93&q=is%3Aissue) before submitting.

Check out our [Wiki](https://github.com/alexa-pi/AlexaPi/wiki/), and make sure to look at [Debugging](https://github.com/alexa-pi/AlexaPi/wiki/Debugging/), and [Audio debugging](https://github.com/alexa-pi/AlexaPi/wiki/Audio-setup-&-debugging) in particular before filing an issue.

In order to help troubleshooting, be sure to include the following information:

**Output of the [audio debugging script](https://github.com/alexa-pi/AlexaPi/wiki/Audio-setup-&-debugging#audio-debugging-script).**

**Your OS (including version) where you are running AlexaPi:**
```
```

**Your hardware platform and model you are running on:**
```

```

**Python release (`python2 --version`):**
```

```

**Description of problem:**
```

```

**Expected:**
```

```

**Problem-relevant `config.yaml` entries:**
```

```

**Steps to reproduce:**

1. 
2. 
3. 

**Traceback (if applicable):**
```

```

**Additional info:**
```

```

renekliment, 2017-03-19:
Hey. Please check the newly added [Post-installation steps](https://github.com/alexa-pi/AlexaPi/wiki/Installation#3-post-installation-steps) section of the Installation guide.

Also, this is not a proper issue report.

EmerickH, 2017-06-14:
@disheet like there's not enough information I close this issue, if you still have the problem, please re-open it and provide more information.

Emerick

closed 2017-06-14 by EmerickH