alexa-pi/AlexaPi #203: Smart home devices discovery

Kinjalk, 2017-03-24:
Hello,
I have followed amazon's documentation on creating Alexa smart home devices and i am successfully able to add devices in my Alexa app using Echosim.io or directly scanning in Alexa app. Also i am able to control devices using Echosim but unable to discover devices using Alexa on raspberry pi. Alexa always says "Could not find any smart home devices" even though all the devices are already in my Alexa app.

Is there any way to control devices using Alexa on raspberry pi ? I know i can use Alexa Skill and AWSIot but then i would have to use invocation. Can i use it without invocation ?
[image]

renekliment, 2017-03-26:
Hey.

> I have followed amazon's documentation on creating Alexa smart home devices and i am successfully able to add devices in my Alexa app using Echosim.io or directly scanning in Alexa app.

Can you elaborate on this? How & where do you have those devices connected? What are they? Do you also have an Echo?

Kinjalk, 2017-03-26:
Hello @renekliment  
This is my lambda-
[lambda.txt](https://github.com/alexa-pi/AlexaPi/files/870676/lambda.txt)

After creating lambda and smart home skill, I activated skill using Alexa app. All four devices were visible in devices section as you can see in the screen shoot. 

No i don't have an Echo , I am trying to control those devices via Alexa on raspberry pi but Alexa is unable to detect them , Surprisingly echosim.io is able to detect when is ask it to "Discover devices" and even control them using utterances like "turn on <device name>".

I think Echosim is using new APIs ! Am I correct ?

renekliment, 2017-03-26:
@Kinjalk I see, that makes sense now.

Yes, it seems like Echoism uses the new API. I'm afraid it might be API version related, even though it doesn't make much sense.

Kinjalk, 2017-03-26:
@renekliment Thanks

shivasiddharth, 2017-04-06:
hi,

I am trying to detect a Wemo switch on the alexa app, but it returns nothing. Can you give a briefing on the steps to follow?
regards

illperipherals, 2017-04-06:
@shivasiddharth , this isn't what you are looking for, but you may find it of interest.
https://www.hackster.io/kuzma/alexa-voice-control-lights-with-nodemcu-1bddb4
join gitter if you have an opportunity.

Kinjalk, 2017-04-07:
@shivasiddharth  
Hello, 
Unfortunately you **wont't be able to detect and control** any local or cloud based smart home devices using this repository as **it implements older APIs**. You can use [alexa-avs-sample-app](https://github.com/alexa/alexa-avs-sample-app) which implements **new APIs** and follow my post [#621](https://github.com/alexa/alexa-avs-sample-app/issues/621) to setup cloud based smart home devices using **AWSIoT** and **Smart home skill**. Also You won't be able to detect any physical devices without writing your own smart home skill. In any case (new or old APIs) Alexa won't be able to connect Wemo switch. 

Ps- You will only be able to use ESP8266 as in link mentoned by @illperipherals iff you have genuine Echo device.

shivasiddharth, 2017-04-07:
@illperipherals , thanks for the share.
@Kinjalk , I thought, both shared the same APIs. I agree with you, alexa-avs-sample-app is really good. I have it running on windows. Thanks for your thoughts..

monsieurtalbot, 2017-04-14:
@Kinjalk  I'd love some more information or links on the exact method you used with the official app...?

I've used a number of methods to get Alexa on my Raspberries around the house - I have yet to find one that's truly flawless other than AlexaPi - problem is the old API just isn't cutting it for me anymore...  I've found methods that feature Autostart but their implementation is touch and go - it doesn't work after reboot sometimes.

Kinjalk, 2017-04-15:
@monsieurtalbot  

1. Install [alexa-avs-sample-app](https://github.com/alexa/alexa-avs-sample-app) as instructed in [wiki](https://github.com/alexa/alexa-avs-sample-app/wiki/Raspberry-Pi) on fresh Raspbian.
   Also add the following under Web Settings 
   Allowed Origins -  `http://localhost:5000` and` http://ALEXA.DEVICE.IP.ADDRESS:5000`
   Allowed Return URLs - `http://localhost:5000/code` and `http://ALEXA.DEVICE.IP.ADDRESS:5000/code`. 

2. Check if installation was successful and everything is fine by running all three command.
3. Go to `Desktop/alexa-avs-sample-app/samples/javaclient/src/main/java/com/amazon/alexa/avs` and replace **AVSApp.java** with **AVSApp.java** in [Alexa.zip](https://github.com/alexa-pi/AlexaPi/files/923395/Alexa.zip)
4. Go to `Desktop/alexa-avs-sample-app/samples/javaclient/src/main/java/com/amazon/alexa/avs/auth/companionservice` . Replace **CompanionServiceAuthManager.java** with **CompanionServiceAuthManager.java** and **CompanionServiceClient.java** with **CompanionServiceClient.java** in [Alexa.zip](https://github.com/alexa-pi/AlexaPi/files/923395/Alexa.zip)
5. Go to `Desktop/alexa-avs-sample-app/samples/javaclient/src/main/resources/res` and Paste **hello.mp3** form [Alexa.zip](https://github.com/alexa-pi/AlexaPi/files/923395/Alexa.zip)
6. Go to `Desktop/alexa-avs-sample-app/samples/javaclient` . Paste **Login.py** and **creds.py** form zip. Make sure you have added **ProductID**, **Security_Profile_Description**, **Security_Profile_ID**, **Client_ID** and **Client_Secret** in **creds.py**
7. `sudo pip install CherryPy` and `sudo pip install requests`
8. `cd ~/Desktop/alexa-avs-sample-app/samples/javaclient` and 
9. `sudo rm -r target/`
10. `mvn validate`
11. `mvn install`
12. `python Login.py`
13. Navigate to `http://localhost:5000` in browser in raspberry pi and follow the steps
14. After successful Login stop the scrip and check if **refresh_token** is added to **creds.py** 
15. Go to `Desktop` and Paste **wakeWordAgent.py** and **alexaClient.py** from [Alexa.zip](https://github.com/alexa-pi/AlexaPi/files/923395/Alexa.zip)
16. `chmod +x /home/pi/wakeWordAgent.py` and `chmod +x /home/pi/alexaClient.py`
17. `cd /lib/systemd/system/`
18. `sudo nano alexaclient.service `
19. Paste  
`[Unit]`
`Description=Start Alexa Client`
`After=multi-user.target`
 `[Service]`
`Type=simple`
`ExecStart=/usr/bin/python /home/pi/alexaClient.py`
`Restart=on-abort`
` [Install]`
`WantedBy=multi-user.target`
20. Save the file and `sudo chmod 644 /lib/systemd/system/alexaclient.service`
21. `sudo systemctl daemon-reload`
`sudo systemctl enable alexaclient.service`
22. `sudo nano alexawakeword.service `
23. Paste  
`[Unit]`
`Description=Start Wake word Agent`
`After=multi-user.target`
 `[Service]`
`Type=simple`
`ExecStart=/usr/bin/python /home/pi/wakeWordAgent.py`
`Restart=on-abort`
` [Install]`
`WantedBy=multi-user.target`
24. Save the file and `sudo chmod 644 /lib/systemd/system/alexawakeword.service`
25. `sudo systemctl daemon-reload`
26. All done ! Now  `cd ~/Desktop/alexa-avs-sample-app/samples/javaclient` and `mvn exec:exec` If everything was fine Alexa will start. You can test it here. 
27. restart Pi and See if Alexa says Hello on boot. It may take some time to start Alexa so wait. 

All the codes are not so clean. Please let me know if it works for you. 
[Alexa.zip](https://github.com/alexa-pi/AlexaPi/files/923395/Alexa.zip)

bjchristenson, 2017-04-26:
@Kinjalk Thank you for putting these steps together. On Step #3, the file **AVSApp.java** is not found in that location nor is it in the repository. So I imagine that this steps would be to place and not replace. Correct?

Kinjalk, 2017-04-26:
@bjchristenson  Steps I mentioned above are for old repo. As the repository was updated on 20 April you will find alot of changes . If you have a copy of previous repo , these steps will work fine . 
New repo persists refresh token hence you no longer need to log in again and again but it still have GUI , you can not start it on boot .

bjchristenson, 2017-04-26:
@Kinjalk Thanks! That worked great. I missed that there was an update to the repo

Kinjalk, 2017-04-26:
@bjchristenson You are welcome. Did you use old or new repo ?

bjchristenson, 2017-04-27:
@Kinjalk I ended up using the New repo. The three terminal commands from the initial wiki were added to individual scripts (initially for testing/validation) and then invocation of the scripts were added to the end of /home/pi/.config/lxsession/LXDE-pi/autostart 
**WorkWordAgent** in the 3rd script requires AVS Auth in script 2 to be complete, so I added a pause of 30 seconds to the 3rd script and then added `omxplayer -o local /home/pi/hello.mp3` prior to `cd wakeWordAgent/src && ./wakeWordAgent -e sensory`

Definitely not elegant, but the goal was to get it up and running as a prototype and all is well.

Kinjalk, 2017-04-27:
@bjchristenson Cool !

J0hn-, 2017-05-19:
Hi!
I have comment line 91,92 and 105 but I have this error…
```
[INFO] --- exec-maven-plugin:1.2.1:exec (default-cli) @ sample-java-client ---
14:08:07.561 [Thread-16] ERROR com.amazon.alexa.avs.auth.AuthSetup - Failed to start companion service client
java.awt.HeadlessException:
No X11 DISPLAY variable was set, but this program performed an operation which requires it.
        at java.awt.GraphicsEnvironment.checkHeadless(GraphicsEnvironment.java:204) ~[?:1.8.0_131]
        at java.awt.Window.<init>(Window.java:536) ~[?:1.8.0_131]
        at java.awt.Frame.<init>(Frame.java:420) ~[?:1.8.0_131]
        at java.awt.Frame.<init>(Frame.java:385) ~[?:1.8.0_131]
        at javax.swing.SwingUtilities$SharedOwnerFrame.<init>(SwingUtilities.java:1758) ~[?:1.8.0_131]
        at javax.swing.SwingUtilities.getSharedOwnerFrame(SwingUtilities.java:1833) ~[?:1.8.0_131]
        at javax.swing.JOptionPane.getRootFrame(JOptionPane.java:1696) ~[?:1.8.0_131]
        at javax.swing.JOptionPane.showOptionDialog(JOptionPane.java:863) ~[?:1.8.0_131]
        at javax.swing.JOptionPane.showConfirmDialog(JOptionPane.java:795) ~[?:1.8.0_131]
        at javax.swing.JOptionPane.showConfirmDialog(JOptionPane.java:757) ~[?:1.8.0_131]
        at javax.swing.JOptionPane.showConfirmDialog(JOptionPane.java:718) ~[?:1.8.0_131]
        at com.amazon.alexa.avs.ui.DialogFactory.showYesNoDialog(DialogFactory.java:37) ~[classes/:?]
        at com.amazon.alexa.avs.auth.companionservice.RegCodeDisplayHandler.handleAuthenticationCopyToClipboard(RegCodeDisplayHandler.java:66) ~[classes/:?]
        at com.amazon.alexa.avs.auth.companionservice.RegCodeDisplayHandler.displayRegCode(RegCodeDisplayHandler.java:60) ~[classes/:?]
        at com.amazon.alexa.avs.auth.companionservice.CompanionServiceAuthManager.requestRegistrationCode(CompanionServiceAuthManager.java:80) ~[classes/:?]
        at com.amazon.alexa.avs.auth.companionservice.CompanionServiceAuthManager.startNewProvisioningRequest(CompanionServiceAuthManager.java:68) ~[classes/:?]
        at com.amazon.alexa.avs.auth.companionservice.CompanionServiceAuthManager.startRemoteProvisioning(CompanionServiceAuthManager.java:63) ~[classes/:?]
        at com.amazon.alexa.avs.auth.AuthSetup$2.run(AuthSetup.java:91) [classes/:?]
```
Login.py managed to get a refresh token
No way to have a real java console app ? I don't want a graphical install on my Raspberry just for this…

Kinjalk, 2017-05-19:
@J0hn-  Did you add Login.py to repo ?? 
You don't need to add any file in New repo . Install Alexa and follow all the steps mentioned in wiki not the steps I wrote for old repo . After  logging in and checking if everything is working fine follow steps for New repo. Comment those lines and recompile .

samiultahir, 2017-11-26:
Can I control one of my smartlights with alexa-pi as it shows up on the alexa app and i can control it from the app but when I ask alexa to control it it just says 'I couldn't find the device'