alexa-pi/AlexaPi #205: Error message at start:Access denied

Muscatnuss, 2017-03-27:
When I try to start AVS with 'systemctl start AlexaPi.service' I get the following error message:
Failed to start AlexaPi.service: Access denied.
I tried the trouble shooting methods of the instructions but nothing worked. I deleted the whole program and reinstalled it three times but i was getting the same results every time.
When i scanned the programm with 'sudo journalctl -u AlexaPi.service' I couldn't see any errors.
I use a Rasperry Pi 3 with Raspbian Jessie wich has python 3
I'm new to Raspberry Pi so please be gentle.

maso27, 2017-03-27:
You need to put _sudo_ in front of the command.
So `sudo systemctl start AlexaPi.service`
And you can make it start at boot with `sudo systemctl enable AlexaPi.service`

Muscatnuss, 2017-03-28:
Thanks for helping me! Now it works.

closed 2017-03-28 by Muscatnuss

renekliment, 2017-03-28:
Added into https://github.com/alexa-pi/AlexaPi/wiki/Installation#3-post-installation-steps 
Thanks for reporting and sorry we missed this in the first place.