alexa-pi/AlexaPi #206: Bug or what?? running for few minutes suddenly not responding.

pleasurous, 2017-03-31:
Hello, 

I'm totally new at this. here it is.. 

I am installed AlexaPi along with official MagicMirror2  by michmich and it is running well. I tried to ask bunch of question to Alexa and it's responded. but it work for few minutes and suddenly stop responding.

when I checked the status with
```
sudo systemctl status AlexaPi.service -l
```
it goes like this
```
● AlexaPi.service - Alexa client for all your devices
   Loaded: loaded (/usr/lib/systemd/system/AlexaPi.service; enabled)
   Active: active (running) since Fri 2017-03-31 15:01:35 WIB; 38min ago
     Docs: https://github.com/alexa-pi/AlexaPi/wiki
 Main PID: 1911 (python)
   CGroup: /system.slice/AlexaPi.service
           └─1911 /usr/bin/python /opt/AlexaPi/src/main.py --daemon

Mar 31 15:06:46 raspberrypi python[1911]: self._send_output(message_body)
Mar 31 15:06:46 raspberrypi python[1911]: File "/usr/lib/python2.7/httplib.py", line 879, in _send_output
Mar 31 15:06:46 raspberrypi python[1911]: self.send(msg)
Mar 31 15:06:46 raspberrypi python[1911]: File "/usr/lib/python2.7/httplib.py", line 841, in send
Mar 31 15:06:46 raspberrypi python[1911]: self.connect()
Mar 31 15:06:46 raspberrypi python[1911]: File "/usr/local/lib/python2.7/dist-packages/requests/packages/urllib3/connection.py", line 284, in connect
Mar 31 15:06:46 raspberrypi python[1911]: conn = self._new_conn()
Mar 31 15:06:46 raspberrypi python[1911]: File "/usr/local/lib/python2.7/dist-packages/requests/packages/urllib3/connection.py", line 150, in _new_conn
Mar 31 15:06:46 raspberrypi python[1911]: self, "Failed to establish a new connection: %s" % e)
Mar 31 15:06:46 raspberrypi python[1911]: NewConnectionError: <requests.packages.urllib3.connection.VerifiedHTTPSConnection object at 0x6ec30ab0>: Failed to establish a new connection: [Errno -2] Name or service not known

```

I tried restart the program with
```
sudo systemctl restart AlexaPi.service
```
It does restarting the program Alexa work again for few minutes and then not responding again.

I am using python 2.7.9 on Raspi3 by the way.

did anyone facing the same problem as me? if it is how did you guys solve it. if there isn't could you guys help me.

renekliment, 2017-04-01:
> `Failed to establish a new connection: [Errno -2] Name or service not known`

This seems like DNS issue.

Provide the whole log ... `journalctl -u AlexaPi`

pleasurous, 2017-04-04:
Definitely DNS issue.. I checked my router setting and did a address binding and it solved. I don't kno if there a compability issue with my router. Anyway.. thanks for the hints. I mark this solved. 💃

closed 2017-04-04 by pleasurous

shivasiddharth, 2017-04-07:
@pleasurous , did you get this error over Wifi or ethernet? I  get this errror when i connect over wifi, but over ethernet it seems to work fine. Also please suggest the fix that you implemented.

pleasurous, 2017-04-07:
@shivasiddharth , I got this problem when I connect over Wifi. I do address binding on my tplink TL-WR940N while I using static ip but since i change the setting to dynamic the problem came again and I solved it by create address reservation.

In your case if alexa works over ethernet then it should be nothing wrong with your alexa installation. check your router first if it is using static or dynamic. and then try do address binding if static or address reservation if it is dynamic. and if the problem still exist open journal by typing ```journalctl -u AlexaPi``` in your terminal and put the result here.

shivasiddharth, 2017-04-07:
@pleasurous Thanks for your inputs