alexa-pi/AlexaPi #207: Chip 4.4 not working (headless)

PrudhviBoggavarapu, 2017-04-03:
Thank you for reporting an issue with **AlexaPi**.

Make sure you are running the latest version of **AlexaPi**, and look through [issues](https://github.com/alexa-pi/AlexaPi/issues?utf8=%E2%9C%93&q=is%3Aissue) before submitting.

Check out our [Wiki](https://github.com/alexa-pi/AlexaPi/wiki/), and make sure to look at [Debugging](https://github.com/alexa-pi/AlexaPi/wiki/Debugging/), and [Audio debugging](https://github.com/alexa-pi/AlexaPi/wiki/Audio-setup-&-debugging) in particular before filing an issue.

In order to help troubleshooting, be sure to include the following information:

**Output of the [audio debugging script](https://github.com/alexa-pi/AlexaPi/wiki/Audio-setup-&-debugging#audio-debugging-script).**
http://www.alsa-project.org/db/?f=d2417659c2275e061d15142fb2363f720fa12f6b
**Your OS (including version) where you are running AlexaPi:**
```
chip headless 4.4
```

**Your hardware platform and model you are running on:**
```
chip
headless
```

**Python release (`python2 --version`):**
```
2.7.9
```

**Description of problem:**
```
The chip script is crashing at 
`● AlexaPi.service - Alexa client for all your devices
   Loaded: loaded (/usr/lib/systemd/system/AlexaPi.service; enabled)
  Drop-In: /etc/systemd/system/AlexaPi.service.d
           └─usergroup-root.conf
   Active: failed (Result: exit-code) since Mon 2017-04-03 22:19:10 UTC; 3s ago
     Docs: https://github.com/alexa-pi/AlexaPi/wiki
  Process: 730 ExecStart=/usr/bin/python /opt/AlexaPi/src/main.py --daemon (code=exited, status=1/FAILURE)
 Main PID: 730 (code=exited, status=1/FAILURE)

Apr 03 22:19:10 chip python[730]: surround50:CARD=AK5370,DEV=0
Apr 03 22:19:10 chip python[730]: surround51:CARD=AK5370,DEV=0
Apr 03 22:19:10 chip python[730]: surround71:CARD=AK5370,DEV=0
Apr 03 22:19:10 chip python[730]: iec958:CARD=AK5370,DEV=0
Apr 03 22:19:10 chip python[730]: dmix:CARD=AK5370,DEV=0
Apr 03 22:19:10 chip python[730]: dsnoop:CARD=AK5370,DEV=0
Apr 03 22:19:10 chip python[730]: hw:CARD=AK5370,DEV=0
Apr 03 22:19:10 chip python[730]: plughw:CARD=AK5370,DEV=0
Apr 03 22:19:10 chip systemd[1]: AlexaPi.service: main process exited, code=exited, status=1/FAILURE
Apr 03 22:19:10 chip systemd[1]: Unit AlexaPi.service entered failed state.`
```

**Expected:**
```
it to work
```

**Problem-relevant `config.yaml` entries:**
```
idk

```

**Steps to reproduce:**
idk it never work
1. 
2. 
3. 

**Traceback (if applicable):**
```
idk
```

**Additional info:**
```
idk
```

renekliment, 2017-04-04:
`sudo journalctl -u AlexaPi.service` to get the full log. Then you'll see the nature of the problem - you need to set the `input_device` correctly. You are given a list of options.

closed 2017-04-04 by renekliment

PrudhviBoggavarapu, 2017-04-06:
CRITICAL: Your input_device 'plughw:1,0' is invalid. Use one of the following:
Apr 04 18:17:27 chip python[217]: null
Apr 04 18:17:28 chip python[217]: default:CARD=sun4icodec
Apr 04 18:17:28 chip python[217]: sysdefault:CARD=sun4icodec
Apr 04 18:17:28 chip python[217]: dmix:CARD=sun4icodec,DEV=0
Apr 04 18:17:28 chip python[217]: dsnoop:CARD=sun4icodec,DEV=0
Apr 04 18:17:28 chip python[217]: hw:CARD=sun4icodec,DEV=0
Apr 04 18:17:28 chip python[217]: plughw:CARD=sun4icodec,DEV=0
Apr 04 18:17:28 chip python[217]: default:CARD=AK5370
Apr 04 18:17:28 chip python[217]: sysdefault:CARD=AK5370
Apr 04 18:17:28 chip python[217]: front:CARD=AK5370,DEV=0
Apr 04 18:17:28 chip python[217]: surround21:CARD=AK5370,DEV=0
Apr 04 18:17:28 chip python[217]: surround40:CARD=AK5370,DEV=0
Apr 04 18:17:28 chip python[217]: surround41:CARD=AK5370,DEV=0
Apr 04 18:17:28 chip python[217]: surround50:CARD=AK5370,DEV=0
Apr 04 18:17:28 chip python[217]: surround51:CARD=AK5370,DEV=0
Apr 04 18:17:28 chip python[217]: surround71:CARD=AK5370,DEV=0
Apr 04 18:17:28 chip python[217]: iec958:CARD=AK5370,DEV=0
Apr 04 18:17:28 chip python[217]: dmix:CARD=AK5370,DEV=0
Apr 04 18:17:28 chip python[217]: dsnoop:CARD=AK5370,DEV=0
Apr 04 18:17:28 chip python[217]: hw:CARD=AK5370,DEV=0
Apr 04 18:17:28 chip python[217]: plughw:CARD=AK5370,DEV=0
I'm sorry but what do I do. This is my first time outside of home automation
I want to use ak5370 usb

maso27, 2017-04-06:
Copy the line for the device that you want to use into _/etc/opt/AlexaPi/config.yaml_ as your input device.

I suspect you want _plughw:CARD=AK5370,DEV=0_.

PrudhviBoggavarapu, 2017-04-06:
yes thanks can you add that to the installation page because niibs like me never used .yaml config files

PrudhviBoggavarapu, 2017-04-06:
Now I'm getting this 
● AlexaPi.service - Alexa client for all your devices
   Loaded: loaded (/usr/lib/systemd/system/AlexaPi.service; enabled)
  Drop-In: /etc/systemd/system/AlexaPi.service.d
           └─usergroup-root.conf
   Active: active (running) since Thu 2017-04-06 22:03:45 UTC; 23min ago
     Docs: https://github.com/alexa-pi/AlexaPi/wiki
 Main PID: 1312 (python)
   CGroup: /system.slice/AlexaPi.service
           └─1312 /usr/bin/python /opt/AlexaPi/src/main.py --daemon

Apr 06 22:03:57 chip python[1312]: INFO: AVS token: Obtained successfully
Apr 06 22:03:58 chip python[1312]: [00c78ff8] alsa audio output error: cannot open ALSA device "plughw:1": No such file or directory
Apr 06 22:03:58 chip python[1312]: [00c78ff8] core audio output error: Audio output failed
Apr 06 22:03:58 chip python[1312]: [00c78ff8] core audio output error: The audio device "plughw:1" could not be used:
Apr 06 22:03:58 chip python[1312]: No such file or directory.
Apr 06 22:03:58 chip python[1312]: [00c78ff8] core audio output error: module not functional
Apr 06 22:03:58 chip python[1312]: [b4212200] core decoder error: failed to create audio output
Apr 06 22:18:00 chip systemd[1]: Started Alexa client for all your devices.
Apr 06 22:24:05 chip systemd[1]: Started Alexa client for all your devices.
Apr 06 22:26:49 chip systemd[1]: Started Alexa client for all your devices.

PrudhviBoggavarapu, 2017-04-06:
now what config is

# See https://github.com/alexa-pi/AlexaPi/wiki/Audio-setup-&-debugging
# to get a better idea of how to set things in here

# Level of log output
# default: INFO
# set to DEBUG for verbose logging
logging: INFO

sound:
  # Name of your ALSA microphone/soundcard in arecord -L
  input_device: "plughw:CARD=AK5370,DEV=0"
  # This is here if none of the listed devices works for you and you need to use a value like "plughw:1"
  # This is an override that is normally not needed and is a last resort.
  allow_unlisted_input_device: false

  playback_handler: "vlc"

  # you can go here with alsa, pulse or others
  # keep blank for the best audio output selected by the underlying code
  output: ""
  # output device for the audio output specified above,
  # currently supports ALSA only
  output_device: "plughw:CARD=sun4i-codec,DEV=0"

  # if left empty, won't be passed to the underlying library / command if it supports it
  # if it doesn't 100% will be used
  default_volume: 80

  # Means the same as above, but is used for media (playing music for example)
  # Leave blank to use the same setting as for speech content
  media_output:
  media_output_device:
  media_default_volume:

  # Some devices clip off the beginning/end of audio files being played back. The
  # value is the number of seconds of silence to pad the beginning and end of the audio. Only
  # the sox handler supports this configuration option.
  playback_padding: "1"

PrudhviBoggavarapu, 2017-04-06:
config is 

 See https://github.com/alexa-pi/AlexaPi/wiki/Audio-setup-&-debugging
 to get a better idea of how to set things in here

 Level of log output
 default: INFO
 set to DEBUG for verbose logging
logging: INFO

sound:
  Name of your ALSA microphone/soundcard in arecord -L
  input_device: "plughw:CARD=AK5370,DEV=0"
  This is here if none of the listed devices works for you and you need to use a value like "plughw:1"
  This is an override that is normally not needed and is a last resort.
  allow_unlisted_input_device: false

  playback_handler: "vlc"

  you can go here with alsa, pulse or others
  keep blank for the best audio output selected by the underlying code
 output: ""
  output device for the audio output specified above,
  currently supports ALSA only
  output_device: "plughw:CARD=sun4i-codec,DEV=0"

  if left empty, won't be passed to the underlying library / command if it supports it
  if it doesn't 100% will be used
  default_volume: 80

  Means the same as above, but is used for media (playing music for example)
  Leave blank to use the same setting as for speech content
  media_output:
  media_output_device:
  media_default_volume:

   Some devices clip off the beginning/end of audio files being played back. The
   value is the number of seconds of silence to pad the beginning and end of the audio. Only
   the sox handler supports this configuration option.
  playback_padding: "1"
(removed the #)

matda59, 2017-04-18:
Hi There - i have exactly the same problem as well with my CHIP.

maso27, 2017-04-18:
@PrudhviBoggavarapu looking at the last thing you posted, it should be working.  It lists the last three starts as being successful.
You can try stopping the service temporarily with `sudo systemctl stop AlexaPi`, then start it in debug mode to watch what happens.  Use `sudo /opt/AlexaPi/src/main.py -d`. It might give you more clues.
Everything will be normal again after a reboot.

@matda59 if your problem is exactly the same, I suggest trying exactly the same thing. :)

But the main gotchas with the CHIP (listed in the wiki) are these:

- You need to run as _root_ to get the button to work
- You get much better audio with an external USB card
- Because of the external USB card, you'll probably have to play around with audio devices a bit