alexa-pi/AlexaPi #208: AVS refresh_token not found in the configuration file. Run the setup again to fix your installation

boospy, 2017-04-05:
Hello, 

i've setuped the software 2 times on my brand new Pi3. But i didn't start. Every setup the same error:

2017-04-06 01:29:48 DEBUG: Setting up playback handler: VlcHandler
2017-04-06 01:29:48 INFO: Checking Internet Connection ...
2017-04-06 01:29:48 DEBUG: Starting new HTTPS connection (1): api.amazon.com
2017-04-06 01:29:48 DEBUG: https://api.amazon.com:443 "GET /auth/o2/token HTTP/1.1" 404 29
2017-04-06 01:29:48 INFO: Connection OK
2017-04-06 01:29:48 CRITICAL: AVS refresh_token not found in the configuration file. Run the setup again to fix your installation (see project wiki for installation instructions).

The yaml configfile is setup correctly with the datas from the registert device on amazon. So where i can get this token? Maybe i understand something wrong...

Thanks a lot for helping. 
:+1:

boospy, 2017-04-05:
Sorry, fixed, didn't see the auth_web.py :)

closed 2017-04-05 by boospy

TinajaLabs, 2017-05-07:
Same issue for me... @boospy, how did you fix this/get your refresh_token
At the end of install:
...
Enter your Client Secret []: mysecret
Ready goto http://192.168.0.172:5050 or http://localhost:5050  to begin the auth process
(Press Ctrl-C to exit this script once authorization is complete)

The browser could not find a web page at: http://192.168.0.172:5050 (which is the internal IP of my alexapi raspi 3)..

So I proceeded to press ^C and moved on.  Was I supposed to install a web server?

running latest upgraded raspian...  

Thanks for any tips...
Chris

TinajaLabs, 2017-05-07:
I found a reference to /opt/AlexaPi/src/auth_web.py...  which can fix the problem of missing refresh_token 

One can simply run it again to start the web page, access it, it bounces to amazon, and fills in the refresh_token in config.yaml.

To start the web page:
```
cd  /opt/AlexaPi/src
sudo python auth_web.py
```

Thanks,
Chris.

Ebenjb, 2017-08-29:
Hi TinajaLabs
I am fresh newby so please excuse the lack of relevent skill set.
I followed the process as described in: https://github.com/xtools-at/AssistantPi/blob/master/README.md
but somehow managed to miss (IMHO) a vital step.
Running the main.py -d I get the error as described above "..missing refresh_token"
Could you please verify my following step? I need to install a web browser to enable the setup script to get the auth done?

Thanking you in advance
Eben.

Ebenjb, 2017-08-29:
Hi TinajaLabs
If I run the sudo python auth_web.py I get Ready goto http://mypiaddress:5050 or http://localhost:5050 to begin the auth process
(Press Ctrl-C to exit script once authorisation is complete)
How do I proceed from here if no auth is taking place?
Once again thanking you in advance
Kind regards
Eben

maso27, 2017-08-29:
@Ebenjb you are referencing the AssistantPi project, which is different from this one.  Issues with that project should be placed in its project pages.

If you are trying to make AlexaPi work, you can start at step 1 [here](https://github.com/alexa-pi/AlexaPi/wiki/Installation).

The _setup.sh_ script automatically runs _auth_web.py_ once it is complete, which is when you enter the information you created at step 1.

Ebenjb, 2017-08-29:
Hey Maso27
Thanks for the quick response
My bad, I searched for the "AVS refresh_token not found in the configuration file" and I ASS U MEd (excuse the pun) that this was what I was looking for.
This is actually my second build of the AlexaPi and the Google Home Assistant running on the same Pi.
Any ideas why I get this "AVS refresh_token not found error..." ?
I updated my security settings on the developer.amazon.com page to include http://localhost:5050 as well as http://192.x.x.x:5050 but still nothing happens when my /opt/AlexaPi/src/main.py is running.

maso27, 2017-08-29:
@Ebenjb 
Here is what you need:

  at developer.amazon.com:
-      allowed origin: http://192.x.x.x:5050
-      allowed return URL: http://192.x.x.x:5050/code
     (localhost stuff really doesn't need to be there)
 

  Run `sudo /opt/AlexaPi/src/auth_web.py` (assuming you've already installed via _setup.sh_)
   Enter:
-      Client_ID:
-      Client_Secret:
-      Device_Type_ID:          (note: this is also known as the Product ID)
-      Security_Profile_Description:
-      Security_Profile_ID:

   On a separate computer, in the same network, navigate to 192.x.x.x:5050
     Sign in with your amazon credentials

That will populate the _refresh_token_ field of _config.yaml_, and everything should work after that.

If that doesn't work, check to make sure you haven't entered something incorrectly.  There have been many troubles with this process, and every time (so far) it has been a typographical error of some sort.

dikkedimi, 2017-10-02:
I'm currently having this issue on a Rpi 3B.

Seem to have audio working, but can't test because of the AVS refresh token BS.
I've tried all suggestions above.

Also, echosim.io is saying "Oops, something went wrong." So maybe Alexa services are down? O_o
```
Oct 02 04:18:43 raspberrypi systemd[1]: Started Alexa client for all your devices.
Oct 02 04:18:48 raspberrypi python[5282]: INFO: Checking Internet Connection ...
Oct 02 04:18:48 raspberrypi python[5282]: INFO: Connection OK
Oct 02 04:18:48 raspberrypi python[5282]: CRITICAL: AVS refresh_token not found in the configuration file. Run the setup again to fix your installation (see project wiki for installation instruct
Oct 02 04:18:49 raspberrypi systemd[1]: AlexaPi.service: Main process exited, code=exited, status=1/FAILURE
Oct 02 04:18:49 raspberrypi systemd[1]: AlexaPi.service: Unit entered failed state.
Oct 02 04:18:49 raspberrypi systemd[1]: AlexaPi.service: Failed with result 'exit-code'.
Oct 02 04:27:11 raspberrypi systemd[1]: Started Alexa client for all your devices.
```

maso27, 2017-10-02:
Open your config file with `nano /etc/opt/AlexaPi/config.yaml`
Scroll down to the _refresh_token_ entry.  Do you have something in there?

If not, you need to run `sudo /opt/AlexaPi/src/auth_web.py` and follow what I wrote above to set it up.

dikkedimi, 2017-10-02:
Like I said, I have followed all these steps. I'll list them again in the order I did them in.
1. ran cd /opt; sudo ./AlexaPi/src/scripts/setup.sh
used default config on raspbian (so debian, raspberrypi, use systemd)
2. chose to also install airplay support (nice, handy, why not?)
3. keep old config or delete or edit? delete
4. entered authentication info in the following order (into fresh config file):
- Device Type ID (aka Product ID)
- Security Profile Description
- Security Profile ID
- Client ID
- Client secret
result:
Ready goto http://192.168.178.14:5050 or http://localhost:5050  to begin the auth process
(Press Ctrl-C to exit this script once authorization is complete)
5. Go to http://192.168.178.14:5050
6. page responds with:
```

> "We're sorry!
> An error occurred when we tried to process your request. Rest assured, we're already working on the problem and expect to resolve it shortly.
> 
> Hide Details
> Error Summary
> 400 Bad Request
> The parameter client_id is required
> Request Details
> scope=alexa%3Aall
> redirect_uri=http%3A%2F%2F192.168.178.14%3A5050%2Fcode
> response_type=code
> scope_data=%7B%22alexa%3Aall%22%3A+%7B%22"

```

all values were copy pasted and checked for correctness before repeating this process for the 5th time. I have made sure that I entered the correct IP adresses for the allowed origin and return url's, as well as localhost

I have the feeling I'm missing some info, but after re-re-re-reading the documentation I found only this topic when googling the log output:

`$ sudo journalctl -u AlexaPi.service
-- Logs begin at Thu 2016-11-03 18:16:42 CET, end at Mon 2017-10-02 18:26:27 CEST. --
Oct 02 04:44:08 raspberrypi systemd[1]: Started Alexa client for all your devices.
Oct 02 04:44:34 raspberrypi python[514]: INFO: Checking Internet Connection ...
Oct 02 04:44:35 raspberrypi python[514]: INFO: Connection OK
Oct 02 04:44:35 raspberrypi python[514]: CRITICAL: AVS refresh_token not found in the configuration file. Run the se
Oct 02 04:44:35 raspberrypi systemd[1]: AlexaPi.service: Main process exited, code=exited, status=1/FAILURE
Oct 02 04:44:35 raspberrypi systemd[1]: AlexaPi.service: Unit entered failed state.
Oct 02 04:44:35 raspberrypi systemd[1]: AlexaPi.service: Failed with result 'exit-code'.
Oct 02 05:02:33 raspberrypi systemd[1]: Started Alexa client for all your devices.
Oct 02 05:02:39 raspberrypi python[2581]: INFO: Checking Internet Connection ...
Oct 02 05:02:40 raspberrypi python[2581]: INFO: Connection OK
Oct 02 05:02:40 raspberrypi python[2581]: CRITICAL: AVS refresh_token not found in the configuration file. Run the s
Oct 02 05:02:40 raspberrypi systemd[1]: AlexaPi.service: Main process exited, code=exited, status=1/FAILURE
Oct 02 05:02:40 raspberrypi systemd[1]: AlexaPi.service: Unit entered failed state.
Oct 02 05:02:40 raspberrypi systemd[1]: AlexaPi.service: Failed with result 'exit-code'.
Oct 02 05:22:39 raspberrypi systemd[1]: Started Alexa client for all your devices.
Oct 02 05:22:43 raspberrypi python[3797]: INFO: Checking Internet Connection ...
Oct 02 05:22:43 raspberrypi python[3797]: INFO: Connection OK
Oct 02 05:22:43 raspberrypi python[3797]: CRITICAL: AVS refresh_token not found in the configuration file. Run the s
Oct 02 05:22:43 raspberrypi systemd[1]: AlexaPi.service: Main process exited, code=exited, status=1/FAILURE
Oct 02 05:22:43 raspberrypi systemd[1]: AlexaPi.service: Unit entered failed state.
Oct 02 05:22:43 raspberrypi systemd[1]: AlexaPi.service: Failed with result 'exit-code'.`

I feel I'm missing part of the process here. Whether it's the inconsistend field labeling or amazon's jungle of a developer dashboard I dunno.

running `sudo /opt/AlexaPi/src/auth_web.py` gave the same result as in steps 4,5 and 6. when I do it with an available config.yaml

7. Thus rebooted after removing all config.yaml files I could find.

8. ```
$ sudo /opt/AlexaPi/src/auth_web.py
Traceback (most recent call last):
  File "/opt/AlexaPi/src/auth_web.py", line 21, in <module>
    with open(alexapi.config.filename, 'r') as stream:
TypeError: coercing to Unicode: need string or buffer, NoneType found
```

9. assume fetal position and cry self to sleep

maso27, 2017-10-02:
I take it you have nothing entered in _config.yaml_ for your refresh_token.

So you've put your address into your account at http://developer.amazon.com.
allowed origin: http://192.168.178.14:5050
allowed return URL: http://192.168.178.14:5050/code

You're going to the web address http://192.168.178.14:5050 from a different computer within your network.

Following all that, the error it's listing is that "The parameter client_id is required".
Are you sure you're using the correct value for Client ID?

I'm not sure what's up with the error when you run _auth_web.py_.  
But this all kind of looks like it's not reading your _config.yaml_ file.

I'm sorry you're frustrated, but I don't really have a lot more to offer here.

dikkedimi, 2017-10-02:
Thanks for the quick reply maso27, really appreciate it.

No, the refresh token was never added to the config file.
I assume the refresh_token gets generated, during authentication, correct?
If not, it's not clear to me where I'm supposed to obtain it :( .

verified by copy pasting that the client ID is correct, formatted like this: amzn1.application-oa2-client.654aesr4654sdrsdre654se654

maso27, 2017-10-02:
Yeah, that's the right format for Client ID.
And yes, the _auth_web_ process is what generates the _refresh_token_.

I suspect that somehow it's not finding _/etc/opt/AlexaPi/config.yaml_, or else that there's something in that file that's preventing it from reading correctly.
From your statement above: all your developer.amazon.com information _is_ making it into _config.yaml_ right?  Client ID, Security Profile ID, etc?

Is there any way that something got added or removed from the config file that AlexaPi is trying to parse?  Also, any reason tabs would be in there instead of spaces or anything?

You could try 
```
sudo mv /etc/opt/AlexaPi/config.yaml /etc/opt/AlexaPi/config.orig.yaml
sudo cp /opt/AlexaPi/src/config.template.yaml /etc/opt/AlexaPi/config.yaml
```
to put the original template in there, then manually editing with 
```
sudo nano /etc/opt/AlexaPi/config,yaml
```
to paste in your parameters, and go from there to see if it helps.

dikkedimi, 2017-10-02:
yeah, I've confirmed that all the information makes it to config.yaml, except for the refresh_token.

If the web_auth.py does that, and it gives an error, maybe i should do a clean install... Do I just remove /opt/AlexaPi to do that?

maso27, 2017-10-02:
I suppose that would work alright for a rough reinstall.  Also, once you've installed the airplay support you don't need to do it again, so that might save some time.

And you could try the dev branch to see if it works differently too.

dikkedimi, 2017-10-02:
ok so cleaning up all config.yaml files and removing the previously cloned master branch I reinstalled using the dev branch. I am prompted to enter my Amazon dev info now, and logging in works. But when prompted to give access to my account it produces the same error. In the response are two tokens, but pasting them in the config file did not fix my issue.

dikkedimi, 2017-10-02:
So I just tried installing it with https://github.com/xtools-at/AssistantPi. The installation script is different, as it offers no option to install on older non-systemctl system. Still the same error though, so by now I'm assuming it's something I did on this install (probably installing google assistant seperately, which also uses python).. If it shows the same behaviour on a fresh raspbian I'll probably file a seperate issue, as we've recently jumped from jessie to stretch?

Praganzi, 2017-12-01:
I'm having identically similar issues to dikkedimi.  Fresh Scratch install tonight, update/upgrade/dist-upgrade.  USB set up, disk expanded, static IP; usual stuff.

Fresh git and repo install tonight.  AWS set up last night.  Did not take note, until recently, of not enabling bluetooth or media capabilities, since corrected.

IDs copied and pasted, and verbally dictation checked.
Used to get some More Info.  That was when I had confused my IDs.  Now I just get a nah.
[image]

s1h4d0w, 2018-02-26:
I have the exact same problem with The Amazon page giving an error. I feel like this is a problem related to Amazon itself?

I successfully ran the setup, entered all the required data, and then started the webservice. I visit the https://internalip:5050 page and get the Amazon page asking me to allow or cancel giving my app permissions. If I click allow it just returns an error:

[image]

[image]

I followed the tutorial as well as I could (needs to be updated, Amazon has changed the Developer Dashboard) and set up everything correctly.

Does anyone have any ideas?

Marcreed48, 2018-10-31:
Http://localhost:5050 reset the refresh token. I’m running AlexaPi.

Marcreed48, 2018-10-31:
You need to see if you entered 3000 or 5050 as your port.