alexa-pi/AlexaPi #212: Have headphone with mic, but want audio output with speakers

Asshen, 2017-04-14:
Hey guys.

I installed a fresh raspbian on my raspberry pi 3, and then installed AlexaPi using the instructions on this site.
AlexaPi seems to work great, but there's one little issue I can't get right :)

I use a headset with microphone.
But I only want the headset to be used for the microphone, I would like the audio output to be my external speakers connected to the mini-jack (no HDMI, I use a VGA monitor).

Can someone help me configure this?

Thank you!
Kris.

orrinwitt, 2017-04-16:
I'm in this exact boat. Cheap USB headset for the mic, speakers plugged into headphone jack.
Hope we get an answer.

renekliment, 2017-04-16:
Change your `output_device` in your config file (`/etc/opt/AlexaPi/config.yaml `). Use `aplay -L` to get the list of your sound devices. 
If you use PulseAudio, use a graphical tool such as `pavucontrol` to set desired sink. There are also some resources on how to do that via the CLI: https://askubuntu.com/questions/844392/trying-to-change-pulseaudio-application-sink-source-from-command-line

orrinwitt, 2017-04-17:
> I tried all device options that were shown using "aplay -L" excluding the
> headset options, but none seem to work.
> I used the command "aplay --device /usr/....../Singer2.wav".
>
> With some option I got the message "Unable to connect: access denied".
> Others seemed to execute, but I couldn't hear any audio playing.
>
> If I try to start "alsamixer", I also get an access denied error and the
> application seems to search for audio devices forever.
>
> Any ideas?

I was eventually able to get it, but I also could make no progress by
editing that file. I ended up getting the VNC server up and using
pavucontrol, but I'm not really even sure what I did other than set the
fallback device as the speaker.
But yeah, it seemed like no matter what I changed in the CLI, I would get
errors when I checked the status.

or mute
the thread

samiultahir, 2017-04-24:
I have the exact same problem!

But i know the cause but not a solution :(

When i enable PCM in my audio device settings the audio comes out of the headphones which i don't want

But when I disable PCM, the audio comes out of the speakers!

However, when I try to talk to alexa via the headsets microphone it wont respond or say anything. :(

EmerickH, 2017-06-14:
Like **renekliment** has said:
you have to change you input and output devices in your config, uses `aplay -L` to get the names.
Uses only `plughw:` devices.
If you don't know how to do, please post here the output of the `aplay -L` command and I will help you.

Emerick

Asshen, 2017-06-15:
I solved it by using Raspbian Lite in stead of the Raspbian version with Pixel.
Works like a charm.

closed 2017-06-15 by Asshen