alexa-pi/AlexaPi #213: Long delay between hotword detection and recording

s-t-e-f-a-n, 2017-04-16:
I use:
- Jessie lite
- Raspi 3
- PS3 Eye as Mic and speaker to line out
- python --version ; Python 2.7.9

Everything works fine except:
Is it a normal behaviour that the time between the hotword detection "Alexa" (which is ack´d by "yes" ) to the start of recording my question to Alexa is about 5..6 seconds? This is annoying and makes it almost useless. It needs to be able to start recording my request right after the "yes" ack of her.

This is the output of "main.py -d" with comments of me:
--- here "Alexa" is recognized ---
2017-04-16 19:32:19 DEBUG: Stopping audio play
2017-04-16 19:32:19 DEBUG: Playing audio: /opt/AlexaPi/src/resources/alexayes.mp3
2017-04-16 19:32:19 DEBUG: Player State: State.Opening
2017-04-16 19:32:19 DEBUG: Player State: State.Playing
2017-04-16 19:32:19 DEBUG: Started play.
--- Alexa says "Yes"
2017-04-16 19:32:20 DEBUG: Player State: State.Ended
2017-04-16 19:32:20 DEBUG: Finished play.
2017-04-16 19:32:20 DEBUG: Setting up recording
--- here it says that the recording starts..... ---
2017-04-16 19:32:20 DEBUG: Start recording
2017-04-16 19:32:20 DEBUG: Starting new HTTPS connection (1): access-alexa-na.amazon.com
--- ....but 6 seconds later the recording starts in reality (checked by saying numbers 1, 2, 3, ... and afterwareds checking /tmp/Alexa.../recording.wav ---
2017-04-16 19:32:26 DEBUG: Start sending speech to Alexa Voice Service
2017-04-16 19:32:28 DEBUG: End recording
2017-04-16 19:32:28 DEBUG: Finished sending speech to Alexa Voice Service
--- now response of Alexa is received and played back ---
2017-04-16 19:32:30 DEBUG: Processing Request Response...
2017-04-16 19:32:30 DEBUG: JSON String Returned: {
  "messageBody": {
    "directives": [
      {
        "namespace": "Speaker",
        "name": "SetMute",
        "payload": {
          "mute": false
        }
      },
      {
        "namespace": "SpeechSynthesizer",
        "name": "speak",
        "payload": {
          "contentIdentifier": "amzn1.as-ct.v1.Domain:Application:Weather#ACRI#WeatherPrompt.e8793d84-e0f1-4162-99b2-34bdcefd4efd",
          "audioContent": "cid:WeatherPrompt.e8793d84-e0f1-4162-99b2-34bdcefd4efd_2096753354"
        }
      }
    ]
  },
  "messageHeader": {}
}
2017-04-16 19:32:30 DEBUG: Stopping audio play
2017-04-16 19:32:30 DEBUG: Playing audio: file:///tmp/AlexaPi-runtime-zn99Fc/WeatherPrompt.e8793d84-e0f1-4162-99b2-34bdcefd4efd_2096753354.mp3
2017-04-16 19:32:30 DEBUG: Player State: State.Opening
2017-04-16 19:32:30 DEBUG: Player State: State.Playing
2017-04-16 19:32:30 DEBUG: Started play.
2017-04-16 19:32:45 DEBUG: Player State: State.Ended
2017-04-16 19:32:45 DEBUG: Finished play.

s-t-e-f-a-n, 2017-04-21:
I'd appreciated very much if anybody just could comment about his delay from hotword detection to start of recording the request speech. 

Thanks a lot in advance!

maso27, 2017-04-21:
So no, that's not common behavior. The PSEye is a bit of a hairy beast to get right, so I don't think anyone feels that they can really speak about it.

I'm not sure it's related though. Do you have a different device you could use as a mic to see if things improve?

If not I'd try a reinstall, maybe with a different SD card.

s-t-e-f-a-n, 2017-04-25:
Thank you for your reply, Mason.

I´ve tried some things, but it does not improve - it still takes 6 seconds from getting "Yes" (after triggering by "Alexa") to its readiness to listen/record my speech - i.e. between "DEBUG: Starting new HTTPS connection (1): access-alexa-na.amazon.com" and "DEBUG: Start sending speech to Alexa Voice Service". Each speech spoken right after "Yes" is not handled / is lost.

I tried:
- reinstalled on a fresh sd card (Jessie Lite; apt-get update / upgrade; sudo git clone https://github.com/alexa-pi/AlexaPi.git; sudo ./AlexaPi/src/scripts/setup.sh; configuring input in config.yaml and .asoundrc; checking that audio works; then start: sudo /opt/AlexaPi/src/main.py -d )
- tried a different usb mic (cat /proc/asound/cards: C-Media Electronics Inc. USB PnP Sound Device at usb-...)

Maybe I lost something at initialization of Alexapi or system? 
What is your delay in seconds between the "yes" ack and start of recording?

Thanks a lot in advance for any comments ;-)

s-t-e-f-a-n, 2017-05-01:
I debugged the code and additionally made some tests by directly sending a wav-file request with the help of a bash script (see: https://miguelmota.com/blog/alexa-voice-service-with-curl/):

The script:
```
TOKEN="Atza|IwEBIO-lPO_abuRS2..."
curl -i \
  -H "Authorization: Bearer ${TOKEN}" \
  -F "metadata=<metadata.json;type=application/json; charset=UTF-8" \
  -F "audio=<askfortime.wav;type=audio/L16; rate=16000; channels=1" \
  -o response.txt \
  https://access-alexa-na.amazon.com/v1/avs/speechrecognizer/recognize
```
Output of curl:
```
  % Total    % Received % Xferd  Average Speed   Time    Time     Time  Current
                                 Dload  Upload   Total   Spent    Left  Speed
  0     0    0     0    0     0      0      0 --:--:--  0:00:06 --:--:--     0

  % Total    % Received % Xferd  Average Speed   Time    Time     Time  Current
                                 Dload  Upload   Total   Spent    Left  Speed
100  134k    0 12803  100  121k   1585  15424  0:00:08  0:00:08 --:--:-- 27322
```
The response code in the response.txt is 200 and the mpeg answer of Alexa.

The same result for AlexaPi on Raspi 3 and with the bash script from the Raspi 3 and also from a Linux PC:
It takes 6 seconds before it starts to stream the audio-data.

I think the problem is the AVS. Does nobody else experience this delay? I mean, that makes it almost useless for a user as you have to wait 6 seconds after the trigger is activated to speak the command...

Please report, how long it takes for you guys....

Frostlock, 2017-05-05:
No idea on how to help you but for me it is practically instantaneous. As soon as she is finished saying yes I can ask my question.

s-t-e-f-a-n, 2017-05-07:
Thank you Frostlock for your comment. That motivated me to analyze my problem further...

I am pretty sure now, that the delay is produced by a 5-6 second DNS lookup time of access-alexa-na.amazon.com on my site.

Wireshark-Dump of relevant part (192.168.1.23 is the Alexa software on my site):
[image]

The first IPv4 DNS query (#13873) seems to be never answered. 
The second query (#13874) is a IPv6 query which promptly is answered (#13875)
The third query (#14061) is again a IPv4 query and sent >5 sec later. This one now is promptly answered  Then again a second same IPv6 query. After this the communication with access-alexa... is started.

Until now I cannot find the reason for this behaviour. Does anybody have a clue?

s-t-e-f-a-n, 2017-05-08:
My curl-tool directly requesting Alexa did also show the DNS delay effect. Now I could solve it for curl by just adding the "-4" parameter to the curl-request

`TOKEN="Atza|IwEBIO-lPO_abuRS2..."
 curl -i \
  -4 \
  -H "Authorization: Bearer ${TOKEN}" \
  -F "metadata=<metadata.json;type=application/json; charset=UTF-8" \
  -F "audio=<askfortime.wav;type=audio/L16; rate=16000; channels=1" \
  -o response.txt \
  https://access-alexa-na.amazon.com/v1/avs/speechrecognizer/recognize`

This one does get the Alexa-response promptly. So this is a workaround.

**But I have no clue how to force requests.post in Python to use ipv4. 
Is there anybody having an idea?**

s-t-e-f-a-n, 2017-05-08:
Yeah. I made it. To solution for me is completely to disable ipv6 on my Raspi by these two measures:

1. Added to /boot/cmdline.txt:
`ipv6.disable=1`

The effect is that ifconfig does not show ipv6 addresses.
But dmesg still shows that systemd loaded the ipv6 module.
The second step cures this:

2. Edited /etc/modprobe.d/ipv6.conf and let it look like this:
`alias net-pf-10 off`
`alias ipv6 off`
`options ipv6 disable_ipv6=1`
`blacklist ipv6`

After rebooting my AlexaPi promptly after the wakeword detection listens to my requests and answers with normal response time. This is great.

Still unclear is the reason for this. I guess either my router is somehow incoompatible to ipv6 or my cable ISP makes problems.

I close this issue. 

PS: Thanks to the forum and all AlexaPi contributors!

closed 2017-05-08 by s-t-e-f-a-n