alexa-pi/AlexaPi #214: Keeps crashing on startup

FreeSpeed, 2017-04-17:
Thank you for reporting an issue with **AlexaPi**.

Make sure you are running the latest version of **AlexaPi**, and look through [issues](https://github.com/alexa-pi/AlexaPi/issues?utf8=%E2%9C%93&q=is%3Aissue) before submitting.

Check out our [Wiki](https://github.com/alexa-pi/AlexaPi/wiki/), and make sure to look at [Debugging](https://github.com/alexa-pi/AlexaPi/wiki/Debugging/), and [Audio debugging](https://github.com/alexa-pi/AlexaPi/wiki/Audio-setup-&-debugging) in particular before filing an issue.

In order to help troubleshooting, be sure to include the following information:

**Output of the [audio debugging script](https://github.com/alexa-pi/AlexaPi/wiki/Audio-setup-&-debugging#audio-debugging-script).**

**Your OS (including version) where you are running AlexaPi:**
PRETTY_NAME="Raspbian GNU/Linux 8 (jessie)"
NAME="Raspbian GNU/Linux"
VERSION_ID="8"
VERSION="8 (jessie)"
ID=raspbian
ID_LIKE=debian

**Your hardware platform and model you are running on:**
Raspberry pi 3

**Python release (`python2 --version`):**
Python 3.4.2

**Description of problem:**

Keeps crashing on startup
 raspberrypi systemd[1]: Unit AlexaPi.service entered failed state.

**Expected:**
```

```

**Problem-relevant `config.yaml` entries:**
```

```

**Steps to reproduce:**

1. just following this guide [http://lifehacker.com/the-simplest-way-to-build-a-raspberry-pi-powered-amazon-1794218212]
2. 
3. 

**Traceback (if applicable):**
```

```

**Additional info:**
```
Thank you
```

maso27, 2017-04-17:
Have you tried any debugging so far?  This isn't really a lot of information to go on.

Note the links at the top of your post for [issues](https://github.com/alexa-pi/AlexaPi/issues?utf8=%E2%9C%93&q=is%3Aissue), [debugging](https://github.com/alexa-pi/AlexaPi/wiki/Debugging/), and [audio debugging](https://github.com/alexa-pi/AlexaPi/wiki/Audio-setup-&-debugging).  There's a fair bit of information on the [wiki](https://github.com/alexa-pi/AlexaPi/wiki/) too.

You can try typing `sudo journalctl -u AlexaPi` to see if there's any pertinent information about why it entered a failed state.

FreeSpeed, 2017-04-18:
I think it is because my audio  settings plughw:CARD=Device,DEV=0' is my headset when i  use    `aplay -l` my headset is a usb headset
-Crash
`[pulseaudio] bluez4-util.c: org.bluez.Manager.GetProperties() failed: org.freedesktop.DBus.Error.UnknownMethod: Method "GetProperties" with signature "" on interface "org.bluez.Man
Apr 17 20:06:19 raspberrypi python[984]: CRITICAL: Your input_device 'plughw:CARD=Device,DEV=0' is invalid. Use one of the following:
Apr 17 20:06:19 raspberrypi python[984]: default
Apr 17 20:06:19 raspberrypi python[984]: null
Apr 17 20:06:19 raspberrypi python[984]: pulse
Apr 17 20:06:19 raspberrypi python[984]: sysdefault:CARD=Headset
Apr 17 20:06:19 raspberrypi python[984]: front:CARD=Headset,DEV=0
Apr 17 20:06:19 raspberrypi python[984]: surround21:CARD=Headset,DEV=0
Apr 17 20:06:19 raspberrypi python[984]: surround40:CARD=Headset,DEV=0
Apr 17 20:06:19 raspberrypi python[984]: surround41:CARD=Headset,DEV=0
Apr 17 20:06:19 raspberrypi python[984]: surround50:CARD=Headset,DEV=0
Apr 17 20:06:19 raspberrypi python[984]: surround51:CARD=Headset,DEV=0
Apr 17 20:06:19 raspberrypi python[984]: surround71:CARD=Headset,DEV=0
Apr 17 20:06:19 raspberrypi python[984]: iec958:CARD=Headset,DEV=0
Apr 17 20:06:19 raspberrypi python[984]: dmix:CARD=Headset,DEV=0
Apr 17 20:06:19 raspberrypi python[984]: dsnoop:CARD=Headset,DEV=0
Apr 17 20:06:19 raspberrypi python[984]: hw:CARD=Headset,DEV=0
Apr 17 20:06:19 raspberrypi python[984]: plughw:CARD=Headset,DEV=0
`
-Audio output configuration
`**** List of PLAYBACK Hardware Devices ****
card 0: ALSA [bcm2835 ALSA], device 0: bcm2835 ALSA [bcm2835 ALSA]
  Subdevices: 8/8
  Subdevice #0: subdevice #0
  Subdevice #1: subdevice #1
  Subdevice #2: subdevice #2
  Subdevice #3: subdevice #3
  Subdevice #4: subdevice #4
  Subdevice #5: subdevice #5
  Subdevice #6: subdevice #6
  Subdevice #7: subdevice #7
card 0: ALSA [bcm2835 ALSA], device 1: bcm2835 ALSA [bcm2835 IEC958/HDMI]
  Subdevices: 1/1
  Subdevice #0: subdevice #0
card 1: Headset [Turtle Beach P11 Headset], device 0: USB Audio [USB Audio]
  Subdevices: 1/1
  Subdevice #0: subdevice #0
`

maso27, 2017-04-18:
Okay, this helps.
See in the crash log, it says "Your input_device... is invalid.  Use one of the following:"
copy and paste one of the lines listed after that into _/etc/opt/AlexaPi/config.yaml_ as your input_device.

Most likely you'll want to use 
_input_device = plughw:CARD=Headset,DEV=0_

FreeSpeed, 2017-04-18:
I don't have a config file only a template in that folder. I'm going to reinstall it
Thank you for your help

maso27, 2017-04-18:
The template located in _/opt/AlexaPi/src_ is copied into _/etc/opt/AlexaPi_ at install.
I think you're looking in the wrong place, and will waste your time on a reinstall.

maso27, 2017-04-18:
If I'm wrong, you can copy the template into the proper place with:
`sudo cp /opt/AlexaPi/src/config.template.yaml /etc/opt/AlexaPi/config.yaml`

Then edit the _input_device_ to be what you need.

FreeSpeed, 2017-04-18:
I already did it oops.
But there was no config file in the folder but i could have used the template.
But thank you

FreeSpeed, 2017-04-18:
i've encountered another problem.
I can hear Alexa say "hello" when it starts up but when is say "alexa" it doesn't respond.
My mic works

maso27, 2017-04-18:
This could be due to a number of things.

1.) Verify you're definitely using the right input device following "checking if your system's audio works" [here](https://github.com/alexa-pi/AlexaPi/wiki/Audio-setup-&-debugging).

2.) Adjust your system volume with `alsamixer`.  You can make any changes permanent over reboots with `alsactl store` (might need a `sudo` in front of that).

3.) Try changing the detection threshold within _config.yaml_.  You can restart and see if the change has helped with `sudo systemctl restart AlexaPi`.

4.) Try a different trigger word by changing it within _config.yaml_.  Alternatively, you could let someone else try saying "Alexa" and see if it triggers.  (My box has never responded to my wife's voice, ever.  It's become a bit of a joke in my house.)

See if any of that helps.

EmerickH, 2017-06-10:
Hello,
It's just that you can't use `plughw`!
Use `hw:CARD=Headset,DEV=0` instead.

Emerick

EmerickH, 2017-06-14:
@FreeSpeed have you solved your issue?

jasonlehman, 2017-06-27:
I used the ps3 eye as well & ran into the same issue of CRITICAL: Your input_device 'plughw:CARD=Device,DEV=0' is invalid. Use one of the following:...

I had to change it to the following to get it to work...
run this...
`sudo nano /etc/opt/AlexaPi/config.yaml`
then change your input_device to "default:CARD=CameraB404271"
If you're using the ps3 eye, you may have different characters after the word Camera.
Here is my sound section of config.yaml
sound:
# Name of your ALSA microphone/soundcard in arecord -L
  input_device: "default:CARD=CameraB404271"
  # This is here if none of the listed device: "raspberrypi"
  # This is an override that is normally not needed and is a last resort.
  allow_unlisted_input_device: false

EmerickH, 2017-07-20:
So it's solved, I'm wrong?

Emerick

closed 2017-07-20 by EmerickH